libplctag/libplctag.NET #4: Support .NET Framework

timyhac, 2019-09-07:
Currently does not support .NET Framework.

Workaround is to manually copy plctag.dll into same directory as libplctag.dll

jkoplo, 2020-07-04:
What's the failure mode? I know I used one of the other libplctag wrappers with Framework. I actually just realized I've been defaulting to core for any tests I've run in CLI lately on this library.

timyhac, 2020-07-04:
I haven't tried it recently but last year when i was working on this i
recall you just needed to copy in the requisite core plctag.dll into the
right directory.

timyhac, 2020-07-08:
This is a useful read: https://ericsink.com/entries/native_library.html

timyhac, 2020-07-08:
I just pushed a change last night that seems to make our library work for both core and framework, I haven't tested many scenarios though.

timyhac, 2020-07-09:
Glad to say this is fixed! The trick was using GetExecutingAssembly instead of GetCallingAssembly. ec5e021

closed 2020-07-09 by timyhac