libplctag/libplctag.NET #8: Expose asynchronous nature of libplctag

timyhac, 2019-09-07:
If timeouts are provided on Write() and Read() in libplctag, the function returns immediately and is queued for later. Expose this to the wrapper in an idiomatic way.

https://github.com/kyle-github/libplctag/wiki/API

timyhac, 2020-07-04:
@jkoplo and @kyle-github - I'm starting work on this feature and need a little clarification on how the locking works.

Which APIs need to be locked - is it enough to lock on plc_tag_read and plc_tag_write?
I've implemented assuming this: https://github.com/libplctag/libplctag.NET/tree/multithread

Are there any situations where you wouldn't want to lock against multithreaded use?

kyle-github, 2020-07-04:
Hi @timyhac,

I am not sure I understand what you are trying to do.   The library is fundamentally thread-safe.  You can call plc_tag_read() from multiple threads and the library will do something sane (perhaps not what you want, but it won't corrupt its internal state).   With very limited exceptions, all API entry points are protected by mutexes.  There is a test in the examples in stress_api_lock.c.

If you provide a timeout to plc_tag_read(), then it will block up to that long before timing out.   Same with plc_tag_write().   If you pass a zero timeout then the functions will queue up the request internally and return immediately.   The only functions with this synchronous/asynchronous behavior are plc_tag_create(), plc_tag_read() and plc_tag_write().   Arguably I probably should have made plc_tag_abort() take a timeout as well, but so far I have been able to make aborts work fully in the background.  All other functions are immediate.   But as noted above, all are protected by a mutex that must be entered in order to examine or change internal tag state.  This includes reading an int32 from a tag, getting its status, or even getting the tag size (which can change if the tag is being read in as you check the size!).

I provide plc_tag_lock() and plc_tag_unlock() to expose higher level locking for systems that do not have good native locking.  Some operations only make sense as a sequence of API calls.   In that case, the following can be done:

```c
rc = plc_tag_lock();
    ... plc_tag_read() ...
    ...
    my_val = plc_tag_get_int32(tag, 42);
    my_val++;
    plc_tag_set_int32(tag, 42);
    ...
    ... plc_tag_write() ...
plc_tag_unlock();
```
The above would read in a value from the PLC, increment it and write it back out.   Without some cross-API level of locking, this would fail to provide correct results in the face of multiple threads trying to run the same code.   Normally, I would suggest that the native system locking be used such as synchronized blocks in Java.   But for those systems that do not have that, there are the library functions.

That probably did not help, did it?

timyhac, 2020-07-04:
In that example that you gave, it will work if the lock()/unlock()
functions were called, because those functions would make the inner code
accessible by only one thread at a time - right?

The reason you say that calling plc_tag_read() from multiple threads might
not be what the user wants, is because they need to also request the data
with a get_uint8 (for example) - and in the intervening period it might
have been updated by a new value - right?

Maybe this is outside the scope of the wrapper, c# has language-level
locking primitives that the user can make use of - its not part of the core
library, so it doesn't need to be part of the wrapper either.

In terms of the end-result I'm going after:
Polling (i.e. using plc_tag_status after a plc_tag_read while  status =
PLCTAG_STATUS_PENDING) isn't super common in the C# sharp world as far as
I'm aware - so I'm looking to make asynchronous versions of Read/Write in a
way that is more idiomatic for C# usage.

Something like this:
public Read(int timeout)                          synchronous version that
exists now
public async Task ReadAsync()              new proposed asynchronous version

timyhac, 2020-07-04:
In the example that you give, it will work if the `plc_tag_lock()`/`plc_tag_unlock()` functions are called, because those functions would make the inner code accessible by only one thread at a time - right?

The reason you say that calling `plc_tag_read()` from multiple threads might not be what the user wants, is because they need to also request the data with a `plc_tag_get_uint8` (for example) - and in the intervening period it might have been updated by a new value - right?

Maybe this is outside the scope of the wrapper, c# has language-level locking primitives that the user can make use of - its not part of the core library, so it doesn't need to be part of the wrapper either.

In terms of the end-result I'm going after:
Polling (i.e. using `plc_tag_status` after a `plc_tag_read` while  `status = PLCTAG_STATUS_PENDING`) isn't super common in the C# sharp world as far as I'm aware - so I'm looking to make asynchronous versions of Read/Write in a way that is more idiomatic for C# usage.

Something like this:
`public Read(int timeout)`                          synchronous version that exists now
`public async Task ReadAsync()`              new proposed asynchronous version

timyhac, 2020-07-04:
I've been thinking about how this could be done. We could either use status polling or use the event system. The event system is interesting because I assume it would mean better performance than polling, but I think polling would be way simpler to implement. After all its only memory that we're polling, its not going across the network.

This excerpt doesn't work but I imagine it would have to be something like this to use the event system.

```csharp
public async Task ReadAsync(CancellationToken cancellationToken = default)
{

    TaskCompletionSource<object> readTask = new TaskCompletionSource<object>();

    Action<int, int, int> eventCallback = delegate(int tag_id, int event_id, int status)
        {

            // Ignore if the event was for a differnt tag
            if (tag_id != pointer)
                return;

            try
            {

                switch ((Events)event_id)
                {
                    case Events.PLCTAG_EVENT_READ_COMPLETED:
                        readTask.SetResult(default);
                        break;
                    case Events.PLCTAG_EVENT_ABORTED:
                    case Events.PLCTAG_EVENT_DESTROYED:
                        readTask.SetCanceled();
                        break;
                    case Events.PLCTAG_EVENT_READ_STARTED:
                    case Events.PLCTAG_EVENT_WRITE_STARTED:
                    case Events.PLCTAG_EVENT_WRITE_COMPLETED:
                        // Irrelevant, do nothing
                        break;
                    default:
                        throw new System.NotImplementedException();
                }

            // We've finished dealing with this, unregister this callback
            plctag.unregister_callback(pointer);
            ThrowIfError();

            }
            catch (Exception e)
            {
                readTask.SetException(e);
            }

        };

    Task.Run(() =>
    {
        plctag.register_callback(pointer, new plctag.callback_func(eventCallback));
        ThrowIfError();

        plctag.read(pointer, 0);

        Task.Delay(Timeout.InfiniteTimeSpan);

    }).Wait(cancellationToken);

    await readTask.Task;

}
```

Polling on the other hand would be much simpler

```csharp
public async Task ReadAsync(CancellationToken cancellationToken = default)
{

    int sleepPeriod = 10;

    Read(TimeSpan.Zero);
    while (GetStatus() == StatusCode.PLCTAG_STATUS_PENDING)
    {
        cancellationToken.ThrowIfCancellationRequested();
        await Task.Delay(sleepPeriod);
    }
}
```

The only issue is that we would have to choose a value for sleepPeriod.
It would be used like this:
```csharp
var myTag = new Tag(IPAddress.Parse("192.168.0.10"), "1,0", CpuTypes.LGX, DataTypes.DINT, "MY_DINT");
while (myTag.GetStatus() == StatusCode.PLCTAG_STATUS_PENDING)
{
    Thread.Sleep(100);
}
myTag.ThrowIfError();

await myTag.ReadAsync();
myTag.ThrowIfError();
```

One thing that this exercise has shown me is that it probably makes sense for the wrapper to do the Status checks as part of the operations. It doesn't really make sense for them to ask the user to do this if they have to do it 100% of the time.

That means we could reduce the above code to be simply:
```csharp
var myTag = new Tag(IPAddress.Parse("192.168.0.10"), "1,0", CpuTypes.LGX, DataTypes.DINT, "MY_DINT");
await myTag.ReadAsync();
```

Maybe this API is less "thin" than it used to be, but it isn't introducing any new functionality - its just converting the core library into something idiomatic for .NET users.

kyle-github, 2020-07-04:
> In the example that you give, it will work if the plc_tag_lock()/plc_tag_unlock() functions are called, because those functions would make the inner code accessible by only one thread at a time - right?

Right.  It protects everything between the lock/unlock calls in a mutex.

> The reason you say that calling plc_tag_read() from multiple threads might not be what the user wants, is because they need to also request the data with a plc_tag_get_uint8 (for example) - and in the intervening period it might have been updated by a new value - right?

Yep.  You really do not want your data changing out from underneath you.   This actually happens on the PLC side when you read a remote tag or get remote data in a consumed tag and it is a PITA to deal with.

> One thing that this exercise has shown me is that it probably makes sense for the wrapper to do the Status checks as part of the operations. It doesn't really make sense for them to ask the user to do this if they have to do it 100% of the time.

OK. I agree with that.   Note that plc_tag_create() does have a timeout on it, so it can be blocking or async.   There is a caveat with the call to plc_tag_create(): Only the first call is going to block for a long time.   Once you have a good connection to the PLC, subsequent calls to get tag handles to the same PLC will complete almost automatically even if they have timeouts.  Internally the library multiplexes the connection automatically.

> That means we could reduce the above code to be simply:

    var myTag = new Tag(IPAddress.Parse("192.168.0.10"), "1,0", CpuTypes.LGX, DataTypes.DINT, "MY_DINT");
    await myTag.ReadAsync();

That seems pretty nice for users!

> Maybe this API is less "thin" than it used to be, but it isn't introducing any new functionality - its just converting the core library into something idiomatic for .NET users.

Thanks for the examples and code.   Now I have better idea of what you are aiming for.   As you note, this does raise the question of just how far the wrapper should go.  Jody, I think, proposed the idea of having either two wrappers or two layers where the low level was a plain one-for-one wrapper around the C library and the next layer would be language/system-specific.   I think that the async stuff falls in the latter bucket.  That said, this is a pretty core piece of how one would expect a .Net library to work.   I am torn on this :-(

I like where you are going with the API.   How does the away/async stuff work in this case?   There is nothing happening until you call `ReadAsync()`.   At that point you have triggered the library in the background.   I do not know .Net at all, so how does the "await" work?   Does that cause the current thread to block until the result is ready?  Can you get a future (or the equivalent) from `ReadAsync()` that you can pass around and then `await` it later?

(random musing...) I wonder how straightforward it would be to make the tag read and write automatically?   I have a background "tickler" call running against very tag in the system frequently.   It would not be hard to add something that would kick off a read every X milliseconds.   Reading in data is "safe" in the sense that you are not going to get a block of RAM swapped out from underneath you since the data access APIs all lock the tag's internal mutex.   You could get data that is partially new and partially old, but that happens on the PLC as well.   I could set a dirty flag whenever on of the plc_tag_set_intX() functions was called...  Hmm, this needs more thought, it could _really_ simplify code if used correctly.  I think I see how to do this in very little code.  Hmm...

Back to today!

In the example code, the definition of `ReadAsync` would be in the wrapper and not something that a user would write, correct?   How would this handle a failure of the read?  Where does the status get propagated?   Perhaps I am not understanding enough of the proposal to grasp how that would work here :-(

I cannot comment on the technical feasibility of this proposal, so hopefully Jody can chime in.

jkoplo, 2020-07-04:
I would love to have async functionality for this library, but I also know
that async is not the norm for more 'old school' developers. We probably
have to expose both - maybe we have a second libplctag.net.async library
with extension methods? TBH I thought async might be something we tackle as
part of a future release and we just leave it to devs to task-out calls
they want to make async.

Also, the _cleanest_ implementation of async would be to use the low level
.net network calls that support async and use libplctag to encode/decode
the messages - but that would be a huge technical undertaking. I think this
is sort-of how the AWS IoT C++ library works - you have to feed it a
network connection from outside the library. I could be wrong though -
haven't played with it a ton.

In terms of truly explaining how async works in .net -> it's COMPLICATED.
I'm not sure I could do it justice. The Joe Duffy blog post is probably one
of the most interesting academic backgrounds on it:
http://joeduffyblog.com/2015/11/19/asynchronous-everything/

Jody

timyhac, 2020-07-04:
Ooft that could be tricky.

Kyle - I think a Task is similar to a Future in C++ or a Promise in javascript, so you can get the task and await it later. Example:
```csharp
var readTask = myTag.ReadAsync();
...
await readTask;
```

If something failed then we can throw an Exception - the [`ThrowIfError()`](https://github.com/libplctag/libplctag.NET/blob/d73fa3d92d440803f1bdf1e72deb76862a5f1314/src/libplctag/Tag.cs#L83) checks the status for all errors and raises an appropriate Exception.

---

Is there any situation where this wouldn't make sense - it seems like you would do it after every API call?

kyle-github, 2020-07-05:
Thanks for the descriptions, timyhac, Jody.  I have some light reading for tonight now :-)

As to the cleanest implementation of async, I am not sure that just providing encoding is sufficient.  There is a lot more than that going on under the hood in the C library.  For instance, all of the handling of partial results that happens when your tag is bigger than a single packet.  It was that level of gratuitous complexity that led me down the path of the current API.

I think that @timyhac's idea using the callback is quite interesting.   More code, but it seems like it fits into the async system a little better because you are explicitly ceding control and getting it back via the callback.

@timyhac 's point about exceptions is an interesting one.   The C library doesn't have them because C does not have them.   But C# and Java and Python and ... pretty much everything but Go has exceptions.   I can see that exceptions might be a reasonable lower level thing.  At the same time, that does blur the lines between the levels.

I think Jody's point about how the users of the library are going to think is a good one.   Most of the code I have seen is extremely straight-line, blocking (or blocking-style) code.      At most the code used threads.   The code I have seen that used OPC tended to be full of straight-line "get this tag value" then "get that tag value" then "compare the two values" rather than "fire requests for both tags, wait for both to complete, then compare the results."

What would a standard API for .Net look like?     What are the elements that would be used?   What would the program flow be like?  Rather than build a higher level wrapper up from what the C library does, perhaps it makes sense to work downward?  I still like the idea of a very thin wrapper/shim around the C library, but that does not mean it cannot start aiming in the direction of the higher level.

timyhac, 2020-07-05:
The reason async/await is popular is because you can write synchronous-looking code, but it does not block the calling thread.
As a theoretical example, if somebody wrote a WinForms application and there was a button that got the value of a tag.

```csharp
void Button1_Click(object sender, EventArgs e){
    myTag.Read(0);
    while (myTag.GetStatus() == StatusCode.PLCTAG_STATUS_PENDING)
    {
        Thread.Sleep(100);
    }
    myTag.ThrowIfError();
    MessageBox.Show($"The value is {myTag.GetInt32()}")
}
```

Just say that the whole transaction took 3 seconds. That means that the entire application would be "Not Responding" until the MessageBox is shown.
You wouldn't be able to close the application, no other screen elements would update, etc, because the UI thread is blocked.

On the other hand this would not block the UI thread, they would still see other screen updates and could interact with the UI.

Putting logic like this inside the UI class is not good practice, but IT IS EVERYWHERE. And providing an async function would provide some partial relief for the actual app user.

```csharp
async void Button1_Click(object sender, EventArgs e){
    await myTag.ReadAsync();
    MessageBox.Show($"The value is {myTag.GetInt32()}")
}
```

To me this feels pretty natural.

kyle-github, 2020-07-08:
I ran across this while on a very boring Zoom call today:

[Async Cancellation in C#](https://devblogs.microsoft.com/oldnewthing/20200701-00/?p=103916)

Raymond is always a fun read.

timyhac, 2020-07-08:
@kyle-github - would you saw that async use is one of the intents of the C API? It feels like that was the intent behind providing an instant return for 0 timeout.

I forgot about Abort() in the earlier examples! So this is what a simple implementation would be:

```csharp
public async Task ReadAsync(CancellationToken cancellationToken = default)
{

    int sleepPeriod = 10;

    Read(TimeSpan.Zero);
    while (GetStatus() == StatusCode.StatusPending)
    {
        if (cancellationToken.IsCancellationRequested)
        {
            Abort();
            cancellationToken.ThrowIfCancellationRequested();
        }
        await Task.Delay(sleepPeriod);
    }
}
```

timyhac, 2020-07-10:
And this is the callback version which seems to work but I'm not sure is the right way to implement this. Playing this has made me realize that we need to surface the "raw" callback APIs into events on the `Tag` object which would make this function much easier to work with.

```csharp
public async Task ReadAsync(CancellationToken cancellationToken = default)
{

    TaskCompletionSource<object> readTask = new TaskCompletionSource<object>();

    Action<int, int, int> eventCallback = delegate (int tag_id, int event_id, int status)
    {

        Console.WriteLine($"{tag_id} {(Event)event_id} {(StatusCode)status}");

        // Ignore if the event was for a different tag
        if (tag_id != pointer)
            return;

        switch ((Event)event_id)
        {
            case Event.ReadCompleted:
                readTask.SetResult(default);
                // Finished, unregister callback
                plctag.unregister_callback(pointer);
                if (status < 0)
                    readTask.SetException(new LibPlcTagException((StatusCode)status));
                break;
            case Event.Aborted:
            case Event.Destroyed:
                readTask.SetCanceled();
                // Finished, unregister callback
                plctag.unregister_callback(pointer);
                if (status < 0)
                    readTask.SetException(new LibPlcTagException((StatusCode)status));
                break;
            case Event.ReadStarted:
            case Event.WriteStarted:
            case Event.WriteCompleted:
                // Irrelevant, do nothing
                break;
            default:
                throw new NotImplementedException();
        }

    };

    try
    {
        Task.Run(() =>
        {
            plctag.register_callback(pointer, new plctag.callback_func(eventCallback));
            Read(TimeSpan.Zero);
            Task.Delay(Timeout.InfiniteTimeSpan);
        }).Wait(cancellationToken);
    }
    catch (OperationCanceledException)
    {
        Abort();
        throw;
    }

    await readTask.Task;

}
```

timyhac, 2020-07-10:
This is a version that makes use of the events that have been exposed in [this branch](https://github.com/libplctag/libplctag.NET/blob/events/src/libplctag/Tag.cs#L143).
It's a little cleaner I suppose, but the main benefit is that it doesn't "hog" the callback - there can be only one callback registered to the C API, and this implementation just adds the ReadCompleted event to the EventHandler list as opposed to registering it to be the callback.

Seems to work OK

```csharp
public Task ReadAsync(CancellationToken cancellationToken)
{

    var awaitable = new TaskCompletionSource<object>();

    EventHandler<LibPlcTagEventArgs> setResult = null;
    EventHandler<LibPlcTagEventArgs> setCanceled = null;

    setResult = delegate (object sender, LibPlcTagEventArgs e)
    {
        removeEventHandlers();
        awaitable.SetResult(default);
    };

    setCanceled = delegate (object sender, LibPlcTagEventArgs e)
    {
        removeEventHandlers();
        awaitable.SetCanceled();
    };

    void removeEventHandlers()
    {
        ReadCompleted -= setResult;
        Aborted -= setCanceled;
        Destroyed -= setCanceled;
    }

    ReadCompleted += setResult;
    Aborted += setCanceled;
    Destroyed += setCanceled;

    using (cancellationToken.Register(() =>
    {
        removeEventHandlers();
        Abort();
        cancellationToken.ThrowIfCancellationRequested();
    }))
    {
        Read(TimeSpan.Zero);
        return awaitable.Task;
    }

}
```

jkoplo, 2020-07-11:
Small nitpick - if the cancellation token is cancelled I *think* that will fire-and-forget the Abort(). It might be cleaner to call the Abort and just depend on the callback/event to set the TaskCompletionSource.

timyhac, 2020-07-12:
I think you're right. This makes me think about the different places an Abort or Destroy could come from.

Currently the Abort and Destroy calls are publicly available on the Tag class, but the only place I feel they could be used is:

Abort - during a Read/Write/ReadAsync/WriteAsync/Dispose
Destroy - during a Dispose

So if we make them private then we could avoid needing to think about it too deeply, in that case it may however call for an optional cancellationToken on the synchronous Read/Write operations in addition to a timeout so that you can make use of the abort behaviour on a synchronous call if needed.

I note that if a timeout occurs then [abort is called in the library](https://github.com/libplctag/libplctag/blob/ca6ebb77d6a1cb635e982a69f3cb21f90f8da6b6/src/lib/lib.c#L1033) but as you mentioned this is a fire-and-forget call, and it would make more sense for this to resolve once the Abort operation is finished, which I assume happens [if we get an Aborted callback](https://github.com/libplctag/libplctag/blob/ca6ebb77d6a1cb635e982a69f3cb21f90f8da6b6/src/lib/lib.c#L875) - @kyle-github  are you able to confirm this?

I'll update the branch with two changes, the first just to only setCancelled on the Aborted callback as you've mentioned, but the second to show what it would look like without public Abort/Destroy methods and a cancellationToken on synchronous Read/Write

timyhac, 2020-07-12:
Done, see the [async branch](https://github.com/libplctag/libplctag.NET/tree/async).

I ended up doing three things:
* Forcing consumers who want async use to make use of the ReadAsync or WriteAsync by checking the timeout parameter >= 0
* Removing Abort from the public API.
* Removing the events from the public API (I couldn't shake the comment you made about the events being in the public API - by themselves they aren't super useful and don't even supply the Tag object (merely the pointer that we use internally) so they weren't ready.

I also set up an Example for asynchronous use, it has two methods. A simple example that just shows how to use Read/Write Async, the second measures the performance of sync vs async versions of the methods.
Something which I can't explain right now is that the Async version (when awaited) are consistently faster than the synchronous versions.
There has to be something wrong, right?

```
This method measures the speed of synchronous vs asynchronous reads
Running 100 Read() calls...     took 59ms per Read on average
Running 100 ReadAsync() calls...        took 46ms per Read on average
```

jkoplo, 2020-07-12:
I've only looked at the code above, but I think getting rid of low level async through C (`Read()` + `Check loop`) actually makes a lot of sense. It does make our wrapper less *thin* but I think we're just exposing async in a way that's better for the C# language. I don't think we're adding/changing functionality if that makes sense...

As for the performance, I'd run a larger sample size (10k reads?) and make sure to do it in `release` and not `debug`. I'm no threading expert and I don't know how everything gets handled in the C code, but I _could_ conceive of ways that async improves throughput by essentially paralleling/pipelining some work to multiple threads that otherwise is all handled synchronously on one thread. That's the magic (and dream) of async!!! Woooo! Or it could just be timing noise <shrug>.

timyhac, 2020-07-12:
Yeah I agree - I think its true to the intent of the libplctag library (otherwise Kyle wouldn't have made a separate code-path for timeout=0) but exposed in a way thats natural for .NET development.

I've just run it in Release mode with 1k reads and the results are:
```
This method measures the speed of synchronous vs asynchronous reads
Running 1000 Read() calls...    took 58ms per Read on average
Running 1000 ReadAsync() calls...       took 46ms per Read on average
```

I'll run it over 10k operations each.

timyhac, 2020-07-12:
While this is running I've been looking into this and found that there are two things that happen in between the READ_COMPLETED callback and returning.
* Something called [rc_dec](https://github.com/libplctag/libplctag/blob/ca6ebb77d6a1cb635e982a69f3cb21f90f8da6b6/src/lib/lib.c#L1058).
* Logging.

Not sure what this rc_dec function does but it seems to involve memory management: I suppose it could explain the extra 11-14ms?
https://github.com/libplctag/libplctag/blob/ca6ebb77d6a1cb635e982a69f3cb21f90f8da6b6/src/util/rc.c#L208

jkoplo, 2020-07-12:
Yeah between the logging and the rc_dec (ref count decrement?), it seems like async might be making a difference. Pretty cool stuff.

timyhac, 2020-07-12:
Got a NullReferenceException during the `ReadAsync()` testing... interesting
```
This method measures the speed of synchronous vs asynchronous reads
Running 10000 Read() calls...   took 57ms on average
Running 10000 ReadAsync() calls...Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
```

timyhac, 2020-07-12:
It was because some of the delegates were being garbage collected. I've updated the branch with a fix for that.

As part of this change, a feature was born; now any active ReadAsync calls will return as soon as one ReadCompleted callback is triggered. This is great because multi-threaded applications can take use of even faster returns - they don't have to wait for another read request to take place. The benefit will only come into effect if a ReadAsync call is made while another is still "in-flight". I suppose we could make this work for the synchonous call as well.

It is getting pretty complex - I'm thinking that maybe we should cut a beta v1 release before having this finished.

kyle-github, 2020-07-12:
Sorry to com in late.   I was out all day yesterday.   Local electronic drop off was finally available for us.  

Trying to catch up...  Sorry this is long.  I am trying to respond to the questions from yesterday all at once.

OK.  Let my try to explain what async is.   Fundamentally it is about control.   If you pass a timeout to the library (thanks for catching that negatives are not handled!), then the library has control and will sit and spin-wait in the current thread.   As far as the application code is concerned, the call to plc_tag_read() blocks.   If you pass a zero timeout to the library, then you are signalling that the application is in control and the library should just trigger actions.

All IO is completely decoupled from the "front end" of the API.   Once you kick off a read (I'll just use read as an example as write and create are similar), the request packet is queued up with a thread that runs the IO for a single PLC.   That thread sends the request to the PLC and checks for response packets.   One key thing the PLC IO thread does is pack requests into single packets to the PLC.  It also unpacks the requests.   As far as the front end is concerned, there is no such thing as packing.

Side note: there are actually three threads involved in any request.   

1. There is the application's thread that is calling plc_tag_read() and possibly plc_tag_status().
2. There is a "tickler" thread that cycles across all tags in the system and calls the routines that check for results packets.
3. There is the IO thread per PLC that looks at the request queue, bundles requests together and unbundles responses.

If you use blocking mode, then thread one sits in a wait loop where it checks the status and then waits.  The wait is more symbolic than anything else as a sleep causes the OS to reschedule to another thread.   So I just wait one millisecond.   Threads generally do not get the CPU back for any number of milliseconds, so the real minimum sleep is generally closer to 10-15ms on Windows.   It can be less on Linux.

If you use non-blocking mode, then the calls to plc_tag_status() calls the tickler function for the tag instead of thread 2 (actually as well as thread 2).   The reason that this is probably a bit faster is that in blocking mode you will use all three threads and thread 2 is the one that processes the request, so you still have to wait until the OS switches back to thread 1 before the response can be acted on.   If you are using async mode, then chances are that thread 1 was the one that called the tickler and immediately got the status that the request is done so you skip a context switch.

I have tried something simpler for Modbus where the PLC IO thread cycles across all associated tags and I do not need the tickler thread.   The Modbus code does no memory allocation during operation whereas the AB code does.   Eventually I want to rework the AB code to be like the Modbus code.  Baby steps toward making the library keep high performance but use very little memory.
 
Back to the discussion... 

There are two ways to take advantage of request packing.   Either you have a lot of threads, usually one per tag, and those all send requests to the PLC IO thread to pack simultaneously, or you use something async and then (at least in C code) you fire all your actions at once (i.e. fire a read across many tags with a timeout of zero) and then wait separately.   In my testing the async method is both slightly faster and uses a lot fewer resources (threads are expensive with the stack etc.).   Threading can be easier to write in application logic.   YMMV.

Now in C#, I have no idea what will happen.  If you have a blocking C call in an async method call, what happens to the thread?   If the thread gets blocked but it is an async executor thread (I think this is how async works in C#), then you will see the rest of the app continue to run fine.

Where you see large differences is in situations where you have more than one tag.   If you use blocking and a single thread, then you serialize all tag access and it is pretty slow.   I see 7-10ms per read.   If you have multiple threads or use async, you can read several tags in the same packet.  I have seen about 200 tags read requests get into a single packet.   The PLC takes a bit longer to respond but the response time averages less than 1ms per tag (sometimes much less).

The core problem with AB PLCs is that their networking is _SLOW_.   It takes many milliseconds for them to respond, notice a packet, do the request and return a response.   The newer L80 series has a gigabit port directly on the CPU module and it is noticeably faster.   Even so, a RaspberryPi can completely flood a PLC with requests without any trouble.  From the tests I have done, I think an Arduino can flood an AB PLC :-(

Without knowing how C# async is implemented, it is difficult for me to advise how the best performance might be obtained.

The rc_dec() is part of the reference counting system to prevent the destruction of the tag's resources while that code is executing.   There is a paired rc_inc() that happens in the lookup_tag() function.  That function takes a mutex guarding the hash table that maps the tag ID/handle to the pointer to the tag object and looks up the tag pointer.  Then lookup_tag() calls rc_inc() inside the mutex.  That function is atomic and if the ref count is already zero, it will return a null pointer.  That is why the first thing in most of tho [... cut ...] r!

I can't thank both of you enough for all the work you are putting into this.   This is the first time that I have had direct contact with people using the library in another wrapper.   My own use has been exclusively in straight C or Java and in Java we already had reason to use a highly threaded model, so we went with a very thin wrapper and use the blocking calls.   This is the first time I have been able to have any insight at all into how other languages would use it in an idiomatic way.

timyhac, 2020-07-13:
Thanks for this extended response Kyle! I've re-read it a couple of times and it will take a while for me to absorb it properly.

A few comments:
* It sounds like wrappers would benefit from using the timeout=0/async/non-blocking mode of plc_tag_read() under the hood, regardless of whether they are providing an async or synchronous API to the caller. This would help maximize request bundling across multiple tags.
* The question around Abort came from the need for a synchronous Abort when cancelling a Read. In the latest update to the branch I actually got rid of that, the only reason it would be useful is if you wanted to stop writing and immediately start reading, and you couldn't wait 10ms for the write-in-progress to finish.
* Something must be wrong with my CPU for me to be getting 60ms synchronous reads....
* Because our async task is an IO-bound task (as opposed to CPU-bound), I believe the TaskCompletionSource technique is the right one to convert the callback provided by the C library into an async non-blocking operation.

In terms of what the API should be, I do think there is room for both `Read()` (synchronous) and `ReadAsync()` versions of the call, as well as the Write equivalents.
The tag creation could also be done asynchronously, but there is no such thing as an async constructor in .NET
There are [factories](https://blog.stephencleary.com/2013/01/async-oop-2-constructors.html) however, that could implement this behaviour, or the `plc_tag_create()` call could be deferred to an `Init()` or `InitAsync()` call. I was thinking that to start off with at least we would just leave it as a blocking call in the constructor - that's what the wrapper does now.
The factory concept could fit in with Jody's idea of having a Controller object that is a factory for Tags.

I think we are getting closer to a great API, Jody's work on the generics should make it really natural for a C# developer to work with this. Async/Await is natural - but Jody made a point that many developers (particularly in our industry) would prefer synchronous methods. This is certainly true in my experience.
Other PLC drivers that I've seen lean heavily on events or background updates - it seems like this is what you're targeting with the transparent operation or auto read/write feature. I think this would be heavily used, but I think the core library should implement this behaviour rather than the wrapper writing its own implementation.

I think one thing missing at the moment is raising Exceptions. I've flipped back and forth on whether this should be in the library or not. Today, I think they should be raised by the wrapper, and I think I'm heading more in this direction again.

kyle-github, 2020-07-13:
A quick note on 60ms reads.   Are you creating a tag, reading it and then destroying it and measuring the total time?   If so, then the create part can easily take upwards of 40ms.  Even destruction can cause the internal PLC connection to be torn down and there is a handshake there too.    If you are taking a timestamp before calling read and then one when the read completes and you are seeing 60ms, then something must be off because you should be getting <10ms.

timyhac, 2020-07-13:
I've run equivalent code that directly invokes the C API (`plctag.read() => plc_tag_read()`) to make sure, its still coming back with ~60ms. hmmm....
I ran some tests last year when working with this and I do recall it being 10ms on average. Must be something wrong with my hardware...
```csharp
int repetitions = 100;
Console.Write($"Running {repetitions} Read() calls...");
var syncStopWatch = Stopwatch.StartNew();
for (int ii = 0; ii < repetitions; ii++)
{
    // We know that it takes less than 1000ms per read, so it will return as soon as it is finished rather than timing out
    plctag.read(tagHandle, 1000);
}
syncStopWatch.Stop();
Console.WriteLine($"\ttook {syncStopWatch.ElapsedMilliseconds / repetitions}ms on average");
```

It's great that you've been able to confirm that async operation is indeed faster than synchronous operation which was the original purpose of these tests.

kyle-github, 2020-07-14:
Weird.   This is what I get with the prerelease version:

```text
2020-06-13 18:04:26.499 thread(1) tag(0) INFO plc_tag_create:578 tag set up elapsed time 40ms
...
2020-06-13 18:04:26.510 thread(1) tag(11) INFO plc_tag_read:1061 elapsed time 11ms
```

So I am seeing 11ms just for the read part and ~50ms for the connection plus the read.   That's from my laptop on WiFi, across my lame router, to a wireline network to an old switch and my PLC.   

Can you read the same tag with tag_rw and use debug level 4 to see if tis taking the same time?

timyhac, 2020-07-14:
I'm using a cross-over cable straight from my laptop into a L32E CompactLogix
```
> ./tag_rw -t sint32 -p "protocol=ab_eip&gateway=192.168.0.10&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=Dummy&debug=4"

...
2389-06-14 11:24:27.492 thread(1) tag(0) INFO plc_tag_create:573 tag set up elapsed time 230ms
...
2389-06-14 11:24:27.585 thread(1) tag(11) INFO plc_tag_read:1050 elapsed time 77ms
...
```

I used the Windows x64 build of the 2.1.10 pre-release.
At least it is roughly the same directly using the C applications as with C# through P/Invoke.

kyle-github, 2020-07-14:
Yikes!  Those are terrible times!   Is your cable bad?

jkoplo, 2020-07-14:
I ran what I could of the example code in the `async` branch - for some reason the async loop was hanging for me. I didn't investigate too much.
```
This method measures the speed of synchronous vs asynchronous reads
Running 1000 Read() calls...    took 14ms per Read on average
```
That's a USB3 Gigabit adapter directly connected to a CompactLogix L16ER.

One thing I did notice - the `Program.cs` in the DotNetCore example was not marked as `async Task` for Main. It also wasn't awaiting/waiting the `async void` examples. I _think_ that maybe some of the examples were running concurrently?

I'll commit back my tweaks.

kyle-github, 2020-07-14:
One way to check whether your async code is really behaving like it should is to set up multiple tags.   E.g. ten.  Then run it and if you see that you have a large multiple of the single tag read time, you might not actually be executing in parallel.   I found some incorrect mutex usage in my core code that way.

I have another example/test program, stress_test.c, that really pounds on the library with multiple threads (not async, but still in parallel).   I use it for performance testing and for burn in tests as I can pass a run time to it.   I usually run it for a few days.

I suggest creating a similar program but using async to really hammer on the code and make sure that everything is working correctly.

timyhac, 2020-07-15:
I'm not sure if this is the most scientific way of doing this (using Task.Run schedules a delegate to be run on a threadpool , so they aren't independent threads, probably only a handful of actual threads - actually I did check this when running on 200 tags, and it was 15 worker threads, 4 real core CPU).
```csharp
Console.Write($"Running {repetitions} ReadAsync() calls on {myTags.Count()} tags simultaneously...");
var asyncStopWatch = Stopwatch.StartNew();

Task.WaitAll(myTags.Select(tag =>
{
    return Task.Run(async () =>
    {
        for (int ii = 0; ii < repetitions; ii++)
        {
            Debug.WriteLine($"{ii} {tag.Name}");
            await tag.ReadAsync();
        }
        Debug.WriteLine(tag.GetInt32(0));
    });
}).ToArray());

asyncStopWatch.Stop();
Console.WriteLine($"\ttook {(float)asyncStopWatch.ElapsedMilliseconds / (float)repetitions}ms on average");
```

The tags its writing to is are all on a single, 1000-element, DINT array but are independent tag objects (i.e. the tag name is MY_DINT_ARRAY[23]), and the test checks status == OK before continuing.
```csharp
var myTags = Enumerable.Range(0, 200).Select(i => new Tag(IPAddress.Parse("192.168.0.10"), "1,0", CpuType.Logix, DataType.DINT, $"MY_DINT_ARRAY_1000[{i}]", 5000));
```

Are these comparisons roughly what you would expect (disregarding my slow hardware)?
```
Running 100 ReadAsync() calls on 1 tags simultaneously...       took 47.03ms on average
Running 100 ReadAsync() calls on 2 tags simultaneously...       took 59.03ms on average
Running 100 ReadAsync() calls on 3 tags simultaneously...       took 64.81ms on average
Running 100 ReadAsync() calls on 4 tags simultaneously...       took 64.75ms on average
Running 100 ReadAsync() calls on 5 tags simultaneously...       took 63.95ms on average
Running 100 ReadAsync() calls on 6 tags simultaneously...       took 61.32ms on average
Running 100 ReadAsync() calls on 7 tags simultaneously...       took 60.79ms on average
Running 100 ReadAsync() calls on 8 tags simultaneously...       took 58.37ms on average
Running 100 ReadAsync() calls on 9 tags simultaneously...       took 58.17ms on average
Running 100 ReadAsync() calls on 10 tags simultaneously...      took 56.49ms on average
Running 100 ReadAsync() calls on 11 tags simultaneously...      took 58.41ms on average
Running 100 ReadAsync() calls on 12 tags simultaneously...      took 61.01ms on average
Running 100 ReadAsync() calls on 13 tags simultaneously...      took 58.96ms on average
Running 100 ReadAsync() calls on 14 tags simultaneously...      took 60.14ms on average
Running 100 ReadAsync() calls on 15 tags simultaneously...      took 59.5ms on average
Running 100 ReadAsync() calls on 16 tags simultaneously...      took 60.97ms on average
Running 100 ReadAsync() calls on 17 tags simultaneously...      took 62.01ms on average
Running 100 ReadAsync() calls on 18 tags simultaneously...      took 63.64ms on average
Running 100 ReadAsync() calls on 19 tags simultaneously...      took 65.14ms on average
Running 100 ReadAsync() calls on 20 tags simultaneously...      took 67.75ms on average
Running 100 ReadAsync() calls on 25 tags simultaneously...      took 71.91ms on average
Running 100 ReadAsync() calls on 30 tags simultaneously...      took 87.32ms on average
Running 100 ReadAsync() calls on 35 tags simultaneously...      took 90.93ms on average
Running 100 ReadAsync() calls on 40 tags simultaneously...      took 114.69ms on average
Running 100 ReadAsync() calls on 45 tags simultaneously...      took 135.4ms on average
Running 100 ReadAsync() calls on 50 tags simultaneously...      took 120.44ms on average
Running 100 ReadAsync() calls on 100 tags simultaneously...     took 297.72ms on average
Running 100 ReadAsync() calls on 200 tags simultaneously...     took 469.84ms on average
Running 100 ReadAsync() calls on 500 tags simultaneously...     took 1139.16ms on average
Running 100 ReadAsync() calls on 1000 tags simultaneously...    took 2236.51ms on average
```

I'm going to run the same test now with more reptitions and more tags

kyle-github, 2020-07-15:
There is a knee in the curve around 25 tags.  If you have 500-byte packets then it sounds about right.  

I just did a similar run using the stress_test.c example.

```text
--- Test 1 updating 50 elements starting at index 0.
--- Test 1, Creating tag (1, 50) with string protocol=ab_eip&gateway=10.206.1.40&path=1,4&cpu=lgx&elem_size=4&elem_count=50&name=TestBigArray[0]&debug=4
*** Test 1, iteration 1 updated 50 elements in 38ms.
...
*** Test 1, iteration 231 updated 50 elements in 36ms.
*** Test 1, iteration 232 updated 50 elements in 42ms.
*** Test 1, iteration 233 updated 50 elements in 38ms.
*** Test 1, iteration 234 updated 50 elements in 44ms.
*** Test 1, iteration 235 updated 50 elements in 57ms.
*** Test 1 terminating after 236 iterations and an average of 40ms per iteration.
``` 

This is for one out of ten threads.  Each thread is hitting 50 DINTs out of a large array (1000 elements).  The threads run by reading and writing these elements over and over.   I ran the test for 10 seconds with 10 threads.   As you can see, on my bad network, I see roughly the same total time as you (though this is reading _and_ writing).   

Once you start doing large packets (the request packets appear to be using about 2000 bytes in my test) and you use multiplexed requests, the response time jumps a lot.  I think the time I see is around 15-20ms.  If I do a single tag read, the tag is small, and there is no multiplexing, then I see more like 7-10ms.

I am also dumping a lot of debug output.   

When I try with 30 threads (30 tags) each of 33 DINTs in size for 60 seconds, I get about 49-50ms per iteration.

kyle-github, 2020-07-15:
I tried with my async version instead of with threads.  This is not apples-to-apples as this only reads a single DINT and does not write it.

```test
Hit ^C to terminate the test.
Read 30 tags in 17ms
Read 30 tags in 11ms
Read 30 tags in 12ms
Read 30 tags in 11ms
Read 30 tags in 16ms
Read 30 tags in 12ms
Read 30 tags in 13ms
Read 30 tags in 11ms
Read 30 tags in 10ms
Read 30 tags in 13ms
Read 30 tags in 10ms
Read 30 tags in 12ms
Read 30 tags in 11ms
Read 30 tags in 13ms
Read 30 tags in 12ms
Read 30 tags in 11ms
Read 30 tags in 11ms
Read 30 tags in 13ms
Read 30 tags in 11ms
Read 30 tags in 12ms
```

This has removed the PLC connection set up time from the equation. 

What is the full tag string?   Are you setting use_connected_msg to any value?

timyhac, 2020-07-16:
The full tag string for one tag is
`protocol=ab_eip&gateway=192.168.0.10&path=1,0&cpu=logix&elem_size=4&elem_count=1&name=MY_DINT_ARRAY_1000[0]&use_connected_msg=1`

I just tried this now with use_connected_msg=0 and it takes about 10-15x longer.

I also tried it without use_connected_msg and it has the same result as use_connected_msg=1.
Is there a difference between use_connected_msg=1 and not supplying this attribute?

kyle-github, 2020-07-16:
No difference.   It defaults to 1 if it is not provided (at least for PLCs that support multi-request packets).

timyhac, 2020-07-16:
Was playing around with this some more and have set up an example called ParallelRead(), which sets up a whole bunch of ReadAsyncs on the same tag and then awaits them all at once. The other tests where (asynchronously) awaiting AsyncReads one at a time.

Where I'm thinking we could go with this is that we could potentially have two callers making calls in close succession (say 5ms apart) to async-mode `plc_tag_read()`, and rather than sending two full read requests, we could just make a single call and when that read completes, return to both awaiters at the same time. The first caller waits the full amount of time, but the second caller gets a 5ms shortcut because it can take advantage of the fact that the read request was already made.

Looking further into this, the async call to `plc_tag_read()` returns a `PLCTAG_ERR_BUSY` status if a read is already in progress, but we can ignore this and just return to all the awaiters when we finally do get a ReadCompleted event.
This is how the code in the branch `async` works at the moment.

But what if it is busy for some other reason (e.g. a write was in progress) - will it still queue up a read? Or do we need to try again once the current operation finishes?
I'm reasonably sure that we have to try again.

The simple solution I suppose would be to let the caller deal with it; but I imagine in multi-threaded applications near-simultaneous Read requests would be fairly common - so throwing an Exception may not be the right way to do it. It does feel like the retry logic should be included in this library.

timyhac, 2020-07-16:
Yeah this is a rabbit hole haha. You're right Kyle, this needs to be an API first approach.

kyle-github, 2020-07-16:
Operations at the top level do not queue.  If the library says it is busy, then the operations is not started.   The app will need to try again.   

The thing is that having multiple threads doing something to a single tag is rarely a good idea.  If you were trying to implement something like the transparent tags, then I could see a very naive implementation using multiple threads to implement a read trigger (one thread) and a write trigger (a write thread).   That would be horribly wasteful and the coordination would not be fun.

timyhac, 2020-07-16:
Yes - it would end up being a race condition, whichever thread happens to be scheduled at the time would be the winner.

Alrighty, best we bubble this back to the caller and ask them to deal with it.

I imagine that consumers will implement this as part of some other PLC access layer (or they should) - so they would be able to implement retry logic / scheduling there.

timyhac, 2020-07-16:
Just making sure I'm not insane...

If we near-simultaneosly trigger an async read on two different tags, `the plc_tag_read()` call should not return `PLCTAG_ERR_BUSY` - right?

kyle-github, 2020-07-17:
Correct.  Different tags have different mutexes.

kyle-github, 2020-07-17:
The only time you should see BUSY is if you try to trigger an operation on a tag while another operations is already in progress.

timyhac, 2020-07-17:
Alrighty I'm abandoning the callback-based async implementation for now, and have done a simple polling system (using an asynchronous Delay so the method is async from the caller's perspective). This is in the simple-async branch.

The async branch has been updated with the latest work on callback-based async, but there are some unresolved issues that appear when using async + destroying/creating tags.
The issues don't appear if we only create the tags once, and it pretty reasonably robust.
I believe the issue has something to do with garbage collection and callback handles.

Maybe something to look at in the future.

Some points raised during this work
* We need to be raising exceptions from within these (and other) methods.
* There might some unsafe Platform Invoke use that needs to be reviewed for the NativeImport library.

timyhac, 2020-07-21:
Fixed in https://github.com/libplctag/libplctag.NET/commit/4c252e7292aaad77dd246bb614ef4213012f3970

closed 2020-07-21 by timyhac