libplctag/libplctag.NET #11: Contribute to libplctag organization?

jkoplo, 2020-05-28:
I'm working with Kyle (libplctag author) to try to create a github organization for [libplctag](https://github.com/libplctag) and the associated wrappers for different languages. I'm primarily a C# programmer and I was planning on starting from the Corsinvest wrapper to build up something a little more featureful that also included the newer features Kyle had added. However, I quickly realized the Corsinvest code is GPL3 and not particularly useful for me in a professional sense.

After some discussion, Kyle is planning on re-licensing libplctag under the MPL. This is a nice balance between some copyleft but still allowing for people to compile the code directly into other binaries.

Would you consider re-licensing your code as MPL and transferring it to the libplctag organization? I suppose it could also be kept as MIT if you feel strongly about it, but I'd prefer the MPL if I'm going to fork it. Also, would you be interested in helping maintain the C# wrapper within the organization going forward?

timyhac, 2020-06-01:
Hi there jkoplo

That sounds like a great plan!

In fact one of the primary reasons my library exists is because of the
incompatibility of the licenses.

The primary difficulty I was trying to overcome was the ease of including
libplctag into other .NET projects so that this great library can be used
far and wide.

Most of the time when developing with C#, you can simply download a Nuget
package and away you go. But with LGPL, you have to either provide a
mechanism to recompile your program with a different libplctag binary (if
using static linking like ilmerge), or provide a way to replace the dll -
which may or may not be simple.
I think its also important that commercial use is unimpeded - because I
believe that most of the use of this project will be of a commercial nature
- I doubt many people have PLCs floating around for hobby projects.

I don't understand the MPL very well, but I feel that the goals of the
project are probably more closely aligned to the MPL than the LGPL and I'm
pretty sure it would be much more easily cover these goals.

I'd be happy to transfer the project to the libplctag organization, as well
as the Nuget package and would love to work with you to make this a great
package.

timyhac

jkoplo, 2020-06-04:
That's great news. I should be upfront that I have not (as of yet) contributed to the library but have gone through all the existing C# libraries in hopes of creating an 'official' C# wrapper for libplctag. The author of [PlcTag.Net](https://github.com/SharpSeeEr/PlcTag.Net) has also been in touch and willing to work on a C# wrapper - he didn't initially realize the wrapper he forked is GPL3.
All discussion for libplctag happens at [libplctag Google Group](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!forum/libplctag). We'll get you added to the organization and you can transfer the repo in.

timyhac, 2020-06-05:
Sounds great - looking forward to it!

This is a great effort you've got going, I think fragmentation of a
community is often its death blow.

closed 2020-06-17 by timyhac