libplctag/libplctag.NET #12: Next steps for c# wrapper

timyhac, 2020-06-17:
We should determine what the next steps are for the c# wrapper now that it has been incorporated into the libplctag organization.

I feel a good place to start with is agreeing on what the goals should be. Maybe these could be similar goals to wrappers for other languages, maybe not. I have some goals written on the README for this project but definitely open to changing these goals.

One issue with the current goals is that we can simplify how this works because we no longer have to support the ability for end-users to supply their own plctag dll with statically linked projects.

timyhac, 2020-06-18:
@libplctag/c-sharp-wrapper - where is the best place to discuss this?

kyle-github, 2020-06-18:
Maybe a kick off on the forum/email list followed by discussion in an issue here?  Or set up an issue with the question and announce it in the forum?   I find that doing both gets a lot more eyeballs.

timyhac, 2020-06-19:
I created a topic here: https://groups.google.com/forum/#!topic/libplctag/6yWU9A2m0K4

closed 2020-07-03 by timyhac