libplctag/libplctag.NET #13: branch name and usage changes

kyle-github, 2020-06-25:
Since I just managed to put a broken version of 2.1.7 out on master, I will take this a kick in the pants to actually redo my branch use that I've been thinking about for a while.

Starting soon (probably this weekend), I will create some new branches in the core library repo:

release - the real main branch.   Nothing goes here except tested code that has "soaked" for a while as a pre-release unless it is an urgent bug fix or such a narrow change that the risk is small.  I will make this the default branch so that when people check out the code without specifying a branch, this is what they get.

prerelease - once I am done with my development and tests, I will merge to prerelease.  All prerelease drops will come from this branch.   When these are validated with more testing and a lack of user bugs, I will merge this over to release.

I am not sure what to do with the master branch.   I will try to keep it up to date with release for a while, but generally I would like to drop it.

How much heartburn will this give you guys?

kyle-github, 2020-06-25:
Note that I would appreciate feedback from you when there is a prerelease: does it break something in your wrapper/tooling/plans etc?   I want to know that so that I can fix it before a full release.

timyhac, 2020-06-25:
No issues from my perspective having a different name for the default
branch.

Do you run an automated test suite on libplctag or do we rely on user bugs
for regressions?

kyle-github, 2020-06-26:
I have a manual test suite that I go through and I have an automated one that is run on check-in via GitHub actions.  The automated one is fairly limited: it just tests a simple tag read of a Logix-style tag access.   I had to write a simulated PLC to do that since there is not a lot of PLC hardware attached to GitHub's machines :-)

My goal with the simulator was to take bugs, as they get reported, and turn those into test cases that are supported by the simulator.

timyhac, 2020-06-26:
Fantastic - that is similar to what I do with a printer interface library I
developed. It is network-protocol heavy, and I take real examples and add
them to a simulator-like unit test suite.

jkoplo, 2020-06-26:
I've had a thought I haven't fully explored about using an sqlite database
in a layer between a class and a PLC to 'record' actual transactions and
then use the database to 'play them back' as part of unit testing. Maybe
it's an idea worth exploring?

closed 2020-07-03 by timyhac