libplctag/libplctag.NET #17: Expose new libplctag APIs

timyhac, 2020-07-03:
New APIs are available in the core dll. Expose these functions to end users.

timyhac, 2020-07-03:
Started work on this - see https://github.com/libplctag/libplctag.NET/tree/new-API

kyle-github, 2020-07-03:
Thanks, Timy (is it Timy, timyhac, Timy Hac? what do you want to be
called?),

The callbacks are probably the hardest part now.   Is there something I can
do to make those better?   Everything else has very simple data types
(intentionally) such as int32 and C-style strings.

Best,
Kyle

timyhac, 2020-07-04:
timyhac probably (my real name is Tim and I've just kept this email address
for ages).

I've just pushed some new example code to the new-APIs branch which checks
the callbacks - see
https://github.com/libplctag/libplctag.NET/blob/new-API/src/ConsoleTest/NativeImportExample.cs
The event callback seems to be working, but I don't know how to check the
logging callback - how can I generate a log?

timyhac, 2020-07-04:
Figured it out, you have to set a debug level in the attribute string. Seems like it works.

kyle-github, 2020-07-04:
Sorry, I must have missed this yesterday :-(

You can set a debug level in the attribute string or you can call `plc_tag_set_debug_level()`.   Setting it to PLCTAG_DEBUG_INFO or PLCTAG_DEBUG_DETAIL will give you plenty of output.  Do not use PLCTAG_DEBUG_SPEW unless you want to test your IO subsystem :-)

Best,
Kyle

timyhac, 2020-07-05:
This is done as part of c61f68a064fff7f33e5e93147498fa80f73dc2af

closed 2020-07-05 by timyhac