libplctag/libplctag.NET #22: Expose Read Cache Ms

timyhac, 2020-07-04:
There is support for a read cache to speed up performance. Expose this in the Tag constructor.
https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes

kyle-github, 2020-07-04:
The motivation for that feature was not actually higher performance.  It was much more mundane.  I added it so that you can use dumb logic in your application but not overwhelm the PLC.   For example, if you need to read some tags every 100ms, some at 200ms and some at 500ms, you can put read_cache_ms settings on the tags for 100ms, 200ms and 500ms respectively, then call plc_tag_read() on all the tags every 50ms and the cache will ensure that the tags are only read from the PLC as often as you wanted.

timyhac, 2020-07-06:
This is fixed in da9cbfc74338782e37d2c3627019c6e1e9893ad2

closed 2020-07-06 by timyhac