libplctag/libplctag.NET #24: Make DebugLevel writable

timyhac, 2020-07-04:
Using plc_tag_set_int_attribute it is possible to set debugLevel after instantiation. Expose this via a property setter on the DebugLevel property
https://github.com/libplctag/libplctag/wiki/API#tag-internal-attribute-functions

kyle-github, 2020-07-04:
There is an API call to do this.   Both will work.   Actually, there are three ways to set the debug level.

1. in the attribute string.
2. using plc_tag_set_debug_level().
3. using plc_tag_set_int_attribute().

It looks like I need to update the API docs again.

Thanks for working through all this!

kyle-github, 2020-07-04:
I updated the API wiki page.  Sigh.  Not sure how I missed the entire section on debugging!

timyhac, 2020-07-05:
Thanks @kyle-github - one question: When using plc_tag_set_debug_level(), how does the core library know which tag we are setting the debug level on, we aren't passing a tag reference in?

I suppose the other question is around plc_tag_get_debug_level; I know the docs reference it but [I couldn't find it](https://github.com/libplctag/libplctag/search?q=plc_tag_get_debug_level&unscoped_q=plc_tag_get_debug_level) in the source.

Perhaps the last one is that the function signature `void plc_tag_set_debug_level(int debug_level)` isn't in the docs. No big deal for me but just raising it in case you think its important.

kyle-github, 2020-07-05:
Hmm, I thought I exposed it.   You can get it via the attribute too.  I look into this to see if I left it out.   Out all day today though.

Debug level is _library wide_, not per tag.   Sadly.

timyhac, 2020-07-05:
Thanks @kyle-github - I've done some checks, and have a few comments here:

* There is no function `plc_tag_get_debug_level()` in the dll (it causes a Dll Exception if trying to call it)
* You can succesfully set debug level using the attribute string in the create_tag method.
* In the documentation for [Tag Internal Attribute Functions](https://github.com/libplctag/libplctag/wiki/API#tag-internal-attribute-functions) it says there is a R/W attribute for "debug_level" - is intentional that this is different to the attribute used during tag creation "debug".
* Neither "debug" nor "debug_level" retrieve the debug level using `plc_tag_get_int_attribute`.
* Setting debug level using plc_tag_set_int_attribute using either "debug" or "debug_level" doesn't seem to work.

```csharp
var tagHandle = plctag.create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=MY_DINT&debug=3", 1000); // Debug level works because there is extra output
while (plctag.status(tagHandle) == 1)
{
    Thread.Sleep(100);
}

var debugLevel = plctag.get_int_attribute(tagHandle, "debug_level", int.MinValue);      // debugLevel = int.MinValue
var debugLevel1 = plctag.get_int_attribute(tagHandle, "debug", int.MinValue);           // debugLevel = int.MinValue
//var debugLevel2 = plctag.get_debug_level();                                           // Dll Exception 'Unable to find an entry point named 'plc_tag_get_debug_level' in DLL 'plctag'.'

var read_cache_ms = plctag.get_int_attribute(tagHandle, "read_cache_ms", int.MinValue); // read_cache_ms = 0
plctag.set_int_attribute(tagHandle, "read_cache_ms", 100);
var read_cache_ms1 = plctag.get_int_attribute(tagHandle, "read_cache_ms", int.MinValue); // read_cache_ms = 100
```

Since debug_level is library wide, I'm thinking that it could make sense in the wrapper to supply a static class that allows them to control this and remove it from the Tag class, something like this:
```csharp
public static class LibPlcTag
{
    public void SetDebugLevel(DebugLevels debugLevel);
    public DebugLevels GetDebugLevel();
}
```

timyhac, 2020-07-06:
Same as my comments on the other issue about version numbers, you need to use a tag pointer of 0 for this to work.

```csharp
var tagHandle = plctag.create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=MY_DINT&debug=3", 1000)
var debugLevel = plctag.get_int_attribute(0, "debug_level", int.MinValue);     // debugLevel = 3
plctag.set_int_attribute(0, "debug_level", 2);
var debugLevel1 = plctag.get_int_attribute(0, "debug_level", int.MinValue);           // debugLevel1 = 2
```

So the only comments I have now are:
* There is no function plc_tag_get_debug_level() in the dll (it causes a Dll Exception if trying to call it).
* In the documentation for Tag Internal Attribute Functions it says there is a R/W attribute for "debug_level" - is intentional that this is different to the attribute used during tag creation "debug".
* Does it make sense to access this with a static class for library-wide utilities as opposed to the Tag?

timyhac, 2020-07-07:
Done

closed 2020-07-07 by timyhac