libplctag/libplctag.NET #25: Expose version number getter

timyhac, 2020-07-04:
It is possible to retrieve the version number of the library by using the get_attribute function.
Expose this as a static function on the Tag object.
https://github.com/libplctag/libplctag/wiki/API#tag-internal-attribute-functions

This may be useful to libraries looking to build on top of this one.

kyle-github, 2020-07-04:
And for the low level wrapper to make sure it is running against a supported version of the library.   I use it in the example programs.   Mostly as an example, but there are some examples that depend on specific functionality being available.

timyhac, 2020-07-06:
Thanks @kyle-github - this is another one that would make sense to be library wide, right? At the moment we can only get the version number by supplying a tag handle.

Actually I haven't been able to get the version numbers using get_int_attribute at all. I know that the dll is version 2.1.9 because the check_lib_version seems to be working.

```csharp
var tagHandle = plctag.create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&cpu=LGX&elem_size=4&elem_count=1&name=MY_DINT", 1000);

while (plctag.status(tagHandle) == 1)
{
    Thread.Sleep(100);
}
var statusBeforeRead = plctag.status(tagHandle);
if (statusBeforeRead != 0)
{
    Console.WriteLine($"Something went wrong {statusBeforeRead}");
}

var majorVersion = plctag.get_int_attribute(tagHandle, "version_major", int.MinValue);  // majorVersion = int.MinValue instead of 2
var minorVersion = plctag.get_int_attribute(tagHandle, "version_minor", int.MinValue);  // minorVersion = int.MinValue instead of 1
var patchVersion = plctag.get_int_attribute(tagHandle, "version_patch", int.MinValue);  // patchVersion = int.MinValue instead of 9

var isAtLeast2_1_9 = plctag.check_lib_version(2, 1, 9);     // returns 0
var isAtLeast2_1_10 = plctag.check_lib_version(2, 1, 10);   // returns -35
```

timyhac, 2020-07-06:
Update - ah I understand now, the tagHandle needs to be 0 for this to work.
https://github.com/libplctag/libplctag/blob/ca6ebb77d6a1cb635e982a69f3cb21f90f8da6b6/src/lib/lib.c#L1236

```csharp
var majorVersion = plctag.get_int_attribute(0, "version_major", int.MinValue);  // majorVersion = 2
var minorVersion = plctag.get_int_attribute(0, "version_minor", int.MinValue);  // minorVersion = 1
var patchVersion = plctag.get_int_attribute(0, "version_patch", int.MinValue);  // patchVersion = 9

var isAtLeast2_1_9 = plctag.check_lib_version(2, 1, 9);     // returns 0
var isAtLeast2_1_10 = plctag.check_lib_version(2, 1, 10);   // returns -35
```

Given that these are library-wide values, I think it does make sense to expose them on a static class as opposed to on the tag object. What do you think?

timyhac, 2020-07-06:
See Pull Request https://github.com/libplctag/libplctag.NET/pull/32 for this implementation

timyhac, 2020-07-07:
Done

closed 2020-07-07 by timyhac