libplctag/libplctag.NET #26: Provide a VB.Net example

timyhac, 2020-07-04:
Many users will be using VB.Net in their projects. Provide a VB.Net example.

timyhac, 2020-07-05:
This is a VB.NET console application and works OK. Targets .NET Framework 4.7.2

```vb
Imports System.Net
Imports System.Threading
Imports libplctag

Module Module1

    Sub Main()

        Dim myTag = New Tag(IPAddress.Parse("10.10.10.10"), "1,0", CpuTypes.LGX, DataTypes.DINT, "PROGRAM:SomeProgram.SomeDINT")
        While (myTag.GetStatus() = StatusCode.PLCTAG_STATUS_PENDING)
            Thread.Sleep(100)
        End While
        myTag.ThrowIfError()

        myTag.SetInt32(0, 3737)
        myTag.Write(TimeSpan.Zero)
        While (myTag.GetStatus() = StatusCode.PLCTAG_STATUS_PENDING)
            Thread.Sleep(100)
        End While
        myTag.ThrowIfError()

        myTag.Read(TimeSpan.Zero)
        While (myTag.GetStatus() = StatusCode.PLCTAG_STATUS_PENDING)
            Thread.Sleep(100)
        End While
        myTag.ThrowIfError()

        Dim myDint = myTag.GetInt32(0)

        Console.WriteLine(myDint)

    End Sub

End Module

```

kyle-github, 2020-07-05:
Nice!   This is really going to help people get going!

timyhac, 2020-07-06:
@libplctag/c-sharp-wrapper should these examples go in a wiki on Github or in an example project in the repository?

kyle-github, 2020-07-07:
I would set it up as a project.  I made the C version into an example program so that it gets built with most builds and is also included in the source.

There is a guy asking about how to list tags in dotnet on the forum.

timyhac, 2020-07-07:
Yep sounds good, that will check them that they can be built - I've seen some projects where the examples don't compile :0

I'll work on this once we've merged in the other outstanding changes.

kyle-github, 2020-07-07:
I build all the examples.  At least on Linux.   I build most, well, maybe half of them on Windows.  Including the tag_rw tool which exercises most of the API.

kyle-github, 2020-07-07:
I should really make an effort to get all the examples compiling on Windows.   The big barrier is that I keep using POSIX/Linux functions in my examples and sometimes it is not obvious how to make that work on Windows.   It would be worth it though.

timyhac, 2020-07-09:
This is done, also the CI builds all the examples now. e326707

closed 2020-07-09 by timyhac