libplctag/libplctag.NET #27: Provide tag listing example

kyle-github, 2020-07-04:
This comes up every couple of weeks either in email directly to me or on the forum.   I think it would be really great if someone could make an example in C# or VB that listed tags.  I know that several users have used one of the C# wrappers and done this by looking at what the C code does.

jkoplo, 2020-07-04:
This is something I feel pretty passionate about and I started looking through the c code to see how it was handled. I can probably take this one.

timyhac, 2020-07-04:
That would be great Jody

kyle-github, 2020-07-05:
Thanks! Let me know if there is any way I can help.  I think I still have the docs around... somewhere.   I got most of it from looking at how pylogix does it.

timyhac, 2020-07-06:
@jkoplo - I started looking at this tonight mainly because I was interested in how it works. This is kind of a converted version of list_tags.c
Does anyone know if there are any other special tags in addition to "@tags"

```csharp
using System;
using System.Text;
using libplctag;
using libplctag.NativeImport;

namespace ExampleConsoleApp
{
    static class ListTags
    {

        static int TAG_STRING_SIZE = 200;
        static int TIMEOUT_MS = 5000;

        static int setup_tag(string ipAddress, string path, string program)
        {

            int tag = (int)StatusCode.PLCTAG_ERR_CREATE;
            string tag_string;

            if (string.IsNullOrEmpty(program))
            {
                tag_string = $"protocol=ab-eip&gateway={ipAddress}&path={path}&cpu=lgx&name=@tags";
            }
            else
            {
                tag_string = $"protocol=ab-eip&gateway={ipAddress}&path={path}&cpu=lgx&name={program}.@tags";
            }

            tag = plctag.create(tag_string, TIMEOUT_MS);
            if (tag < 0)
            {
                Console.WriteLine($"Unable to open tag!  Return code {plctag.decode_error(tag)}\n");
                return 0;
            }

            return tag;

        }

        static void get_list(int tag, string prefix/*, tag_entry_s[][] tag_list, program_entry_s[][] prog_list*/)
        {
            int rc = (int)StatusCode.PLCTAG_STATUS_OK;
            int offset = 0;

            rc = plctag.read(tag, TIMEOUT_MS);
            if (rc != (int)StatusCode.PLCTAG_STATUS_OK) {
                Console.WriteLine($"Unable to read tag!  Return code {plctag.decode_error(tag)}\n");
                return;
            }

            /* process each tag entry. */
            do
            {
                uint tag_instance_id = 0;
                ushort tag_type = 0;
                ushort element_length = 0;
                ushort[] array_dims = new ushort[3];
                string tag_name = "";

                /* each entry looks like this:
                uint32_t instance_id    monotonically increasing but not contiguous
                uint16_t symbol_type    type of the symbol.
                uint16_t element_length length of one array element in bytes.
                uint32_t array_dims[3]  array dimensions.
                uint16_t string_len     string length count.
                uint8_t string_data[]   string bytes (string_len of them)
                */

                tag_instance_id = plctag.get_uint32(tag, offset);
                offset += 4;

                tag_type = plctag.get_uint16(tag, offset);
                offset += 2;

                element_length = plctag.get_uint16(tag, offset);
                offset += 2;

                array_dims[0] = (ushort)plctag.get_uint32(tag, offset);
                offset += 4;
                array_dims[1] = (ushort)plctag.get_uint32(tag, offset);
                offset += 4;
                array_dims[2] = (ushort)plctag.get_uint32(tag, offset);
                offset += 4;

                var tag_name_len = plctag.get_uint16(tag, offset);
                offset += 2;

                int prefix_size;
                if (!string.IsNullOrEmpty(prefix))
                {
                    tag_name = $"{prefix}.";
                    prefix_size = prefix.Length + 1;
                }
                else
                {
                    prefix_size = 0;
                }

                byte[] tagNameBytes = new byte[tag_name_len];
                for (int ii = 0; (ii < tag_name_len) && (ii + prefix_size < TAG_STRING_SIZE * 2 - 1); ii++)
                {
                    var character = plctag.get_uint8(tag, offset);
                    tagNameBytes[ii] = Convert.ToByte(character);
                    offset++;
                }

                tag_name = Encoding.ASCII.GetString(tagNameBytes);

                Console.WriteLine($"{tag_type} \t {tag_name}");

            } while (rc == (int)StatusCode.PLCTAG_STATUS_OK && offset < plctag.get_size(tag));

        }

        public static void main()
        {
            var host = "192.168.0.10";
            var path = "1,0";
            var tag = setup_tag(host, path, null);

            //get_list(tag, null, &tags, &programs);
            get_list(tag, null);

        }

    }
}

```

timyhac, 2020-07-24:
Added in 8e382bc

closed 2020-07-24 by timyhac

kyle-github, 2020-07-25:
Nope.  Nothing else special.

Best,
Kyle

timyhac, 2020-07-25:
Hey @kyle-github - was that around the question whether there were other special tags besides `@tags`?

kyle-github, 2020-07-25:
Sorry, drive by response.   There are no other special tag names.   Yet :-)