libplctag/libplctag.NET #40: DataType Enum Design

timyhac, 2020-07-09:
The design of the DataType class is unusual. Its main purpose currently is to provide element size of various Tag types.

Another time where Tag type is important is during Tag listing, where the underlying integer values are different as they refer to CIP types vs element size. (See [this](https://github.com/libplctag/libplctag.NET/issues/28#issuecomment-655682550) comment)

Is it possible to reduce the DataType to a basic enum (as opposed to a class of static values), and have some mapping elsewhere that provides the appropriate element size?

Proposal would be something like this:

```csharp
public enum DataType
{
    SINT,
    INT,
    DINT,
    REAL,
    ....
}
```

The Tag constructor could take either a DataType enum or a data size (separate constructors).

Other concerns:
* There are many other base types missing (LREAL, UINT, etc).
* Should these use Pascal case (C# naming conventions) or just use upper case (AB naming convention).
* Should other members appear that duplicate AB types (as they do now) (Int8, Int32, etc) - we took the stance for CpuTypes that there should only be one alternative for each unique thing.

timyhac, 2020-07-09:
@jkoplo - I'd be keen to get your view on this.

jkoplo, 2020-07-09:
Yeah I found that enum a bit odd. I need to sit down and think though
datatypes + generics. Part of the issue is that the 'type' is actually
outside of the c# code and so to be strongly typed a tag either has to be
'known' before compile or dynamically handled.

I think there's a way to thread this needle though. It might just end up
causing runtime exceptions if you have a type mismatch though...

Jody

timyhac, 2020-07-09:
What about something like this:

The consumer provides the preferred C# they want to use, and they provide the method by which the tag is encoded/decoded.

```csharp
public class Tag<T, U> where T : IDataType<U>, new()
{

    T dataType = new T();

    Tag tag;

    public void Read(TimeSpan timeout)
    {
        tag.Read(timeout);
        Value = dataType.Decode(tag);
    }

    public void Write(TimeSpan timeout)
    {
        dataType.Encode(tag, Value);
        tag.Write(timeout);
    }

    public U Value { get; set; }
}
```

```csharp
public interface IDataType<T>
{
    int ElementSize { get; }
    int CipCode { get; }
    void Encode(Tag tag, T t);
    T Decode(Tag tag);
}
```

Our library could implement the DataType for the basic types but if they wanted to extend this (PID tags, other UDTs, etc) they could just implement IDataType.

This is probably better than the abstract classes I did last year for my own projects in my opinion. The main thing missing is array indexers.

jkoplo, 2020-07-10:
That's really nice and just about what I had in mind. I particularly like the interface. You kinda did the work already, but I'm hoping I'll have time to integrate this tonight.

timyhac, 2020-07-10:
Cool, nice - I haven't compiled/tried it but seems roughly right. Do you know if there is a way to make classes be arrays in a generic way:
i.e. a single class can have 0, 1, 2 or 3 dimensions?

kyle-github, 2020-07-10:
Nice!

Perhaps you could call something like your DataType.Decode(), but it would take two arguments: the tag and an offset.   The offset would be the byte offset to that element of the array in the tag's data.  After Read you would calculate the element offsets, and then call Decode() on one element at a time?

In C++ this is painful.  Easier to have a class that overloads the array operator and pretends to be an array than to use the normal notation for array declarations.  I think that there is some template magic to make this a bit more palatable.

Minor nit: in your example code, if the timeout is too short or zero, Read() will return before there is any data and perhaps throw an exception.

timyhac, 2020-07-10:
@kyle-github I think we're thinking along the same lines: maybe we should only support 0D and 1D access because that is what libplctag does. I know that AB have 2d and 3d arrays, but not sure about other systems.
If users want to inherit this class and provide 2d or 3d indexers and different constructors (with array sizes) they could do so. I don't know how this would work in C++but in C# it is fairly straightforward to provide 2D indexers.

So the base type would provide access via a `Value` property, and 1d access via a 1D indexer.

What do you think?

```csharp
public class Tag<T, U> where T : IDataType<U>, new()
{

    // .....

    public U Value { get; set; } // check that element_count == 1 and throw error if not
    public U this[int index] { get; set; } // Of course do checks on index bounds
}
```

Hmmmm... it doesn't feel right, maybe only the indexer? hmmmm...

timyhac, 2020-08-06:
This is long done

closed 2020-08-06 by timyhac