libplctag/libplctag.NET #42: Unreachable IP hangs indefinitely

jkoplo, 2020-07-09:
While testing the examples I ran one that was trying to hit a tag at a 192 IP address. Instead of any sort of error message it simply sat in the StatusPending loop. I tried again and waited quite a bit to see if it would timeout, but no luck.

The examples run with a keyword `default` value for timeout on the tag constructors. I'm guessing this default value is infinity. However, there wasn't any device at the IP we were attempting to connect to so I would expect this to fail immediately and not wait on a timeout.

jkoplo, 2020-07-09:
I just tested this with the `list_tags.exe` pointing at a non-existent IP. It seems to wait for the timeout duration (attempting to reconnect?) before throwing a fault. I think we just need a sensible default timeout - 30 secs?

kyle-github, 2020-07-09:
Was this in the C library examples?   If so, please let me know which one!   That should not happen.  I set the timeouts to discrete amounts every time, or I should!

timyhac, 2020-07-09:
I think default for TimeSpan is TimeSpan.Zero which would run forever. What do you think the default should be? The [list_tags application](https://github.com/libplctag/libplctag/blob/d4afe282c49574f5ffec5db345303a552acbe089/src/examples/list_tags.c) has a timeout value of 5000ms. Should that default be in the C api and it is accesed with a -1 timeout or something, or should the wrapper decide what the timeout value is?

timyhac, 2020-07-09:
Maybe at least in the examples we can set a timeout of 5seconds like the C examples do?

jkoplo, 2020-07-09:
I think maybe we just set the default parameter value for the timespan in the tag constructor to something reasonable like 5 sec. I was thrown mostly b/c I expected an unreachable address to just fault and not timeout.

kyle-github, 2020-07-09:
Five seconds was picked because it was short enough that someone would reasonably wait through it.   It was long enough that if it really did time out, there was a very strong chance that the PLC was not accessible or off.   It was not very scientific.   Three seconds would have been just about as good IMO.

timyhac, 2020-07-09:
İ think that the examples should definitely have a timeout like the C examples.
I'd be a little bit unsettled if the wrapper had a different opinion about the timeout than the underlying library.

Maybe we don't make it an optional parameter and force the user to consciously choose the timeout.

İ think this is my general opinion about what the wrapper should be; true to the intent of the underlying library without providing new opinions.

İf we do put a 5second or similar timeout, we should get all the wrappers to have the same value

jkoplo, 2020-07-10:
I think we force a timeout in the ctor. Are you married to `Timespan` instead of a simple `int` representing msec?

And I think 5 sec for the examples is probably a good idea.

timyhac, 2020-07-10:
Not married to timespan no, we should put the units in the parameter name (i.e. millisecondsDelay like [Task.Delay](https://docs.microsoft.com/en-us/dotnet/api/system.threading.tasks.task.delay?view=netcore-3.1#System_Threading_Tasks_Task_Delay_System_Int32_System_Threading_CancellationToken_))
Do we check for negative timeouts in the core library? Do we need to prevent this?

There are some precedents with default timeouts:
[HttpWebRequest](https://docs.microsoft.com/en-us/dotnet/api/system.net.httpwebrequest.timeout?view=netcore-3.1) has a 100second default timeout.
[SerialPort](https://docs.microsoft.com/en-us/dotnet/api/system.io.ports.serialport.readtimeout?view=dotnet-plat-ext-3.1) has a default infinite timeout.

Main reason I'm arguing this is that I think the wrappers should be consistent.

jkoplo, 2020-07-10:
I like `int` because it serializes easier and we don't have the need for large time ranges. Agree about units in the parameter name.
I'm good with no default timeout or whatever is default in the lower library.

kyle-github, 2020-07-10:
Defaults?   Uh...  It is C.  There are no defaults :-)

If you want defaults, perhaps it is better to add that as a parameter to the constructor and record that in the object, then have getters/setters to change it?   If I had been just a little smarter I would have had plc_tag_read(tag, timeout) and plc_tag_read_async(tag).   I overloaded the timeout value of 0 so you have to know what data you are passing to see the behavior of the program.   Not ideal.

Negative timeouts?   Let me check.   Huh.   Apparently that is something I never checked because, no, I do not check that they are negative.  Right now the result would be to immediately time out unless int is 64 bits.  I should fix that.   Bug #155 filed.   Thanks for making me check that!

closed 2020-07-12 by timyhac