libplctag/libplctag.NET #51: Expose Arrays as part of Generic Tag implementation

jkoplo, 2020-07-16:
We have the base types working in #50, now we need to implement the Arrays of base types. I've started work in branch `generic-typed-arrays`.

jkoplo, 2020-07-16:
One of the harder things here is that base types don't need any sort of value for length. Arrays need a length upon tag instantiation and we can't embed this into a type, so we either need to pass an instance of our ArrayType for encode/decode with length defined or we need to add length into GenericTag instantiation. Or both.

kyle-github, 2020-07-16:
Would it make sense to use the array version as the base and then create a special wrapper for non-array tags that used the array version under the hood?

jkoplo, 2020-07-16:
That's a really interesting idea. I was definitely thinking of it in the standard terms - an array is made up of base types. But flipping it (a base type is an array of one element) might clean up some of my troubles.

I'll keep running down my current direction a bit more but it might end up converging with that concept anyway.

timyhac, 2020-07-16:
Our tag constructors (both Tag and GenericTag) already takes an optional elementCount, defaulting to 1 (if this helps).

jkoplo, 2020-07-17:
Sleeping on it helped. I'm juggling a bunch of generics which I'm not hugely fond of, but I did get an implementation that compiles (and should work - but my PLC isn't here right now for test).

The class signature is a bit ugly. It's essentially a PlcType that is generically defined with another PlcType. All the DotNetTypes have to match or it throws compile time errors (good) but I can't figure out a way to not define the DotNetType (bad - feels like duplication).
```c#
public class PlcTypeArray<TPlcType, TDotNetType> : IPlcType<TDotNetType[]> where TPlcType : IPlcType<TDotNetType>, new()
```
In practice it looks like this:
```c#
var myArrayTag = new GenericTag<PlcTypeArray<PlcTypeDINT, int>, int[]>(IPAddress.Parse("10.10.10.10"), "1,0", CpuType.Logix, "TestArray", 1000, ARRAY_LENGTH);
```
And the actual code for PlcTypeArray isn't bad:
```c#
    public class PlcTypeArray<TPlcType, TDotNetType> : IPlcType<TDotNetType[]> where TPlcType : IPlcType<TDotNetType>, new()
    {
        private readonly TPlcType plcType;

        public PlcTypeArray()
        {
            //Instantiate our definition
            //TODO: These could be singleton or a private static lookup for performance

            plcType = new TPlcType();

        }

        //public TDotNetType this[int index] { get => plcType.Decode(tag, index); set => plcType.Encode(tag, value, index); }

        public int ElementSize { get => plcType.ElementSize; }

        //TODO: Not sure if this gets the base CIP code or a special 'Array' code
        public byte CipCode { get; } = 0;

        public TDotNetType[] Decode(Tag tag, int index = 0)
        {
            TDotNetType[] array = new TDotNetType[tag.ElementCount];
            for (int i = 0; i < array.Length; i++)
            {
                array[i] = plcType.Decode(tag, i);
            }

            return array;
        }

        public void Encode(Tag tag, TDotNetType[] t, int index = 0)
        {
            if (t.Length != tag.ElementCount) throw new ArgumentException($"{tag.Name} is an array of {tag.ElementCount}. Argument is length {t.Length}");

            for (int i = 0; i < tag.ElementCount; i++)
            {
                plcType.Encode(tag, t[i], i);
            }
        }
    }
```
If I'm being honest, the whole thing feels a little bit too _clever_ and like I'm creating layers of generics in some sort of C# version of Inception.

jkoplo, 2020-07-17:
@timyhac - What's your take on this? I realized that I *could* hide some complexity by making classes that hide the levels of generics. Something like:
```c#
PlcTypeDINTArray : PlcTypeArray<PlcTypeDINT, int>
``` 
Which would give us:
```c#
var myArrayTag = new GenericTag<PlcTypeDINTArray, int[]>(IPAddress.Parse("10.10.10.10"), "1,0", CpuType.Logix, "TestArray", 1000, ARRAY_LENGTH);
```

There'd be a lot of child classes and it would need better naming, but they're all one liners and could be in a single file.

jkoplo, 2020-07-18:
I can confirm the generic arrays work with the nested generic strategy. I also tested the `PlcTypeDINTArray` class which is basically just a readability fix and it works as well. I need to clean up this branch (implement for all the other types) and then PR it. I'll do `strings` in another branch and eventually deal with multidimensional arrays (though I _think_ those are a nice-to-have).

timyhac, 2020-07-18:
That's great!

I might have a play around and see.

timyhac, 2020-07-18:
Hey @jkoplo - I've put together my interpretation of how generics could be implemented. Its in the generics-timyhac branch.

```csharp
var MyDintArray = new Tag<DINTIntMarshaller, int>(IPAddress.Parse("192.168.0.10"), "1,0", CpuType.Logix, "MY_DINT_ARRAY_1000[0]", 1000, 1000);

for (int ii = 0; ii < 1000; ii++)
    MyDintArray[ii] = ii;

MyDintArray.Write(1000);

var MyStringArray = new Tag<STRINGStringMarshaller, string>(IPAddress.Parse("192.168.0.10"), "1,0", CpuType.Logix, "MY_STRING_ARRAY_10[0]", 1000, 10);

for (int ii = 0; ii < 10; ii++)
    MyStringArray[ii] = (ii*1111).ToString();

MyStringArray.Write(1000);
```

```csharp
public class DINTIntMarshaller : IMarshaller<int>
{
    public int ElementSize => 4;
    public int Decode(Tag tag, int index) => tag.GetInt32(ElementSize * index);
    public void Encode(Tag tag, int index, int value) => tag.SetInt32(ElementSize * index, value);
}
```

The only thing that is a little bit weird about it is that the consumer would need to use the indexer regardless of whether it was an array or not.

But I suppose, then again, if the PLC used 2d or 3d arrays then it would again be a little unnatural - maybe we could supply some "sugar" classes that marshal the 2d or 3d indexes into a 1d array. Same with non-arrays, it could provide a "Value" property instead of an indexer that just maps to array element 0.

jkoplo, 2020-07-18:
Yeah I thought about the custom indexer but I was a little uncomfortable with where it lead to. I really don't want to treat Arrays differently than base types. And when we get to UDTs I'd really like to just return a class that matches the UDT. If arrays require indexing out, this is going to get messy.

I also didn't want users to have to unpack the arrays out of the class.
Say I have `FailedSerialNumbersDINT[1000]` that represents a list of all failed serials for a lot on a machine. But the list gets initialized to all zeros on lot startup and the PLC just increments an index and adds to it throughout the run. I'd really like to do this:
```c#
var plcFailedSerials = new GenericTag<PlcTypeDINTArray, int[]>(IPAddress.Parse("10.10.10.10"), "1,0", CpuType.Logix, "FailedSerialNumbersDINT", 1000, 1000);
plcFailedSerials.Read(500);
var actualFailedSerials = plcFailedSerials.Value.Where(serial => serial != 0).ToList();
//Do more things
``` 
If we rely on accessing the values by indexer, then we either need to use a for loop to first unpack the tag into an `Array`, or we need to implement `IEnumerable` on the Tag class itself. I think going down the road of trying to make the tag class equivalent to an `Array` is going to be messy. It's easier (and more consistent with the base types) to just return an `Array` on `Tag.Value` and let the users do whatever they want with it.

jkoplo, 2020-07-18:
Oh, and don't judge my class/property/method naming too harshly. I tend to use a name that makes sense to me, and then go back through and rename things when the code is in place and I can see where things ended up.
Actually we should probably do that for the whole library API before we cut a 1.0 - we're gonna be stuck with the naming once it's released.

jkoplo, 2020-08-20:
Closing this. Single dimension arrays are implemented as part of the latest work on generics. #73

closed 2020-08-20 by jkoplo