libplctag/libplctag.NET #56: Throwing exceptions

timyhac, 2020-07-19:
C# APIs generally throw errors rather than returning status codes. This ticket requests that all operations are checked for errors and then throw the appropriate Exception.

timyhac, 2020-07-19:
Hi @kyle-github - I'm looking into this now and was just hoping for some clarification of how errors are returned for some data getters.

`int plc_tag_get_bit(int32_t tag, int offset_bit);`
`double plc_tag_get_float64(int32_t tag, int offset);`
`float plc_tag_get_float32(int32_t tag, int offset);`

I'm currently assuming that, because these functions return signed values, there would be an error if they (respectively) returned:
`int.MinValue`
`double.MinValue`
`float.MinValue`

Is this correct?

kyle-github, 2020-07-19:
Mostly correct, signed int getters return INTx_MIN.  Unsigned return UINTx_MAX.  The bit getter returns standard PLCTAG_ERR_x codes, or 0 or 1.  The floating point getters all look like they have a bug.   When I refactored everything into the central library routines, it looks like I either copied it incorrectly or did not fix the errors.   

Here is what _should_ happen:

- signed integer getters should return INTx_MIN.
- unsigned integer getters should return UINTx_MAX.
- floating point getters should return FLTx_MIN.
- bit getters return 0 for false, 1 for true and one of the PLCTAG_ERR_x values for errors.

This is probably a mistake in the API.   I should have had another input that was a default to use if there was an error.  That way, the app could pick the value.  I use this with the attribute functions and it is a much better way to do things.  Perhaps in API 3.0, the getters will change.   That would break _all_ code though :-(

C does not provide a nice mechanism for returning multiple values and has no way to throw an exception (there are libraries that provide exception-like features but they are not standard).

I will add an ER to the library to document the return codes in the API docs more clearly.   I am frustrated that I never really noticed that not all the getters are returning useful values.  It looks like the floating point getters will return zero if there is an error.

Thanks very much for noticing this!

kyle-github, 2020-07-19:
This is fixed in the latest prerelease.  I updated the handling of the error conditions.

timyhac, 2020-07-19:
Thanks! - I've updated the branch to reflect the upcoming release as opposed to the 2.1.9 behaviour.

closed 2020-07-21 by timyhac