libplctag/libplctag.NET #64: Native runtime is not replaced on NativeImport package update

timyhac, 2020-07-26:
When a developer needs to make use of an update to the libplctag.NativeImport package that includes updated native runtimes, they will need to manually delete the existing plctag.dll/libplctag.so/libplctag.dylib.
New applications do not suffer from this issue because they will always extract the correct runtime.

NativeImport only replaces the runtime if it does not already exist on disk.
Furthermore, Visual Studio Build->Clean does not delete these files, it only deletes files it has created, so the developer must manually delete these files.

A separate, but related, issue is when application developers change their application's target platform. The target platform needs to match the native runtime architecture.

Some options
* Always extract the runtime to disk rather than checking if it already exists (however this will prevent prototyping with pre-release runtimes).
* Use nuget packaging with native runtimes (unclear if this will work across .NET Core *and* .NET Framework applications, and it will force the application developer to target a specific platform as opposed to using "Any CPU")
* Provide an option to the application developer to choose whether to extract a runtime or use the one on the file system, this could be done by lazily extracting the appropriate library, allowing the application to first select whether to extract from assembly or use the runtime on disk.

timyhac, 2020-07-26:
Branch https://github.com/libplctag/libplctag.NET/tree/lazy-native-extraction has an implementation for lazy runtime extraction.

This potentially introduces a small performance overhead for each method call - each method will need to check if it is the first use of the native library, if it is then it will run the ExtractLibrary method, if not then it will call into the native runtime.

It is literally the smallest possible overhead since it checks an in-memory boolean value (rather than going to disk or performing some other more expensive computations).

kyle-github, 2020-07-26:
Would it be possible to name the C library DLL with the SHA of the contents?   Then when you try to load a specific version, the SHA hash will only match the right DLL.  Or just name the DLL plctag_x.y.z.dll?   Hmm, but that only would load the exact version, not a possibly newer version.

timyhac, 2020-07-26:
I was thinking about IanRobo's situation where we provide a pre-release to do some tests, where it might have the same version number. I suppose its also possible someone wants to run their own (potentially modified) compiled version of libplctag which would have a different hash.

The other option is to ask the application developer to call some `ExtractNative()` function (or not) - I suppose we could do this in the "friendly" wrapper so most of the time people wouldn't need to think about it. The only thing going against that would be separation of concerns, where I've thought that the NativeImport package is the package that sorts out all of that. Hmmm...

Maybe some performance statistics could help guide this decision.

kyle-github, 2020-07-26:
For what it is worth, the library does support a convenience function that tells you whether the current version of the library is compatible with the one you want.   See `plc_tag_check_library_version()`.   I am not really seeing how that could help but maybe there is some way you could leverage it?

timyhac, 2020-07-26:
I've just created a benchmark which evalutes the performance of calls to the raw `plc_tag_status()` vs it being called after a check to ensure that the library has been initialized. It does this for 1000tags, and runs `plc_tag_status()`s once for each tag, and this is not done in parallel.

The difference is on the order of 2ms but not in the direction I expected.. It is possible that the CLR does some tricky optimization which is skewing the results. It could be the garbage collector getting involved differently.. I don't understand..

When running this on one tag, it still showed the same difference (divided by 1000), but I thought it was just benchmarking noise..

This is quite useful actually, I might push this benchmarking project to the repository separately..

```
// * Summary *

BenchmarkDotNet=v0.12.1, OS=Windows 10.0.19041.388 (2004/?/20H1)
Intel Core i7-6700HQ CPU 2.60GHz (Skylake), 1 CPU, 8 logical and 4 physical cores
.NET Core SDK=5.0.100-preview.7.20366.6
  [Host]     : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT  [AttachedDebugger]
  DefaultJob : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT

|       Method |     Mean |    Error |   StdDev | Gen 0 | Gen 1 | Gen 2 | Allocated |
|------------- |---------:|---------:|---------:|------:|------:|------:|----------:|
|          Raw | 14.53 ms | 0.285 ms | 0.427 ms |     - |     - |     - |       4 B |
| ExtractFirst | 12.22 ms | 0.241 ms | 0.322 ms |     - |     - |     - |         - |

```

kyle-github, 2020-07-26:
Can you show the code?    Calling status is probably about the worst API call with respect to overhead in the library.   It has to take a mutex, look up the tag handle in a hash table, then dereference a single pointer and read a single int.   The mutex and hash table are probably 99% of the overhead.

timyhac, 2020-07-26:
Yep sure.

In this branch, I've wrapped each [native cal](https://github.com/libplctag/libplctag.NET/blob/lazy-native-extraction/src/libplctag.NativeImport/plctag.cs#L350)l in an [another method](https://github.com/libplctag/libplctag.NET/blob/lazy-native-extraction/src/libplctag.NativeImport/plctag.cs#L106) which extracts the runtime if required.

[`ExtractLibraryIfRequired()`](https://github.com/libplctag/libplctag.NET/blob/lazy-native-extraction/src/libplctag.NativeImport/plctag.cs#L26) should be very fast all except the first method call.

The benchmarking project is [here](https://github.com/libplctag/libplctag.NET/blob/lazy-native-extraction/src/libplctag.NativeImport.Benchmarks/Program.cs)

I'll try this perhaps on a `get_int32()` call.

timyhac, 2020-07-26:
With a get_int32 call the ExtractFirst is slower by 100-200us over 1000 tags.

```
// * Summary *

BenchmarkDotNet=v0.12.1, OS=Windows 10.0.19041.388 (2004/?/20H1)
Intel Core i7-6700HQ CPU 2.60GHz (Skylake), 1 CPU, 8 logical and 4 physical cores
.NET Core SDK=5.0.100-preview.7.20366.6
  [Host]     : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT  [AttachedDebugger]
  DefaultJob : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT

|       Method |     Mean |    Error |   StdDev | Gen 0 | Gen 1 | Gen 2 | Allocated |
|------------- |---------:|---------:|---------:|------:|------:|------:|----------:|
|          Raw | 13.30 ms | 0.261 ms | 0.339 ms |     - |     - |     - |         - |
| ExtractFirst | 13.44 ms | 0.264 ms | 0.293 ms |     - |     - |     - |         - |
```

timyhac, 2020-07-26:
What do you think would be the best API call to compare?

kyle-github, 2020-07-26:
We would want a call that does not lock a mutex.   There are two problems with calls that lock:

1. Locking a mutex can be expensive if there is contention.
2. It serializes access.

I think either `plc_tag_decode_error()` or `plc_tag_check_lib_version()` will work.   Neither take a mutex.   Both rely on very simple code that should be fast in general internally.   That should mean that the overhead of the library loading check will be more obvious.

timyhac, 2020-07-26:
This is using `plc_tag_check_lib_version()` - I also tried it with decode_error but since it was returning a string there was a significant marshalling cost due to string return type, which further reduced the difference between the raw and ExtractFirst versions. There was also memory allocation so we were really just measuring the speed of garbage collection.

On my laptop the difference was between 1-2ns on consecutive runs. I personally think this is small enough that it doesn't matter; most API calls take several orders of magnitude more time and will swamp the time cost of ensuring the library is loaded.

I suppose we could expose the library extraction feature to the application developer and ask them to do it before calling into any methods.
We would save the 2ns and the internal code would be simpler, but the API is slightly more complex. Although it would fail pretty quickly and the developer would quickly learn. Hmmm, maybe thats the better option.

At least the results make sense now!

```
// * Summary *

BenchmarkDotNet=v0.12.1, OS=Windows 10.0.19041.388 (2004/?/20H1)
Intel Core i7-6700HQ CPU 2.60GHz (Skylake), 1 CPU, 8 logical and 4 physical cores
.NET Core SDK=5.0.100-preview.7.20366.6
  [Host]     : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT  [AttachedDebugger]
  DefaultJob : .NET Core 3.1.6 (CoreCLR 4.700.20.26901, CoreFX 4.700.20.31603), X64 RyuJIT

|       Method |     Mean |     Error |    StdDev | Gen 0 | Gen 1 | Gen 2 | Allocated |
|------------- |---------:|----------:|----------:|------:|------:|------:|----------:|
|          Raw | 6.989 ns | 0.1440 ns | 0.1347 ns |     - |     - |     - |         - |
| ExtractFirst | 8.279 ns | 0.1578 ns | 0.1476 ns |     - |     - |     - |         - |
```

kyle-github, 2020-07-26:
Given that reading a single data element is going to take a mutex, 2ns does not seem like a problem.  I am all in favor of low friction for the common case,  If the hit is 2ns to make it transparent to the end user, then it seems worth it.

timyhac, 2020-07-27:
Alrighty - I might push this into the NativeImport package.

This doesn't affect the public API, but we could potentially expose the Initialization step to eliminate the overhead entirely.

timyhac, 2020-07-27:
The way it works now is that it will always dump the native dll onto disk unless you set `plctag.ForceExtractLibrary = false`

closed 2020-07-27 by timyhac