libplctag/libplctag.NET #96: No timeout on async

jkoplo, 2020-10-07:
During some testing I realized that if I attempt InitializeAsync or ReadAsync with no network connection to the PLC, it hangs indefinitely (no timeout).

timyhac, 2020-10-09:
Really, shouldn't it time out after 100s?

P.S Even if this isn't the issue, perhaps we could reduce the default timeout, because i imagine that many of our users would have a LAN connection to their PLC. İ think 100s was chosen after the default timeout for HttpClient which is often used for internet communications

jkoplo, 2020-10-10:
Hmm, perhaps I wasn't being patient enough...

timyhac, 2020-10-10:
How do you feel about reducing the default timeout to, say, 10s? That would be super generous for all but the worst networks I imagine.

jkoplo, 2020-10-10:
10 seconds seems reasonable. We should confirm that the sync and async actually timeout when no PLC is present.

kyle-github, 2020-10-10:
For what it is worth, I have no default timeout when you use the async mode on read/write/create.   That was intentional at the time.

It might be worth revisiting that in the core library by adding a new attribute to set a default timeout period.

timyhac, 2020-10-11:
What was the reason Kyle?

İt would be better if it was part of the core library i think.

kyle-github, 2020-10-11:
The reason at the time was primarily that I had no way (and had not thought of it) to change any values in a tag at run time.   I also wanted a very clear demarkation between the library being opinionated (used in blocking mode) and unopinionated.   And until the last few years, the error handling of the library was not as consistent as it is now so sometimes the best thing to do was to close all tags and try again.

So that could be revisited now.   Adding a default timeout param would not be that hard to do but it would change the behavior of existing code when you add it.

closed 2020-10-21 by jkoplo

kyle-github, 2020-10-22:
Hmm, another issue with having a default timeout is that this becomes a race condition in the C code.   I do have a "tickler" that runs a function on every tag every couple of milliseconds.   This is used to handle all the back-end async behavior.   If I add a default timeout, then I will need to handle the cases:

1. There is a default timeout and the user also calls plc_tag_read()/write() with a valid timeout value.   One should win, probably the explicit call.
2. There is a default timeout and the user is also monitoring the time.   The user could call plc_tag_abort() when the timeout has not happened.   The abort will need to clean up anything in flight at that time.
3. There is a default timeout and there are auto-sync times.   If the timeout is longer than the read period (for instance), what happens?   

It gets a bit tricky.  Mostly I just need to decide what makes the most sense, but it is not as simple as I was thinking.