libplctag/libplctag.NET #98: Dispose Tag

romanka5150, 2020-10-14:
Hi.
I started to uses your wrapper. In that time libplctag.0.0.27-alpha13 were latest release so is still in use on my side
Today I have some doubts if understand your examples, or there is some bug.

On simple example bellow I see that tag after .Dispose() has still status .isInitialized = true.
Is there a way how to Dispose, Release Tag ? Am i doing this in correct way ? 

                Tag sStringOut_LEN = new Tag
                {
                    Gateway = "192.168.7.100",
                    Path = "1,0",
                    PlcType = PlcType.ControlLogix,
                    Protocol = Protocol.ab_eip,
                    Name = $"PC_DATA.String_In.LEN",
                    ElementSize = 4,
                    ElementCount = 1
                };

                // Tag.isInitialized = false
                sStringOut_LEN.Initialize(5000);

                // Tag.isInitialized = true
                sStringOut_LEN.Read(5000);

                var someData = sStringOut_LEN.GetInt32(offset: 0);

                // Tag.isInitialized = true
                sStringOut_LEN.Dispose();

                // Tag.isInitialized = true   . Why ??

timyhac, 2020-10-15:
This is a good pickup, according to once of my go-to sources:
> Once disposed, an object is beyond redemption. It cannot be reactivated, and calling its methods or properties (other than Dispose) throws an ObjectDisposedException.
> Albahari, Joseph; Albahari, Ben. C# 7.0 in a Nutshell (p. 514). O'Reilly Media. Kindle Edition. 

FYI Internally, IsInitialized is separate to IsDisposed - the tag handle creation does some communication with the PLC, and libplctag core library has the concept of an async constructor. Of course in C# there is no such thing as an async constructor, so we decided to separate the initialization, so it can be lazily initialized (or if you wanted to create several hundred tags at once, you could do them all at once, asynchronously, rather than one after the other).

timyhac, 2020-10-15:
Btw @romanka5150 - what you've done is fine. You don't need to call `Dispose()` yourself because the GarbageCollector will do this for you.

jkoplo, 2020-10-15:
@romanka5150 - it also looks like you're using the lower level (native) class for Tag. That's fine by all means, but there is also an implementation that handles types and offsets for you.

It looks like your setting up a length read in order to eventually retrieve a string. 

Here's an example of the higher level syntax, in this case writing an array of strings:

```c#
//Strings and String Arrays var stringTag = new Tag<StringPlcMapper, string[]>() { Name = "MY_STRING_1D[0]", Gateway = gateway, Path = path, Protocol = Protocol.ab_eip, PlcType = PlcType.ControlLogix, ArrayDimensions = new int[] { 100 }, }; var r = new Random((int)DateTime.Now.ToBinary()); for (int ii = 0; ii < stringTag.Value.Length; ii++) stringTag.Value[ii] = r.Next().ToString(); stringTag.Write();
```

More examples here: https://github.com/libplctag/libplctag.NET/tree/master/src/Examples/CSharp%20DotNetCore

jkoplo, 2020-10-21:
This should be closed by #99.
It'll make it into the next build.

closed 2020-10-21 by jkoplo