libplctag/libplctag.NET #101: StringPlcMapper does not null out full string

jkoplo, 2020-10-21:
One of my co-workers discovered this and it's a bit annoying.

The way we implemented the `StringPlcMapper` if you write a string like "12345" it will write a value of 5 to the `string.len` tag and then it will write 5 bytes into the actual byte array for the string.

The issue comes about if you then write a shorter string, "abc". The `string.len` will get updated to 3 and the first three bytes will be overwritten. *BUT* the value for the string array in the PLC will now be [a,b,c,4,5,\0,\0...] and this confuses some PLC functions and external programs.

[StringPlcMapper.cs](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/StringPlcMapper.cs)

I fixed this locally by creating a new, local mapper in my project that always writes the full max string length - padding it out with zeros. But it raises some questions:

1. Are there any performance implications to writing extra bytes for strings?
2. Should we do this for the different PLC types or only CLgx?
3. Should this be optional functionality or default?

@kyle-github and @timyhac - What do you guys think?

timyhac, 2020-10-21:
Interesting problem! I reckon it should pad with zeros as you've done it, and that is the standard behaviour - I'm not sure about other controller types but I imagine that zero padding would be the conservative way, so we could safely bet on that.

timyhac, 2020-10-21:
@ianrobo75 - do you have any advice on what the default behaviour should be for other controller types?

IanRobo75, 2020-10-21:
I believe a single byte padding of \0 should be sufficient in the cases I know of, especially HMIs/ SCADAs; If they are not reading .LEN then they are looking for the \0 terminator.
This would apply for all PLCs I believe.

kyle-github, 2020-10-22:
I have seen mangled strings in the PLC like that.   If you are using the fixed-width AB string types, then you can zero them out.   In the tag buffer, you will have all 82 bytes no matter how much string data you have.   It is safe to just start writing zeros until you get back a PLCTAG_ERR_OUT_OF_BOUNDS error.

That said, that is not particularly portable.   

And if you really have 82 bytes of string data, then there will not be a terminating zero.

I think in my string examples I may zero out the whole buffer first and then write in the new string.   I can't remember.   Whatever I did, it is probably wrong on some PLC.

I need to dredge up the string API ideas again and see what would happen in this case.   This is very PLC specific.

But first I need to stop working 12-16 hour days and finish up the Java and Android stuff.  All I need are six 28-hour days in the week and it would be great.

timyhac, 2020-10-30:
I think with this issue we should take the conservative approach and implement Jody's logic. In my opinion low performance is not as important as impaired functionality.

kyle-github, 2020-10-31:
You have to write the whole string every time anyway.   So there is no difference in the amount of IO.   Generally zero padding will be safer because someone somewhere will assume that all the strings in the world are C strings.  The IO overhead is so vastly larger than the overhead of writing out 60-70 extra bytes of zeros into an array in memory that is not an issue.

timyhac, 2020-10-31:
@kyle-github - fantastic!

@IanRobo75 - after looking at this a little bit closer, I'm 51% sure that we can simplify the [`LogixPcccEncode`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/StringPlcMapper.cs#L138) logic.

What its doing at the moment is padding a \0 character only when the string length is odd, to guarantee an even string length.

I would find it a little strange if _always_ padding the entire structure with \0s would break other cases.
What do you think about this approach?

kyle-github, 2020-10-31:
FWIW, I just wrote from the end of the valid characters to the end of the string in my string API.  I should dust that off and see if I can rebase on the current release branch.  It is a bit stale.

Something like (sorry, C) for PCCC PLCs:

```c
        /* set the string length */
        rc = plc_tag_set_uint16(tag, i * elem_size, (uint16_t)strlen(str));

        /* copy the string bytes into the tag. */
        for(j=0; (j < str_size) && (j < STR_BUF_MAX); j++) {
            int char_index = (i*elem_size) /* each string has a full buffer. */
                           + 2             /* skip the string length INT */
                           + (j ^ 0x01);   /* byteswap the index. */
            rc = plc_tag_set_uint8(tag, char_index, str[j]);
            if(rc != PLCTAG_STATUS_OK) { ... }
        }

        /* zero out the remaining bytes. */
        for(j = str_size; j < STR_BUF_MAX; j++) {
            int char_index = (i*elem_size) /* each string has a full buffer. */
                           + 2             /* skip the string length INT */
                           + (j ^ 0x01);   /* byteswap the index. */
            rc = plc_tag_set_uint8(tag, char_index, 0);
            if(rc != PLCTAG_STATUS_OK) { ... }
        }
```

The first loop copies the valid characters into the string buffer.   The next loop zeroes out the remaining bytes.  For CIP PLCs like ControlLogix, you need to use uint32 for the string length, 4 bytes for the offset and skip the byteswapping.

timyhac, 2020-10-31:
Oh right! the byte swapping..

jkoplo, 2020-10-31:
@timyhac - your code is basically the same as my fix.

IanRobo75, 2020-11-01:
I’m not sure what you are getting at.
I wrote \0 because the string “ian” would be stored in a LogixPCCC PLC as \61 \69 \00 \6E.  It makes sense to keep adding \00 from the next byte (5th byte here) until the end of the string (82nd byte), assuming the string is less than 82 bytes.  Seems pretty safe to me, for all the PLCs I know, and extra time should be negligible.

kyle-github, 2020-11-01:
@timyhac I was thinking about your earlier question about performance and thought of something that I missed.   Every call to `plc_tag_set_uint8()` takes a mutex (all API functions that read or set tag state do).   On Linux and Windows, that may not be too bad as mutexes are generally fairly well optimized.   I am not so sure about macOS.   But definitely on embedded platforms this could be a problem when mutexes are implemented in a correct way but not optimized.

So while calling the uint8 setter function works, it may not be ideal on some platforms.   Again, however, it is very unlikely that this will dominate the IO time.  When that is usually measured in milliseconds, you can run a lot of code without materially changing the response time of the result.

That said, perhaps I should add an API function for setting a range of bytes.  Then I can use `memset()` or the equivalent to go as fast as possible.   Something like:

```c
    int plc_tag_set_bytes(int32_t tag_id, int starting_offset, int length, uint8_t val);
```

You would use this to zero out a string like this:

```c
    int str_total_len = plc_tag_get_string_total_length(tag, str_offset);
    int rc = plc_tag_set_bytes(tag, str_offset, str_total_len, 0);

    /* ... code to set the string bytes and length ... */
```

Thoughts?   I can see it still being useful for some cases outside of string handling.

jkoplo, 2020-11-01:
That seems like an unnecessary optimization at this point. Since all the bytes are getting sent to the PLC, I think optimizing the local memory calls is not worth it.

If we really cared about the performance here we could optimize it by checking the new value against the old value and only padding the necessary elements if the new string is shorter. Even that is probably overkill though.

timyhac, 2020-11-02:
@kyle-github - do you have some performance metrics for the platforms where mutexes aren't performant?

Having raw buffer APIs could be useful where the data being transferred is something more like a binary file. We've exposed a [GetBuffer](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/Tag.cs#L428) function - I can't remember why but it must have been a use-case for something resembling raw buffer access.

kyle-github, 2020-11-02:
@timyhac sorry, no.  The comparison is always the other way when reading about how Linux mutex perform.  "Looks how great Linux mutexes are, see these numbers!"   And then no numbers for other platforms.

That said, there is a fairly large difference between a naive implementation of a mutex and one that tries very hard to be cheap when there is no contention.   Linux tries very, very hard in that direction.   That entails a lot of code which embedded systems usually leave out just due to code size considerations.

commit f084310a (2020-11-03): Merge pull request #107 from libplctag/null-out-string-pccc

Simpler logic for LogixPcccEncode + "Nulls out" the remainder of the string buffer.

Should completely resolve #101
closed 2020-11-03 by timyhac