libplctag/libplctag.NET #102: Extraction of native libraries fails for Windows Service

jkoplo, 2020-10-21:
I haven't fully isolated this one, but I built a ASP.NET Core Windows Service and found out the hard way that it won't extract the `plctag.dll` during service startup.

I'm assuming this might be do to the permissions the service is running as. For now I just have to include `plctag.dll` in my build as a file.

timyhac, 2020-10-21:
Wow good find. It must be a permissions thing.
I don't think its got anything to do with it, but there is an option to modify the extraction behaviour, whether to [ForceExtract](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag.NativeImport/plctag.cs#L13) or not (i.e. whether to overwrite the binary already there). Should probably put this on the README actually.

timyhac, 2020-10-30:
@jkoplo I was able to get this Worker Service working. Do you have any faultfinding ideas on the ASP.NET Core service?

Program.cs
```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading.Tasks;
using Microsoft.Extensions.DependencyInjection;
using Microsoft.Extensions.Hosting;
using Microsoft.Extensions.Logging;

namespace WorkerService1
{
    public class Program
    {
        public static void Main(string[] args)
        {
            CreateHostBuilder(args).Build().Run();
        }

        public static IHostBuilder CreateHostBuilder(string[] args) =>
            Host.CreateDefaultBuilder(args)
                .UseWindowsService()
                .ConfigureLogging(logging =>
                {
                    logging.AddFile(@"test.log");
                })
                .ConfigureServices((hostContext, services) =>
                {
                    services.AddHostedService<Worker>();
                });
    }
}
```

Worker.cs
```csharp
using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading;
using System.Threading.Tasks;
using Microsoft.Extensions.Hosting;
using Microsoft.Extensions.Logging;
using libplctag;

namespace WorkerService1
{
    public class Worker : BackgroundService
    {
        private readonly ILogger<Worker> _logger;

        public Worker(ILogger<Worker> logger)
        {
            _logger = logger;
        }

        protected override async Task ExecuteAsync(CancellationToken stoppingToken)
        {
            while (!stoppingToken.IsCancellationRequested)
            {
                var version = $"{LibPlcTag.VersionMajor}.{LibPlcTag.VersionMinor}.{LibPlcTag.VersionPatch}";
                _logger.LogInformation("Worker running at: {time} with version {version}", DateTimeOffset.Now, version);
                await Task.Delay(1000, stoppingToken);
            }
        }
    }
}
```

WorkerService1.csproj
```xml
<Project Sdk="Microsoft.NET.Sdk.Worker">

  <PropertyGroup>
    <TargetFramework>netcoreapp3.1</TargetFramework>
    <UserSecretsId>dotnet-WorkerService1-E9305C20-9018-4797-A576-93F0EF349A4D</UserSecretsId>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="libplctag" Version="1.0.0" />
    <PackageReference Include="Microsoft.Extensions.Hosting" Version="3.1.9" />
    <PackageReference Include="Microsoft.Extensions.Hosting.WindowsServices" Version="3.1.9" />
    <PackageReference Include="NReco.Logging.File" Version="1.1.0" />
  </ItemGroup>
</Project>
```

jkoplo, 2020-10-30:
I'd have to come up with a good test. It works fine in debug and as a standalone exe. It's only when it's installed as a service and has to extract the file that there's an issue.

Maybe a fix is to always extract to a temp directory? Or an appdata directory? I think it's the creation of a file in Program Files that is the underlying issue.

kyle-github, 2020-10-30:
FWIW, the JNA code (Java Native Access) copies the native binaries out of the JAR and into a temp directory on all platforms.  It took me an embarrassingly long time to figure out why /tmp had a copy of libplctag.so :-(

timyhac, 2020-10-30:
I should have been clearer: the above works for me when installed as a Service.

timyhac, 2020-10-30:
I'm OK if we extract to a temp directory, its probably better that the wrappers work the same way in this regard - @kyle-github is there a way for a user to substitute another native binary if they need to (i.e. run their own build or a pre-release)?

kyle-github, 2020-10-31:
With the Java wrapper, if the system path (Java class path plus system library path) contains a DLL that matches, then that one will be used.  If not, then the one in the JAR file will be used.   So I can test newer versions easily and still ship with one embedded in the JAR file.

JAR files are just ZIP files with delusions of grandeur and some gratuitous XML thrown in because Java and XML go together like chocolate and frog eggs.

timyhac, 2020-10-31:
By "contains a DLL that matches", you mean that it passes the `plc_tag_check_lib_version(...)` check? Yeah that could work.

kyle-github, 2020-10-31:
No, it isn't that good.   It just matches the CPU/OS.   It is probably possible to set up a comparison function in the code that does the search for the DLL.  I have not looked into it.

timyhac, 2020-10-31:
Okidoke - the current procedure came out of https://github.com/libplctag/libplctag.NET/issues/64.

What if we changed the behaviour slightly so that by default, it extracted the dll to tmp and used it, but the tinkerer had the option to supply a path to a dll which it used instead.
This would work in the situation where there was a new/different version of the core library that didn't have a new version number.

jkoplo, 2020-10-31:
What if we check the application directory for a dll first and if it doesn't exist we extract to temp? This gives us the ability to use any dll version we drop next to the exe for tinkering or really optimizing startup time, but in the default case it just extracts to temp on startup ensuring it uses the latest bundled dll when you upgrade the nuget package.

timyhac, 2020-10-31:
Ah yes that works too as far as I can see, sounds good.

kyle-github, 2020-10-31:
That's basically what the Java version does.   If I was smart, I would version the core DLL and then I could possibly do some version checking before trying to load it.   But I am not smart :-(

timyhac, 2020-11-10:
I have done a bit of research on how to do load from a folder in temp. Some ideas for further investigation:

* Manually load the dll into memory, and then hopefully pinvoke uses that instead of one that is on disk. https://www.pinvoke.net/default.aspx/kernel32.loadlibrary
* Add that folder to environment path. Pinvoke should check environment path for appropriate libraries.
* Use platform specific APIs to set the dll load path: e.g. https://docs.microsoft.com/en-au/windows/win32/api/winbase/nf-winbase-setdlldirectorya?redirectedfrom=MSDN

https://docs.microsoft.com/en-us/dotnet/core/dependency-loading/loading-unmanaged

timyhac, 2020-11-29:
Hi @jkoplo - have you had a chance to review any test cases that show this issue?

I haven't been able to figure out how to load plctag.dll from another directory :(

jkoplo, 2020-11-29:
Bummer. 

No I haven't come up with any other good test cases. Adding the extracted DLL as part of the install/build isn't really that bad. Maybe it's just a known workaround for people creating a Windows service? Not sure how big a population that is anyway...

timyhac, 2020-11-29:
@jkoplo - yes could be. I was able to get it working with a (.NET Core 3.1) Worker Service, I haven't tried with an ASP.NET Core Service - could be something specific there. Should we leave this ticket open do you think?

P.S. In my company we have one Windows Service; it proved to be a pain to debug and update so we figured that the service itself should just be an invoker of other (console) apps, because they're easier to debug, and we can more easily upgrade them. I'm not sure what your experience has been with them but that was the path we chose.

jkoplo, 2020-12-01:
Yeah let's close it. I haven't had the time to focus on much of anything lately, but maybe this will roll back around at some point.

closed 2020-12-01 by jkoplo