libplctag/libplctag.NET #108: Exception thrown when value in PLC is at the lower bound of the data type.

trinnan42, 2020-11-13:
I'm getting an exception when I set a value in the PLC to the lower bound of the data type.  For example, setting TestSINT.7 to 1 with all other bits set to 0 will result in this exception.  I have had this happen with SINTs and DINTs.  I haven't investigated if other data types have the same error.  All other reads and writes are working fine, it's only when I have _only_ the last bit set on a PLC tag.

`Exception of type 'libplctag.LibPlcTagException' was thrown.`

```
   at libplctag.Tag.GetInt8(Int32 offset)
   at libplctag.DataTypes.SintPlcMapper.Decode(Tag tag, Int32 offset)
   at libplctag.DataTypes.PlcMapperBase`1.DecodeArray(Tag tag)
   at libplctag.DataTypes.PlcMapperBase`1.libplctag.DataTypes.IPlcMapper<T[]>.Decode(Tag tag)
   at libplctag.Tag`2.DecodeAll()
   at libplctag.Tag`2.Initialize()
   at [MyMethod].ReadPLCData(LptLink link) in [MyMethod].cs:line 293
```

My ReadPLCData method just initializes the tag with a tagname passed in, reads it, and returns the data converted to an array of strings.

`{libplctag.Tag<libplctag.DataTypes.SintPlcMapper, sbyte[]>}`

Please let me know if you need any more information on how to recreate.

timyhac, 2020-11-14:
Trinnan - what is the tag name you're using? It seems like your using the SINT mapper, but are using it more like an array of bools. The tag type that the tag name points to should match the Mapper you use (i.e. "TestSINT", not "TestSINT.1").

trinnan42, 2020-11-14:
For this example, the tagname I'm attempting to read in the C# side would be "TestSINT".  When I refer to TestSINT.7 I'm referring to what I'm setting on the PLC side.  So I'm trying to read the whole SINT but in the PLC I'm setting only bit 7 (the 8th bit) to 1 for a value of -128.  This is when I get the exception.

timyhac, 2020-11-15:
Ah now I understand. You've hit a bug that I believe has been fixed in #100, but hasn't been released yet.
We were discussing whether to increase the minor version or release as a patch, I think it is backwards compatible so I will release now as a patch version.

trinnan42, 2020-11-16:
@timyhac That fixed it.  Thanks!

timyhac, 2020-11-16:
Awesome @trinnan42 - thanks!

closed 2020-11-16 by timyhac