libplctag/libplctag.NET #109: PLC not connected Timeout doesn't seem to use the Timeout property of the tag.

trinnan42, 2020-11-13:
When a PLC is disconnected, the Initialize method seems to take about 15 seconds to timeout.  It seemed to me in the documentation of the code that it would use whatever value is set to the Timeout property.  That Timeout property seems to work as expected for read/write timeouts if I unplug the PLC while my program is running, but it doesn't seem to for the first Initialize timeout if I start my program with the PLC disconnected.

timyhac, 2020-11-14:
@trinnan42 - both synchronous and asynchronous forms of Initialize are supported, I think you are using Initialize() not InitializeAsync(). For `InitializeAsync()`, the libplctag.NET wrapper handles timeout, for `Initialize()`, the underlying core library handles the timeout.

Could you by any chance reveal a bit of the code you've used for this?

trinnan42, 2020-11-16:
Yes, I was using Initialize(). I switched to InitializeAsync() and now it's doing what I expected it to do. That should resolve any issues I have with that.  Thanks.

timyhac, 2020-11-29:
@trinnan42 - thanks for reporting this. I believe the issue around `Initialize()` is in the core runtime, or with the way I (the wrapper author) expects the core runtime to work.

To check where the issue is, I've been using variants of the [NativeImportExample](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/NativeImportExample.cs#L9).

### Case 1: Cable plugged in and not taken out - All OK

```csharp
var tagHandle = plctag.plc_tag_create("protocol=ab_eip&gateway=192.168.0.10&path=1,0&plc=LGX&elem_size=4&elem_count=1&name=EncoderFactor&debug=4", 1000);

while (plctag.plc_tag_status(tagHandle) == 1)
    Thread.Sleep(100);

var statusBeforeRead = plctag.plc_tag_status(tagHandle);
if (statusBeforeRead != 0)
    Console.WriteLine($"Something went wrong {statusBeforeRead}");

plctag.plc_tag_read(tagHandle, 1000);
while (plctag.plc_tag_status(tagHandle) == 1)
    Thread.Sleep(100);

var statusAfterRead = plctag.plc_tag_status(tagHandle);
if (statusAfterRead != 0)
    Console.WriteLine($"Something went wrong {statusAfterRead}");

var theValue = plctag.plc_tag_get_uint32(tagHandle, 0);

plctag.plc_tag_destroy(tagHandle);
```

#### Output
```
2020-11-29 13:42:52.957 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2020-11-29 13:42:52.957 thread(2) tag(0) INFO tag_tickler_func:189 Starting.
2020-11-29 13:42:52.958 thread(1) tag(0) INFO ab_tag_create:165 Starting.
2020-11-29 13:42:52.958 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:172
2020-11-29 13:42:52.958 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-11-29 13:42:52.958 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 000002D9BA0F65E0
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL ab_tag_create:178 tag=000002D9BA0F65E0
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL get_plc_type:904 Found ControlLogix/CompactLogix PLC.
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL get_plc_type:904 Found ControlLogix/CompactLogix PLC.
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2020-11-29 13:42:52.959 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL session_create_unsafe:449 Session should use connected messaging.
2020-11-29 13:42:52.959 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2020-11-29 13:42:52.959 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-11-29 13:42:52.959 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 000002D9D3FA0590
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 1,0.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 1.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 2 in string 1,0.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 0.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router object.
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 6
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL cip_encode_path:456 Done
2020-11-29 13:42:52.960 thread(1) tag(0) DETAIL add_session_unsafe:310 Starting
2020-11-29 13:42:52.961 thread(1) tag(0) DETAIL add_session_unsafe:320 Done
2020-11-29 13:42:52.961 thread(1) tag(0) INFO session_create_unsafe:553 Done
2020-11-29 13:42:52.961 thread(1) tag(0) INFO session_init:569 Starting.
2020-11-29 13:42:52.963 thread(1) tag(0) DETAIL mutex_create:632 Starting.
2020-11-29 13:42:52.964 thread(1) tag(0) DETAIL mutex_create:659 Done.
2020-11-29 13:42:52.965 thread(1) tag(0) DETAIL thread_create:790 Starting.
2020-11-29 13:42:52.965 thread(1) tag(0) DETAIL thread_create:824 Done.
2020-11-29 13:42:52.966 thread(1) tag(0) INFO session_init:584 Done.
2020-11-29 13:42:52.967 thread(1) tag(0) DETAIL session_find_or_create:300 Done
2020-11-29 13:42:52.967 thread(3) tag(0) INFO session_handler:979 Starting thread for session 000002D9D3FA0590
2020-11-29 13:42:52.968 thread(1) tag(0) DETAIL ab_tag_create:266 using session=000002D9D3FA0590
[... 964 lines cut ...]
2020-11-29 13:53:20.215 thread(1) tag(0) DETAIL mutex_destroy:747 destroying mutex 0000022BE4E0E8E8
2020-11-29 13:53:20.216 thread(1) tag(0) DETAIL mutex_destroy:760 Done.
2020-11-29 13:53:20.216 thread(1) tag(0) DETAIL mutex_destroy:747 destroying mutex 0000022BE4E0E8F0
2020-11-29 13:53:20.217 thread(1) tag(0) DETAIL mutex_destroy:760 Done.
2020-11-29 13:53:20.217 thread(1) tag(0) INFO ab_tag_destroy:837 Finished releasing all tag resources.
2020-11-29 13:53:20.218 thread(1) tag(0) INFO ab_tag_destroy:839 done
2020-11-29 13:53:20.219 thread(1) tag(0) INFO refcount_cleanup:268 Done.
2020-11-29 13:53:20.220 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -32 not found.
2020-11-29 13:53:20.221 thread(1) tag(0) WARN plc_tag_status:1293 Tag not found.
[... cut ...]
2020-11-29 13:53:20.242 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -32 not found.
2020-11-29 13:53:20.243 thread(1) tag(0) WARN plc_tag_status:1293 Tag not found.
Something went wrong -19
2020-11-29 13:53:20.245 thread(1) tag(0) WARN lookup_tag:2853 Tag with ID -32 not found.
2020-11-29 13:53:20.245 thread(1) tag(0) WARN plc_tag_get_uint32:1994 Tag not found.
2020-11-29 13:53:20.246 thread(1) tag(0) INFO plc_tag_destroy:1067 Starting.
2020-11-29 13:53:20.246 thread(1) tag(0) WARN plc_tag_destroy:1070 Called with zero or invalid tag!
```

timyhac, 2020-11-29:
Hi @kyle-github - would appreciate if you could run your eye over the previous comment to check whether the behaviour shown is indeed unexpected. If you think it is, then I'll do some more investigation before raising a libplctag bug.

kyle-github, 2020-11-29:
Something seems off here.   

If you are using blocking mode, then you should be getting PLCTAG_ERR_TIMEOUT in scenario 2 and either PLCTAG_ERR_CREATE or PLCTAG_ERR_TIMEOUT in scenario 3.   

Let me do some investigation.   I think this is in the core library.

timyhac, 2020-11-30:
@kyle-github - yep OK thanks for checking. I'll do a bit morefaultfinding and get back with better details.

kyle-github, 2020-11-30:
Initial check shows I get the expected values on Linux.   I think.  I checked just scenario 3, which seems like the real outlier.   However, I just tested it by pointing to a PLC that does not exist.   It went right through and then shutdown with a timeout error.

I wonder if there is something different about that test than what you did?

Getting PLCTAG_ERR_TIMEOUT is somewhat annoying as in this case, the PLC does not exist.   But there is no way I can tell that from a PLC that is simply offline due to maintenance.   So the library just carries on being dumb and retries internally after five seconds.      Now that the library retries, about the only way you'll get a PLCTAG_ERR_CREATE is if the tag string itself is garbage.

Using the tag_rw2 (coming soon!) tool I can pass different timeouts and those are honored.

Perhaps this behavior differs somehow on Windows?

timyhac, 2021-01-27:
Just going to assume that this replicates #115 and is therefore fixed by #125

closed 2021-01-27 by timyhac