libplctag/libplctag.NET #110: try red O0:0.0  error code =-29

LittleWll, 2020-11-16:
try red   
"var tag4 = new Tag("192.168.0.100", CpuType.SLC, "O0:0.0", DataType.Int16, 5);"

return error code =29

timyhac, 2020-11-16:
Hi @LittleWll - this looks like you're either using a very old version or are using a different wrapper. Can you confirm you are using this nuget package: https://www.nuget.org/packages/libplctag/

timyhac, 2020-11-19:
@LittleWll the solution from libplctag.NET's perspective would be for you to update to v1.0.1

I daresay the error code relates to the core library error codes which you can find information around here: https://github.com/libplctag/libplctag/wiki/API#status-codes

closed 2020-11-19 by timyhac

LittleWll, 2020-11-22:
try codes  in  https://github.com/libplctag/libplctag
use the code   Examples---> CSharp DotNetFramework---ExampleRW.cs !
`Console.WriteLine($"\r\n*** ExampleRW ***");

            const int TIMEOUT = 5000;

            //DINT Test Read/Write
            var myTag = new Tag<IntPlcMapper, short>()
            {
                //Name of tag on the PLC, Controller-scoped would be just "SomeDINT"
                Name = "N7:0",

               
                //PLC IP Address
                Gateway = "192.168.1.110",

                //CIP path to PLC CPU. "1,0" will be used for most AB PLCs
                //Path = "1,0", 

                //Type of PLC
                PlcType = PlcType.Slc500,

                //Protocol
                Protocol = Protocol.ab_eip,

                //A global timeout value that is used for Initialize/Read/Write methods
                Timeout = TimeSpan.FromMilliseconds(TIMEOUT), 
            };
            myTag.Initialize();

            //Read tag value - This pulls the value from the PLC into the local Tag value
            Console.WriteLine($"Starting tag read");
            myTag.Read();`
connect the 1747-L552 cpu   variable initialization error
initialization string  {protocol=ab_eip&gateway=192.168.1.110&cpu=SLC&elem_size=2&elem_count=64&name=N7:0}