libplctag/libplctag.NET #113: Build script should tag releases

jkoplo, 2020-12-09:
We've had a couple requests for source code for a particular release. I think Nuget will always be the primary way the package is consumed, but it's probably worth adding releases/tags to our build script.

kyle-github, 2020-12-09:
For what it is worth, when I make a release on GitHub, it tags it for me.  Right now, making a release is a manual thing.  The CI does all the testing and building of the artifacts, but I have to pull the trigger.

Can Nuget handle source packages?  Maybe you can release a binary and a source package?

timyhac, 2021-01-06:
What should the tagging system be?

We have two different packages that are released which potentially have different version numbers, so just tagging them with v1.0.0 won't work. We could have something like:

libplctag-v1.0.2
libplctag.NativeImport-v1.0.1

I suppose an alternative could be to bump the version on both packages when either of them change, that will be tricky with the package references because when building libplctag it needs to have NativeImport already in nuget before it can build. This way we could just get away with the single version number.

jkoplo, 2021-01-09:
Good point about the separate packages, but I think your suggestion of including the package name in the tag is perfectly good.

timyhac, 2021-01-11:
Looks like it worked!

closed 2021-01-11 by timyhac