libplctag/libplctag.NET #117: Issue with reading from Tag-Based alarms in a ControlLogix PLC

darko-liberati, 2021-01-08:
I found an issue with the original libplctag code where it would not allow you to read from tag-based alarm tags. I notified the original developers/supporters of libplctag library and they have fixed the issue. I reached out to the c# wrapper library developer on github and they are no longer supporting their code and pointed me to here.

Can you include the new changes to the libplctag to this wrapper so that people can read tag-based alarms?

Thanks,

Here is the original thread from the original libplctag git-hub.

https://github.com/libplctag/libplctag/issues/222

timyhac, 2021-01-09:
Is anything required beyond packaging the new core libplctag runtime?

kyle-github, 2021-01-09:
Nope.  Just pick up 2.1.22.

timyhac, 2021-01-11:
Hey there @darko-liberati - please use latest packages which reference the 2.1.22 runtimes.

closed 2021-01-11 by timyhac