libplctag/libplctag.NET #120: Error on System.BadImageFormatException

ricky5151, 2021-01-12:
Hi, 

When I run .NET core project, the code is running fine. However, when I try to run on .NET framework project, it gave an error on this. I am running at "Any CPU" with an option "Prefer 32-bit" at ".NET Framework 4.7.2".

**System.BadImageFormatException: 'An attempt was made to load a program with an incorrect format. (Exception from HRESULT: 0x8007000B)'**

timyhac, 2021-01-12:
Hi @ricky5151 , a few questions if you don't mind:

* What is the actual platform you hit that issue on? x64 / x86? Mac/Windows/Linux?
* What are the range of target platforms you're targeting?
* Does it work with "prefer 32-bit" disabled?
* Does it work with .NET core with "prefer 32-bit" enabled?
* Are you supplying different libplctag binaries or using the ones supplied by libplctag.NativeImport?
* Are you using "ForceExtractLibrary = false"?

Do you need to run "Prefer 32-bit" with "Any CPU"? According to [this](https://stackoverflow.com/questions/12066638/what-is-the-purpose-of-the-prefer-32-bit-setting-in-visual-studio-and-how-does) the reason you would do that is to support both ARM (I guess x64?) and windows x86 - and the reason you target x86 is because you have a dependency in your application that requires it to be x86 when on windows.

timyhac, 2021-01-27:
HI @ricky5151 - I guess you have solved this problem since you haven't found any more issues.

closed 2021-01-27 by timyhac