libplctag/libplctag.NET #124: List All Tag in UDT

ricky5151, 2021-01-19:
Are we able to pull out all the tag info from UDT?

jkoplo, 2021-01-19:
It depends what you're asking about...

1. If we know the structure of the UDT, can we pull all the values, either individually or as one object? **Yes**
2. If we are given a PLC, can we discover the structure of a UDT on that PLC? **No**

The discovery of UDT (and other 'subtags') is something that I'd love to see @kyle-github support in the base libplctag library. It's basically a required feature for most of my professional work. But, it's complicated and made even more so by the fact that `libplctag` supports multiple PLCs, some of which don't even have the concept of UDTs.

ricky-ge, 2021-01-19:
Hi thank for the comment. The answer that I would like to know is from point 2. Hope that the new feature to get 'subtag' can be implemented in this library.

jkoplo, 2021-01-19:
Can you tell us a little more about your use-case? I'm assuming CLogix, right? And it sounds like you don't know the PLCs tag structure at C# compile time? Are you looking for a particular tag or a number of instances, or what? Even if you can auto-discover at runtime from the PLC, things can get really dynamic really quick depending what you are trying to do with the discovered tags... Whatever you can share would be helpful.

Another option - would it be feasible to use an L5X that's read in at runtime to enumerate tags?

ricky-ge, 2021-01-19:
I am actually doing real-time data collection by getting data from PLC and stored into database. For the ease of usage for this library, I had created an user interface for this work so that the user interface can be deploy in different machine. For my user interface, I able to retrieve a list of tag available in the PLC, which including the tagname, datatype, array size, etc. Unfortunately, the UDT type, I only able to get the tagname and its UDT name, but not the property name inside the UDT ('subtag'). The reason why I need to get 'subtag' is because there are some useful information that required to get the value from 'subtag'.

kyle-github, 2021-01-19:
I do plan to support this as soon as possible in the core library, but at this point, it is stacked up behind completion of the new event core.   I just got this to read a tag in a PLC-5, so while it is making progress, it is far from finished.  All the tag-metadata operations will be based on the new core as it drastically reduces the amount of code required.

Unfortunately, I cannot provide ETAs for this work.   My day job is keeping me very busy and while COVID means fewer outside activities, I do need to spend time with my family :-)  I had hoped to get the event core completed over the holiday break, but only got the happy path flow to work for PLC-5 last night.

That's a long way of saying that the UDT operations are a priority but I cannot give an ETA yet as a key project is in front of it.

ricky-ge, 2021-01-20:
No problem, take your time. Just part of my suggestion for this improvement. Will be very useful in future if the feature is available.

timyhac, 2021-01-20:
It looks like this one is outside the scope of the .NET wrapper project - is somebody OK to move this to https://github.com/libplctag/libplctag?

kyle-github, 2021-01-20:
The core library will support this similar to how it supports tag listing.   A lot will be left to the upper layers.  I would keep this open.   I may just do direct raw CIP support as that it going to be a lot easier with the new internal core.   I went back and forth on the right level before so I am not sure.   It would be nice to have *Logix tags automatically use symbol IDs instead of name segments.   Less overhead and faster responses.

timyhac, 2021-01-20:
Are there any issues in the core library that this can be linked against?

jkoplo, 2021-01-20:
@kyle-github - I imagine this could be a lot like the "@tags" in that it's really more of a data parsing problem than a protocol issue? And pretty specific to a PLC type?

kyle-github, 2021-01-22:
Very specific to at least a PLC make.   AB only supports tag listing of any kind in some PLCs.  I.e. not in Micro800 apparently. To list out "tags" in a PLC-5, SLC or MicroLogix, it appears that you need to read out a fair amount of the PLC's low level memory structures and decode them.   I have not looked too hard for examples for that.

I will be looking for examples of how to pull in multiple UDTs at a shot.   The docs do not give too many details.

timyhac, 2021-02-06:
For anyone who stumbles upon this thread, support will depend on some implemtation in the core libplctag library. Conversation is tracked here: https://github.com/libplctag/libplctag/issues/231

TrongViet94, 2021-05-19:
Can you update list all tag in UDT(@udt/...) for .NET? 
@kyle-github  @timyhac

timyhac, 2021-05-19:
@TrongViet94 Support will depend on some implemtation in the core libplctag library. Conversation is tracked here: libplctag/libplctag#231

timyhac, 2021-09-13:
Hi team - there is a pre-release version of libplctag which supports accessing of UDT definitions - Kyle has been discussing (see https://github.com/libplctag/libplctag/issues/231#issuecomment-914881260).

There is some support for this in a branch on libplctag.NET - see https://github.com/libplctag/libplctag.NET/blob/%23124-udt/src/Examples/CSharp%20DotNetCore/ListUdtDefinitions.cs

To run this you'll need to get the pre-release version of the core library and use `plctag.ForceExtractLibrary = false`.

The output will look something like this:
```
Id=4046 Name=ASCIISTRING82 NumFields=2
    Name=LEN Offset=0 Metadata=0 Type=196 Size=0
    Name=DATA Offset=4 Metadata=82 Type=8386 Size=0
Id=3970 Name=COUNTER NumFields=8
    Name=CTL Offset=0 Metadata=0 Type=211 Size=0
    Name=PRE Offset=4 Metadata=0 Type=196 Size=0
    Name=ACC Offset=8 Metadata=0 Type=196 Size=0
    Name=CU Offset=3 Metadata=7 Type=193 Size=0
    Name=CD Offset=3 Metadata=6 Type=193 Size=0
    Name=DN Offset=3 Metadata=5 Type=193 Size=0
    Name=OV Offset=3 Metadata=4 Type=193 Size=0
    Name=UN Offset=3 Metadata=3 Type=193 Size=0
Id=3971 Name=TIMER NumFields=6
    Name=CTL Offset=0 Metadata=0 Type=211 Size=0
    Name=PRE Offset=4 Metadata=0 Type=196 Size=0
    Name=ACC Offset=8 Metadata=0 Type=196 Size=0
    Name=EN Offset=3 Metadata=7 Type=193 Size=0
    Name=TT Offset=3 Metadata=6 Type=193 Size=0
    Name=DN Offset=3 Metadata=5 Type=193 Size=0
Id=3969 Name=CONTROL NumFields=11
    Name=CTL Offset=0 Metadata=0 Type=211 Size=0
    Name=LEN Offset=4 Metadata=0 Type=196 Size=0
    Name=POS Offset=8 Metadata=0 Type=196 Size=0
    Name=EN Offset=3 Metadata=7 Type=193 Size=0
    Name=EU Offset=3 Metadata=6 Type=193 Size=0
    Name=DN Offset=3 Metadata=5 Type=193 Size=0
    Name=EM Offset=3 Metadata=4 Type=193 Size=0
    Name=ER Offset=3 Metadata=3 Type=193 Size=0
    Name=UL Offset=3 Metadata=2 Type=193 Size=0
    Name=IN Offset=3 Metadata=1 Type=193 Size=0
    Name=FD Offset=3 Metadata=0 Type=193 Size=0
 .....
```

timyhac, 2021-09-22:
Hi all - the feature to determine the strucuture of UDT's for a controller has been released and can be used in libplctag.NET v1.0.10.

See https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ListPlc.cs for an example program

jkoplo, 2021-09-22:
NICE!

closed 2021-09-23 by timyhac