libplctag/libplctag.NET #129: Support libplctag2.2.0

jc835433715, 2021-01-26:
hi,timyhac, I used libplctag.NativeImport through NuGet, but it doesn’t seem to support the acquisition of strings in [libplctag2.2.0](https://github.com/libplctag/libplctag/releases/tag/v2.2.0), Can I update it soon, thank you

timyhac, 2021-01-27:
The work has been done, but it has not been uploaded to Nuget because it hasn't been tested.

I currently have no way of testing this - are you able to help?

To test, download this repository and modify one of the examples from [here](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/NativeImportExample.cs) to make use of the [string API](https://github.com/libplctag/libplctag/wiki/API#handling-strings).

timyhac, 2021-02-04:
Released as v1.0.3

closed 2021-02-04 by timyhac