libplctag/libplctag.NET #133: Package cant be downloaded/installed form nuget

uffebjorklund, 2021-02-04:
I was trying to install the libplctag package in a dotnet-script and failed with the message `The author primary signature's timestamp found a chain building issue: UntrustedRoot: self signed certificate in certificate chain`

At first I thought it was some limitation in dotnet-script, but I switched to a console application and still get the same error. 
I can install other packages from nuget, but not libplctag. My colleague has the same issue.

I have also tried earlier versions of the package without success.

dotnet --info
```
.NET SDK (reflecting any global.json):
 Version:   5.0.102
 Commit:    71365b4d42

Runtime Environment:
 OS Name:     ubuntu
 OS Version:  20.10
 OS Platform: Linux
 RID:         ubuntu.20.10-x64
 Base Path:   /usr/share/dotnet/sdk/5.0.102/

Host (useful for support):
  Version: 5.0.2
  Commit:  cb5f173b96

.NET SDKs installed:
  3.1.404 [/usr/share/dotnet/sdk]
  5.0.101 [/usr/share/dotnet/sdk]
  5.0.102 [/usr/share/dotnet/sdk]

.NET runtimes installed:
  Microsoft.AspNetCore.App 3.1.10 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.1 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 5.0.2 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.1.10 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.1 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 5.0.2 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

```

uffebjorklund, 2021-02-04:
This is an issue on Linux Ubuntu - Groovy Gorilla, closing

closed 2021-02-04 by uffebjorklund

timyhac, 2021-02-04:
Thanks for investigating @uffebjorklund

uffebjorklund, 2021-02-04:
If someone else have issues with nuget and certificates on Groovy Gorilla... 

## Two solutions...
 1. Use a custom nuget.config and add ALL owners to the <owners> section....
```
<?xml version="1.0" encoding="utf-8"?>
<configuration>
 <packageSources>
    <add key="NuGet.org" value="https://api.nuget.org/v3/index.json" protocolVersion="3" />
  </packageSources>
  <config>
    <add key="signatureValidationMode" value="accept" />
  </config>
  <trustedSigners>
    <author name="Microsoft">
      <certificate fingerprint="3F9001EA83C560D712C24CF213C3D312CB3BFF51EE89435D3430BD06B5D0EECE" hashAlgorithm="SHA256" allowUntrustedRoot="true" />
      <certificate fingerprint="AA12DA22A49BCE7D5C1AE64CC1F3D892F150DA76140F210ABD2CBFFCA2C18A27" hashAlgorithm="SHA256" allowUntrustedRoot="true" />
    </author>
    <repository name="NuGet.org" serviceIndex="https://api.nuget.org/v3/index.json">
      <certificate fingerprint="0E5F38F57DC1BCC806D8494F4F90FBCEDD988B46760709CBEEC6F4219AA6157D" hashAlgorithm="SHA256" allowUntrustedRoot="true" />
      <owners>Microsoft;dotnetframework;grpc-packages;serilog;domaindrivendev;protobuf-packages;dotnetfoundation</owners>
    </repository>
  </trustedSigners>
</configuration>
```
 2. The correct solution
```
apt remove ca-certificates
apt purge ca-certificates
apt update
apt install ca-certificates
```
Note that this removed some other stuff from my machine like kubectl for example.