libplctag/libplctag.NET #134: Create C# based test app

jkoplo, 2021-02-08:
Would it help with testing if we made a C# based app that people could grab and run? Maybe have it do a bunch of different permutations of calls and save the results to a file (CSV, json log, whatever) that could be emailed?

_Originally posted by @jkoplo in https://github.com/libplctag/libplctag/issues/231#issuecomment-774754390_

jkoplo, 2021-02-08:
A couple thoughts on why and what the use case would be:

The why:

1. Help @kyle-github so he doesn't have to spend as much time building/maintaining/etc user-focused test executables
2. Capture more data that can be used to help debug both the base library and libplctag.net - c# makes data logging simpler than in a pure c executable.
3. Dogfood our own library.

What (some scattered thoughts):

1. Do we need a UI or is command line fine? Do we want both?
2. Is this a separate repo or just another project in this repo? Do publish releases (probably)?
3. What's the first use case to target? Just duplicate functionality from the C test executables?

kyle-github, 2021-02-08:
@GitHubDragonFly made a really nice little Python front end.  See the [C DLL issue](https://github.com/libplctag/libplctag/issues/228).  Maybe something like that?   

Alternatively, I noticed that a lot of people are still using VB.Net and older Windows Forms.  Perhaps an app that lets you browse tags?

timyhac, 2021-02-08:
I think this is a good idea - providing a pre-built binary executable with a User Interface will maximize the number of people that can gather testing data. This should lead to gathering test data over a wide variety of platforms (Operating System, PLC Controller).

Some other benefits:
* Make the tests repeatable, as we will know exactly what code has been executed.
* The test data can be in a format of our choice (making it easy to compare between platforms).
* Reduce debugging and support effort. If people have problems, we can ask them to check if the pre-built binary works, if it does, then their code is wrong.

In terms of testing the [UDT feature](https://github.com/libplctag/libplctag/issues/231#issuecomment-774711892) and other internals of libplctag core, I think developing a C application would be the most comprehensive solution as it will run on all platforms. But I think that is beside the point - we want to cast as wide a net as possible to get as much feedback as possible.

* I think a User Interface would be best.
* Should be cross platform (I'm sure there are some frameworks we can use for this e.g. https://avaloniaui.net/).
* I think we should publish prebuilt binary executables.

jkoplo, 2021-02-08:
CLI is a User Interface right??? 😛

I've done some research into cross platform UI for .NET. This article is a pretty good summary:
[.NET 6 Desktop Dev Options: WPF, WinForms, UWP, .NET MAUI, Blazor ...](https://visualstudiomagazine.com/articles/2021/02/03/net-6-desktop.aspx?m=1)

Basically things are kind of a mess right now but should be better in the future. With that in mind I think we should:

1. Start with a CLI app -> Make something shippable first. Build a UI (or multiple) over it later.
2. Create a new repo -> It'll make cutting releases easier.
3. Decide on a small initial target for a test app -> @kyle-github, what would be helpful?

timyhac, 2021-02-09:
Haha yes I guess so, I suppose I assumed that there would be some interactivity in the interface as well. I understand that there are some CLI frameworks that help with that but I'm not familiar with them or how to write interactive CLIs.

There are some other (G)UI frameworks not listed in the article that seem OK. Not Microsoft supported but could be worth looking at:
* https://platform.uno/
* https://avaloniaui.net/

Out of all frameworks, I'm most familiar with WPF - but this won't ever be cross-platform. The two above seem somewhat close to WPF. Blazor is cross-platform but I imagine there would have hurdles setting it up as a Desktop application as opposed to a Server application. It doesn't seem fit for purpose.

If we do go down the CLI path, whats the benefit of doing this in C# instead of C? C would be more portable, and would more directly test libplctag core.

kyle-github, 2021-02-09:
The examples in C include a lot of one-off test code.   But they are generally not very flexible.   The tag_rw and tag_rw2 programs are probably among the few that are usable in a greater context.   The CI uses both the test programs and tag_rw directly for testing.

kyle-github, 2021-02-09:
Note that @GitHubDragonFly has started to make several test apps:

https://github.com/GitHubDragonFly/AB-Modbus-Master

That one is in VB.Net.   They also have one in Python that is coming along nicely.

jkoplo, 2021-02-13:
Sadly, they're using some modified version of the old Mesta wrapper, I think. Also, I'm allergic to VB. I break out in hives - I think it's due to repeated exposure to Excel Macros...

After reflecting on this, I have a couple thoughts.

- If we want a rich UI, maybe we just use WPF. The nature of the PLC industry is such that everyone pretty much uses Windows and WPF is a known entity.

- That being said, it seems like Uno Platform is the most promising choice if we want to build a cross platform UI. We aren't accountable to any business objective so maybe trying Uno would be fun?

- I think a server-type app might be another project (with shared core) and should have a web based UI. Then it could run on an RPi for longer term data logging.

@timyhac - thoughts?

timyhac, 2021-02-13:
I think that there is a lot of truth with the Windows being the primary platform. I don't have a wide set of experiences but I imagine that most of the PLC programming platforms only run on Windows. People doing PLC interfacing apps are probably also running the PLC programming platform in my view. Kyle might be an anomaly in this regard?

There would definitely be a fun element for me playing with Uno. I would use a Windows and an Android application. I also have access to iPhone and Macbook, so I could test there too.

kyle-github, 2021-02-14:
The primary platform is not clear.   At first, almost all users were Linux.   I think it has shifted towards Windows, but I also find signs of Mac usage (seems odd) now too.  There are a fair number of people who are doing embedded Linux (for some definition of embedded).  C# does run, somewhat, on Linux and macOS, so it is not out of the running.   Mesta's video and initial wrapper really kickstarted Windows use.

There are some old school cross platform toolkits like Tk, Qt, wxWindows, and a few others that I cannot remember. Dear Imgui?   Nuklear?   All the cool kids like immediate mode.

There are ways to package Python as a single executable.   I have not used them.  Same for Java if you want to go to Java 14.

timyhac, 2021-02-21:
After getting the ARM builds working, doing cross platform builds with Uno is quite simple, although I must say its not seamless.

To check this I put together a simple page which on button press, retrieves the value of a DINT and displays it in a dialog box.

The Gtk output looks like this:
[image] After adding the ARM builds to libplctag.NativeImport, this just worked.

The UWP output looks like this:
[image]
I was hitting #137 when trying to run the UWP project, so I copied the native dll in manually.

I couldn't get the WPF project to run properly.

I think the main pull of Uno is that you can allegedly make "native" experiences for each platform, rather than having the same UI for all platforms. i.e. the GTK widgets for linux apps, WinUI components for UWP apps. Pretty neat!

I haven't tried this yet with Android but it allegedly works for iOS and Android - I'm not sure which libplctag library would be the right one for iOS, and I believe that special builds are necessary for Android - I wonder if possible to add to CI...

jkoplo, 2021-06-16:
I've been doing more Blazor recently and there _seems_ like there might be upcoming desktop Blazor support (a la Electron-style apps). I might give this a go sometime soon. I'll report back if I get anything worth showing off.

timyhac, 2021-11-03:
With the tag and UDT listing now possible, this would be a really compelling project.

jkoplo, 2021-11-03:
Yeah, I've been designing it in my head for months! Just gotta find the time to put it into code.

jkoplo, 2022-02-16:
So this is started as [Crude Observatory](https://github.com/jkoplo/CrudeObservatory).

It started as a very 'plugin' type app, but once I got it working with InfluxDB I started wondering whether targets like CSV or JSON provided that much value. 

It doesn't have a UI, and therefore doesn't implement tag listing yet, but that's probably the next to do on the list. I'll do the UI in Blazor Hybrid so I can support the app as either a web/server install or as a cross platform desktop app.

Anyway, it's started and it's viable so I'm gonna close this.

closed 2022-02-16 by jkoplo