libplctag/libplctag.NET #142: UWP Support

timyhac, 2021-02-15:
I'm getting an [SEHException](https://docs.microsoft.com/en-us/dotnet/api/system.runtime.interopservices.sehexception?view=net-5.0) when calling `plc_tag_destroy(...)` on a tag in a UWP project.
Other API calls appear to work, it is only the plc_tag_destroy call that fails.

Using mixed debugging mode (both managed and native), I narrowed the error down to an invalid handle on [this line of code](https://github.com/libplctag/libplctag/blob/24f83dc79917f14b4d18253d9dc5a7182610ea85/src/platform/windows/platform.c#L1281). While destroying a tag, `socket_close` gets called twice, and it fails on the second call with `exception 0xC0000008: An invalid handle was specified.`
I'm essentially running [this](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/NativeImportExample.cs#L9).

It fails on both my PC (Windows x64) and on RPi (ARM 32bit) running Windows IOT Core.
It fails on both Debug and Release builds.
It does not fail when running as a .NET Core console application, nor a .NET Core WPF application, nor a .NET Framework console application.

kyle-github, 2021-02-15:
That line of code looks backward to me.   closesocket() should return 0 on success, so this is definition checking the wrong result.  I just fixed something like that in the new core, so maybe I just carried that over by cut-and-paste.

kyle-github, 2021-02-15:
I raised #240 on the C library for this.

timyhac, 2021-02-15:
Thanks @kyle-github - it is quite bizarre that other types of apps don't have this issue. Perhaps UWP apps use a different kind of debugger or at least different debugging preferences than win32 apps. I don't know or understand this.

I've put together a UWP project which invokes the problem (and also some which do not).
https://github.com/timyhac/socketclose_investigation

kyle-github, 2021-02-15:
Looks like this was a lurking bug for a long, long time.   Even on the POSIX/Linux side the same kind of thing could happen.  It is just not as obvious there.   The fundamental problem was an erroneous return value check.  The secondary problem was that there is an internal flag keeping track of whether the socket is open.   That flag is not cleared until after the close.   So if the close fails, the flag is never reset and thus a second attempt to close the socket will call `closesocket()` again rather than terminating early.

I am building a new release as I write this.  It is 2.3.2.   I just blindly bumped the version and see that I skipped 2.3.1.  Oh well.

timyhac, 2021-02-15:
Holy moly that was fast! - nice job, fixed with libplctag core release v2.3.2

I'll release this as part of libplctag.NativeImport later today (along with a fix for #141

closed 2021-02-15 by timyhac

timyhac, 2021-02-16:
Interestingly - UWP apps _are_ different to other applications.
https://blogs.windows.com/windowsdeveloper/2015/08/20/net-native-what-it-means-for-universal-windows-platform-uwp-developers/
https://docs.microsoft.com/en-us/dotnet/framework/net-native/getting-started-with-net-native
https://www.youtube.com/watch?v=RxPs2Pb70OQ

UWP apps compile to native code (like C++) when running a Release build.
This doesn't fully explain what is happening, because I have been debugging with a Debug build, and apparently these still build an Intermediate Language binary which runs on a .NET runtime - but it is strong enough evidence to me that UWP are fundamentally different, and to me this is a good enough answer as to why the behaviour was different under a UWP app vs a .NET core or framework win32 app.