libplctag/libplctag.NET #144: Test ARM/ARM64 builds

kyle-github, 2021-02-19:
Using the command line cross-arch build config that @timyhac has in his PR, I added the ARM builds to the main CI process.   I attached the results here since I have no way of testing them.   Can one of you please give these a look and see if they are usable?

[libplctag_2.3.3_windows_ARM64.zip](https://github.com/libplctag/libplctag.NET/files/6007619/libplctag_2.3.3_windows_ARM64.zip)
[libplctag_2.3.3_windows_ARM.zip](https://github.com/libplctag/libplctag.NET/files/6007620/libplctag_2.3.3_windows_ARM.zip)

If these work, they will be built every time I update the C DLL release.

timyhac, 2021-02-19:
I can confirm that the list_tags.exe works on Windows IoT Core Raspberry Pi 3B with ARM CPU (32bit), I don't have access to an ARM64 device.

I guess that confirms that both list_tags.exe and plctag.dll.

timyhac, 2021-02-19:
Running [`file` ](https://www.man7.org/linux/man-pages/man1/file.1.html)on these files produces the following:

#### ARM
`libplctag_2.3.3_windows_ARM/Release/plctag.dll: PE32 executable (DLL) (console) ARMv7 Thumb, for MS Windows`

#### ARM64
`libplctag_2.3.3_windows_ARM64/Release/plctag.dll: PE32+ executable (DLL) (console) Aarch64, for MS Windows`

I'm not sure what this proves besides the fact that file can identify it correctly.

I'd be reasonably confident that the 64bit version works.
If there is something wrong with it, I think it is likely that it just doesn't work at all - which is not that terrible in the scheme of things.

kyle-github, 2021-02-19:
Thanks!   I will work on adding these to the standard build on the `release` branch and getting the CI status showing in the main page.

timyhac, 2021-02-19:
Thanks @kyle-github - appreciate your work here.

I will also work on getting linux ARM working.

closed 2021-02-19 by timyhac

kyle-github, 2021-02-20:
I am going to try to get this working on GitHub actions via cross-compilation for Linux.   Seems like it should be possible.