libplctag/libplctag.NET #150: Trouble reading SINT Array

TheFern2, 2021-03-04:
This is more of question, I could be possibly doing something wrong. I am able to read each type of tag individually, I am able to read each type of Array, but for some reason SINT is not returning an array, like all the other types. I can read a SINT indivual tag, and also an individual array index, but not array. Looks like is returning hex values? I haven't worked with sbyte before, should I be casting to something else?

Thanks for the help!

```csharp
 private async Task<sbyte[]> ReadSintArray(string tagName, int arrayLength)
{
    var tag = new Tag<SintPlcMapper, sbyte[]>()
    {
        Name = tagName,
        Gateway = _ipAddress,
        Path = _path,
        PlcType = PlcType.ControlLogix,
        Protocol = Protocol.ab_eip,
        Timeout = TimeSpan.FromSeconds(_timeout),
        ArrayDimensions = new int[] { arrayLength }
    };

    await tag.InitializeAsync();
    await tag.ReadAsync();

    return tag.Value;
}
```

[image]

timyhac, 2021-03-04:
I think whatever you're using for stringification is seeing an array of signed bytes and converting them into a hexadecimal representation.  i.e. If you look at "BaseSINTArray[0]" I can see that you get a value of 9. The first byte in the SINT Array hex represetation is "09".

timyhac, 2021-03-04:
P.S. - @TheFern2 - would appreciate if you could use the v1.0.4-beta.1 during development.
There are a few bug fixes regarding async/await, but its also a fairly significant refactor.

I'm just waiting for feedback from testers before marking as a proper release.

TheFern2, 2021-03-04:
Thanks, I'll take a look a bit closer tomorrow morning. Not sure if I can switch to development, but if I can I'll definitely switch. Is there a changelog where I can see what are the new changes? If not I'll do a quick swap and see if anything breaks. 

I am thinking serilog console library is doing something weird with sbyte. Strangely every other type outputs fine. :)

timyhac, 2021-03-05:
Cool thanks!
No there isn't, but there are no changes to the public API - just bug fixes and refactoring for testability.

closed 2021-03-05 by timyhac

TheFern2, 2021-03-05:
@timyhac I have switched to v1.0.4-beta.1, I'll let you know if I run into any issues.