libplctag/libplctag.NET #152: Boolean Strange Results

TheFern2, 2021-03-09:
Hi, I am unable to write to a boolean single tag to make it go false, it does however work for making it true. I also tried a boolean array individual index, and I can make the first index BaseBOOLArray[0] true or false with no problems, but nothing else in the array.

I tried both async and non async with same results.
```csharp
var tag2 = new Tag<BoolPlcMapper, bool>()
{
    Name = "BaseBOOL",
    Gateway = "192.168.1.196",
    Path = "1,2",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};

tag2.Initialize();
tag2.Value = false;
tag2.Write();
```

I am on version 1.0.4 stable, I tried on 1.0.3 with same results.

kyle-github, 2021-03-09:
Do you have the possibility of trying with just the C DLL?   Use the `tag_rw` executable on the boolean tag and see if you can set it to 0 (false) or 255 (true)?   If that works, then we know that the underlying DLL is good.  If it does not work, then the C DLL is the problem.

TheFern2, 2021-03-09:
> Do you have the possibility of trying with just the C DLL? Use the `tag_rw` executable on the boolean tag and see if you can set it to 0 (false) or 255 (true)? If that works, then we know that the underlying DLL is good. If it does not work, then the C DLL is the problem.

Yeah, I can do that I was trying to find a good example for that. Is been a while since I messed with c, is there a good example you can point  me to?

kyle-github, 2021-03-09:
You can grab a ZIP file with all these example programs already compiled on the [releases page](https://github.com/libplctag/libplctag/releases).   The `tag_rw` tool allows you to enter tag strings on the command line without recompiling.   You can use `-d 4` to get debugging output (it will be a lot!).

If you want to do it the "hard way" you could look at the `toggle_bool.c` example.

Let me know if that helps!

TheFern2, 2021-03-09:
> You can grab a ZIP file with all these example programs already compiled on the [releases page](https://github.com/libplctag/libplctag/releases). The `tag_rw` tool allows you to enter tag strings on [...]

tag_rw.exe utility seems to write 0 and 255 to a bool with no issues.

TheFern2, 2021-03-09:
Here are the results from my previous test just in case:

```
C:\Users\fernandob\Downloads\libplctag_2.3.5_windows_x64\Release>tag_rw.exe -p "pro
tocol=ab_eip&gateway=192.168.1.196&path=1,2&plc=LGX&elem_size=1&elem_count=1&deb
ug=1&name=BaseBOOL" -w 0 -t uint8
Library version 2.3.5.
Arg[0]=tag_rw.exe
Arg[1]=-p
Arg[2]=protocol=ab_eip&gateway=192.168.1.196&path=1,2&plc=LGX&elem_size=1&elem_c
ount=1&debug=1&name=BaseBOOL
Arg[3]=-w
Arg[4]=0
Arg[5]=-t
Arg[6]=uint8
Wrote 0
Done

C:\Users\fernandob\Downloads\libplctag_2.3.5_windows_x64\Release>tag_rw.exe -p "pro
tocol=ab_eip&gateway=192.168.1.196&path=1,2&plc=LGX&elem_size=1&elem_count=1&deb
ug=1&name=BaseBOOL" -w 255 -t uint8
Library version 2.3.5.
Arg[0]=tag_rw.exe
Arg[1]=-p
Arg[2]=protocol=ab_eip&gateway=192.168.1.196&path=1,2&plc=LGX&elem_size=1&elem_c
ount=1&debug=1&name=BaseBOOL
Arg[3]=-w
Arg[4]=255
Arg[5]=-t
Arg[6]=uint8
Wrote 255
Done
```

TheFern2, 2021-03-09:
You're welcome it seems no matter where I go bools are always the evil brothers haha.

https://github.com/TheFern2/thefern.libplctag.NET I'm building an high level api for clx. Is a bit rough as c# isn't my main language but I need a dotnet core service. On that repo I already ran unit tests for all individuals tag types successfully except bools of course.

I need to build tests for all the arrays still.

If you want to build the unittests and setup to print an xml report or something along those lines, I can run against a clx plc. And submit report as a PR.

TheFern2, 2021-03-09:
@timyhac Yes, I visually saw the boolean tag get toggled in my rslogix5000 project online. I did it a few times, 0, 255.

TheFern2, 2021-03-10:
Ok, so I fired up my Softlogix VM at home, I was trying to check the interface for the Set Bit in the wrapper, but I went ahead and ran tag_rw with debugging level 4. One thing to note is that the exe help notes do not list bit as a valid type. However I did attempt to run with type bit, and the utility did not warned me, so I assumed it was good. Running with bit type did not change the online tag at all.

I captured 4 logs, two successful runs as uint8 with 0 and 255. And two unsuccessful runs with bit.

```
Usage:
 tag_rw -t <type> -p <path> [-w <val>] [-d <lvl>]
  -t <type> - type is one of 'uint8', 'sint8', 'uint16', 'sint16',
               'uint32', 'sint32', or 'real32'.  The type is the type
              of the data to be read/written to the named tag.  The
              types starting with 'u' are unsigned and with 's' are signed.
              For floating point, use 'real32'.
  -p <path> - The path to the device containing the named data.
  -w <val>  - The value to write.  Must be formatted appropriately
              for the data type.  Optional.
```

uint8 write 0:

```
tag_rw.exe -p "protocol=ab_eip&gateway=192.168.56.106&path
=1,2&plc=LGX&elem_size=1&elem_count=1&debug=4&name=BaseBOOL" -w 0 -t uint8
Library version 2.3.5.
Arg[0]=tag_rw.exe
Arg[1]=-p
Arg[2]=protocol=ab_eip&gateway=192.168.56.106&path=1,2&plc=LGX&elem_size=1&elem_count=1&debug=4&name=BaseBOOL
Arg[3]=-w
Arg[4]=0
Arg[5]=-t
Arg[6]=uint8
2021-03-09 19:44:30.259 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2021-03-09 19:44:30.259 thread(1) tag(0) INFO ab_tag_create:171 Starting.
2021-03-09 19:44:30.259 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:178
2021-03-09 19:44:30.259 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000000000273FF0
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL ab_tag_create:184 tag=0000000000273FF0
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL get_plc_type:916 Found ControlLogix/CompactLogix PLC.
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL get_plc_type:916 Found ControlLogix/CompactLogix PLC.
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2021-03-09 19:44:30.259 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2021-03-09 19:44:30.259 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL session_create_unsafe:449 Session should use connected messaging.
2021-03-09 19:44:30.260 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2021-03-09 19:44:30.260 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000000000276910
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 1,2.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 1.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 2 in string 1,2.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 2.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router obje
ct.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 6
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL cip_encode_path:456 Done
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL add_session_unsafe:310 Starting
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL add_session_unsafe:320 Done
2021-03-09 19:44:30.260 thread(1) tag(0) INFO session_create_unsafe:553 Done
2021-03-09 19:44:30.260 thread(1) tag(0) INFO session_init:569 Starting.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL thread_create:791 Starting.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL thread_create:825 Done.
2021-03-09 19:44:30.260 thread(1) tag(0) INFO session_init:584 Done.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL session_find_or_create:300 Done
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL ab_tag_create:272 using session=0000000000276910
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL get_tag_data_type:481 Starting.
2021-03-09 19:44:30.260 thread(1) tag(0) DETAIL setup_tag_listing:2432 Starting.
[... 1645 lines cut ...]
2021-03-09 19:51:35.870 thread(1) tag(0) INFO lib_teardown:160 Tearing down library.
2021-03-09 19:51:35.871 thread(1) tag(0) INFO lib_teardown:165 Tearing down tag tickler thread.
2021-03-09 19:51:35.871 thread(1) tag(0) INFO lib_teardown:172 Tearing down tag lookup mutex.
2021-03-09 19:51:35.871 thread(1) tag(0) DETAIL mutex_destroy:748 destroying mutex 000007FEF80186B8
2021-03-09 19:51:35.872 thread(1) tag(0) DETAIL mutex_destroy:761 Done.
2021-03-09 19:51:35.872 thread(1) tag(0) INFO lib_teardown:178 Destroying tag hashtable.
[... cut ...]
2021-03-09 19:51:35.876 thread(1) tag(0) INFO mb_teardown:1794 Starting.
2021-03-09 19:51:35.877 thread(1) tag(0) DETAIL mb_teardown:1798 Destroying Modbus mutex.
2021-03-09 19:51:35.877 thread(1) tag(0) INFO mb_teardown:1804 Done.
2021-03-09 19:51:35.877 thread(1) tag(0) INFO lib_teardown:160 Tearing down library.
2021-03-09 19:51:35.877 thread(1) tag(0) INFO lib_teardown:185 Done.
2021-03-09 19:51:35.878 thread(1) tag(0) DETAIL plc_tag_unregister_logger:945 Starting
2021-03-09 19:51:35.878 thread(1) tag(0) DETAIL plc_tag_unregister_logger:949 Done.
```

TheFern2, 2021-03-10:
Debugging information:

bit write 1 Successfully wrote True to the PLC tag:
```
PS C:\libplctag> .\bool.bat

C:\libplctag>tag_rw.exe -p "protocol=ab_eip&gateway=192.168.56.106&path=1,2&plc=LGX&elem_size=1&elem_count=1&debug=4&nam
e=BaseBOOL" -w 1 -t bit
Library version 2.3.5.
Arg[0]=tag_rw.exe
Arg[1]=-p
Arg[2]=protocol=ab_eip&gateway=192.168.56.106&path=1,2&plc=LGX&elem_size=1&elem_count=1&debug=4&name=BaseBOOL
Arg[3]=-w
Arg[4]=1
Arg[5]=-t
Arg[6]=bit
2021-03-09 20:37:35.612 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2021-03-09 20:37:35.612 thread(1) tag(0) INFO ab_tag_create:171 Starting.
2021-03-09 20:37:35.612 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:178
2021-03-09 20:37:35.612 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000000000305CF0
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL ab_tag_create:184 tag=0000000000305CF0
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL get_plc_type:916 Found ControlLogix/CompactLogix PLC.
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL get_plc_type:916 Found ControlLogix/CompactLogix PLC.
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2021-03-09 20:37:35.612 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL session_create_unsafe:449 Session should use connected messaging.
2021-03-09 20:37:35.612 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2021-03-09 20:37:35.612 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-03-09 20:37:35.612 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000000000308610
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 1,2.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 1.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 2 in string 1,2.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 2.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router obje
ct.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 6
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL cip_encode_path:456 Done
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL add_session_unsafe:310 Starting
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL add_session_unsafe:320 Done
2021-03-09 20:37:35.613 thread(1) tag(0) INFO session_create_unsafe:553 Done
2021-03-09 20:37:35.613 thread(1) tag(0) INFO session_init:569 Starting.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL thread_create:791 Starting.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL thread_create:825 Done.
2021-03-09 20:37:35.613 thread(1) tag(0) INFO session_init:584 Done.
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL session_find_or_create:300 Done
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL ab_tag_create:272 using session=0000000000308610
2021-03-09 20:37:35.613 thread(1) tag(0) DETAIL get_tag_data_type:481 Starting.
2021-03-09 20:37:35.614 thread(1) tag(0) DETAIL setup_tag_listing:2432 Starting.
2021-03-09 20:37:35.614 thread(1) tag(0) INFO setup_tag_listing:2526 Tag is not a tag listing request.
2021-03-09 20:37:35.614 thread(1) tag(0) INFO get_tag_data_type:647 Setting element size to 1.
2021-03-09 20:37:35.615 thread(1) tag(0) DETAIL get_tag_data_type:671 Done.
2021-03-09 20:37:35.615 thread(1) tag(0) DETAIL ab_tag_create:338 Setting up Logix tag.
2021-03-09 20:37:35.615 thread(1) tag(0) DETAIL parse_symbolic_segment:912 Starting with name index=0 and encoded name i
ndex=1.
2021-03-09 20:37:35.616 thread(1) tag(0) DETAIL parse_symbolic_segment:951 Parsed symbolic segment "BaseBOOL" in tag nam
e.
2021-03-09 20:37:35.616 thread(1) tag(0) INFO tag_read_start:282 Starting
2021-03-09 20:37:35.616 thread(1) tag(0) INFO build_read_request_connected:400 Starting.
2021-03-09 20:37:35.616 thread(1) tag(0) DETAIL session_create_request:2334 Starting.
2021-03-09 20:37:35.617 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2343
2021-03-09 20:37:35.617 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-03-09 20:37:35.617 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000000003027C0
[... 774 lines cut ...]
2021-03-09 20:39:31.073 thread(1) tag(0) INFO lib_teardown:165 Tearing down tag tickler thread.
2021-03-09 20:39:31.073 thread(1) tag(0) INFO lib_teardown:172 Tearing down tag lookup mutex.
2021-03-09 20:39:31.073 thread(1) tag(0) DETAIL mutex_destroy:748 destroying mutex 000007FEF2BE86B8
2021-03-09 20:39:31.073 thread(1) tag(0) DETAIL mutex_destroy:761 Done.
[... cut ...]
2021-03-09 20:39:31.075 thread(1) tag(0) DETAIL mb_teardown:1798 Destroying Modbus mutex.
2021-03-09 20:39:31.076 thread(1) tag(0) INFO mb_teardown:1804 Done.
2021-03-09 20:39:31.076 thread(1) tag(0) INFO lib_teardown:160 Tearing down library.
2021-03-09 20:39:31.077 thread(1) tag(0) INFO lib_teardown:185 Done.
2021-03-09 20:39:31.077 thread(1) tag(0) DETAIL plc_tag_unregister_logger:945 Starting
2021-03-09 20:39:31.078 thread(1) tag(0) DETAIL plc_tag_unregister_logger:949 Done.
PS C:\libplctag>
```

timyhac, 2021-03-10:
Thanks for doing this - that does narrow it down.

So it could be either:
* libplctag.NET should not use the bit accessor APIs for bool tags.
* There is an issue with libplctag core around these methods.

@kyle-github - are the [bit accessor methods](https://github.com/libplctag/libplctag/wiki/API#bit-accessors) suitable for ordinary BOOL tags or BOOL arrays?

TheFern2, 2021-03-10:
@timyhac No prob. From reading the core lib api and plc_tag_set_bit function, it does sound like plc_tag_set_bit should work.
Did the same test but using the native import in c#, just to make sure it was the same results.

```csharp
// plctag.plc_tag_set_uint8(tagHandle, 0, 255);    // Works for false and true
plctag.plc_tag_set_bit(tagHandle, 0, 1);           // Works only for true
plctag.plc_tag_write(tagHandle, 1000);
plctag.plc_tag_read(tagHandle, 1000);
```

kyle-github, 2021-03-10:
Hi, sorry for the delay.  Work is taking all my waking hours right now.

The bit functions are for individual bits within other words.   For instance picking out a single bit in a DINT (`name=MyDint.23`) or single flags within BOOL arrays.   If you read/write BOOL tags (not arrays of BOOL!), you get a single byte back.   In that case, IIRC, false is 0 and true is 255.

Are you using the `release` branch?   When I test I want to make sure that I am using the same baseline as you.  

Thanks for bringing this up.   This is really important!

GitHubDragonFly, 2021-03-11:
This issue seems to be in the library itself.

The VB app that I posted was doing the same thing, writing True but not writing False.
I ended up changing it to set uint8 instead of bit for BOOL type.

timyhac, 2021-03-11:
@kyle-github - yes the libplctag native libraries packaged with libplctag.NET are always from the release branch. And currently are using v2.3.5.

Unless @TheFern2 - you've supplied your own native libraries?

I've started working on splitting up the `PlcMapper` classes for bool types - [one for arrays of various dimensions](https://github.com/libplctag/libplctag.NET/blob/%23152/src/libplctag/DataTypes/BoolArrayPlcMapper.cs) which uses bit accessors, and [another for single bools](https://github.com/libplctag/libplctag.NET/blob/%23152/src/libplctag/DataTypes/BoolPlcMapper.cs) which uses uint8 APIs.

TheFern2, 2021-03-11:
> @kyle-github - yes the libplctag native libraries packaged with libplctag.NET are always from the release branch. And currently are using v2.3.5.
> 
> Unless @TheFern2 - you've supplied your own native libraries?
> 
> I've started working on splitting up the `PlcMapper` classes for bool types - [one for arrays of various dimensions](https://github.com/libplctag/libplctag.NET/blob/%23152/src/libplctag/DataTypes/BoolArrayPlcMapper.cs) which uses bit accessors, and [another (new) one for standalone bool tags](https://github.com/libplctag/libplctag.NET/blob/%23152/src/libplctag/DataTypes/BoolPlcMapper.cs) which uses uint8 APIs.
> 
> I don't have access to a controller at the moment so I won't be able to test it. @TheFern2 - any chance you could try these two to check that they work. I would check the single tag, as well as 1d, 2d and 3 dimensional arrays.
> 
> You can just copy the class into your own project and then use that as the type argument for the Tag.
> 
> ```
> var boolTag = new Tag<BoolPlcMapper, bool>(){ ... };
> var boolArrayTag = new Tag<BoolArrayPlcMapper, bool[]>(){ ... };
> ```

I will give it a try, and let you know. My libplctag.NET install is straight from nuget packages.

TheFern2, 2021-03-11:
@timyhac with the separate new bool mappers, everything is passing. Bool arrays worked previously, but I went ahead and still use your new mapper. Individual bool tags are now passing the unit tests. 🎉 

[image]

I rarely use multi dim arrays in my line of work, but I will still test it anyways.

Do you have a clear example on how to declare a tag for 2d and 3d? I was doing something like this, but ArrayDimensions only wants a one dim.

```csharp
var tag = new Tag<BoolArrayPlcMapper, bool[,]>()
{
    Name = tagName,
    Gateway = _ipAddress,
    Path = _path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(Timeout),
    ArrayDimensions = new int[,] { firstDim, secondDim }
};
```

TheFern2, 2021-03-11:
got caught up in another project sorry for the late reply, 2D, and 3D read and write are all in good working order. : ) Feel free to close this issue.

timyhac, 2021-03-11:
Cool thanks, appreciate it!

TheFern2, 2021-03-11:
> Cool thanks, appreciate it!

Np, Thank you!

closed 2021-04-03 by timyhac