libplctag/libplctag.NET #153: Feedback: Tag class, Mapper and params abstraction

TheFern2, 2021-03-10:
This is just some feedback. I think the Tag class could use a bit more of abstraction to promote DRY. I don't know if this is all feasible, but nevertheless these are my thoughts.

Current tag creation:
```csharp
var tag = new Tag<BoolPlcMapper, bool>()
{
    Name = tagName,
    Gateway = _ipAddress,
    Path = _path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(Timeout)
};
```

OR:
```csharp
var tag = new Tag<BoolPlcMapper, bool[]>()
{
    Name = tagName,
    Gateway = _ipAddress,
    Path = _path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(Timeout),
    ArrayDimensions = new int[] { arrayLength }
};
```

1) I don't think BoolPlcMapper should be a worry for the end user, since you can deduct mapper according to the type. Bool BoolPlcMapper, Int DintPlcMapper, and so on...
2) The connection details could be abstracted to a Connection class. End result could look something like this.

Handle the connection once:
```csharp
var conn = new Connection()
{
    Gateway = _ipAddress,
    Path = _path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(Timeout)
};
```

Tag creation could look something like this:
```csharp
var tag = new Tag<bool>()
{
    Name = tagName,
    Connection = conn
};
```

```csharp
var tag = new Tag<bool>()
{
    Name = tagName,
    ArrayDimensions = new int[] { arrayLength },
    Connection = conn
};
```

Or even:
```csharp
var tag = new Tag<bool>()
{
    Name = tagName,
    ArrayDimensions = new int[] { arrayLength }
};

tag.UseConnection(conn);
```

timyhac, 2021-03-10:
Thanks @TheFern2 - I agree with the first one, we can just make an assumption about which Mapper to use based on the type argument. When that assumption is wrong, you'll know before compiling and you can manually specify the Mapper.

The second one we've gone back and forth on - it has indeed been implemented in [the groups branch ](https://github.com/libplctag/libplctag.NET/tree/groups) as TagGroups and AttributeGroups:
* https://github.com/libplctag/libplctag.NET/blob/groups/src/Examples/CSharp%20DotNetCore/ExampleGroups.cs
* https://github.com/libplctag/libplctag.NET/blob/groups/src/libplctag/TagGroup.cs
* https://github.com/libplctag/libplctag.NET/issues/71

What I'll do here is create a new Github issue for the single type argument version of `Tag` so we can discuss each of these items in the appropriate issue.

closed 2021-03-10 by timyhac

TheFern2, 2021-03-10:
Awesome! thanks for the details. Groups looks pretty good!

timyhac, 2021-03-10:
Yes thanks - I really appreciate the level to which you've been providing feedback.