libplctag/libplctag.NET #154: Can't catch Exceptions

WpIT1640, 2021-03-10:
I'm trying to learn how to catch Exceptions and it's not working for some reason. I've tried a try catch block like this
// Read the value from the PLC
            try
            {
                dryerStack1.Read();
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);
                Console.WriteLine(ex.StackTrace);
                throw;
            }
           

As well as the Appdomaon catch handler on the msdn site. I looked through the code examples but I didn't see any error handling.
I'm able to read from my plc's just fine but I introduced an error (incorrect ip address) in an attempt to prepare for network latency.

timyhac, 2021-03-10:
Unless you've changed the timeout somewhere, the default timeout is 10seconds. It should throw an error after timing out.
That would be with libplctag.NET v1.0.4

What happens if you try this:

```csharp
try
{
    1 / 0 ;
}
catch (Exception ex)
{
    Console.WriteLine(ex.Message);
    Console.WriteLine(ex.StackTrace);
    throw;
}
```

WpIT1640, 2021-03-10:
Attempted to divide by zero.
   at crashtest.Program.Main() in C:\Users\aaron\source\repos\crashtest\Program.cs:line 16

Unhandled Exception: System.DivideByZeroException: Attempted to divide by zero.
   at crashtest.Program.Main() in C:\Users\aaron\source\repos\crashtest\Program.cs:line 22
   
   That's what I get when I try and run it. SO maybe there's something hoofy with my visual studio install.

WpIT1640, 2021-03-10:
I know with the c library there is a way to test for this and then move on gracefully, is that possible with the .net implementation?

timyhac, 2021-03-10:
There is, if you look at the Native import examples you'll see how that's done.

I would suggest figuring out how to deal with exceptions though, IMO it's essential to c# development

WpIT1640, 2021-03-10:
The solution was staring me in the face.
catch (libplctag.LibPlcTagException)
And now I can move on. Thanks!

closed 2021-03-10 by WpIT1640