libplctag/libplctag.NET #155: Single type argument version of Tag

timyhac, 2021-03-10:
The current API provides two versions of `Tag`.
* `Tag`
* `Tag<M,T>`

This issue suggests that we add a third type that takes a guess at which Mapper class to use based on a single type argument for the CLR type to use. i.e.
`Tag<T>`

Some discussions where this was discussed:
* https://github.com/libplctag/libplctag.NET/issues/28#issuecomment-653854167
* https://github.com/libplctag/libplctag.NET/issues/153

What needs to be decided?
* What should the mappings be between CLR type and Mapper types.

jkoplo, 2021-03-10:
I remember considering this during the design process for Mappers/Tags. I don't remember why I didn't implement it. I think there's something really weird with the way generics are handled that made this not feasible. Or maybe it just seemed like added complexity until we were comfortable with the Mapper strategy...

timyhac, 2021-03-11:
One reason was that there isn't a 1 to 1 mapping between PLC types and CLR types.

timyhac, 2021-03-11:
@jkoplo - I've taken a stab at what this could look like here: https://github.com/libplctag/libplctag.NET/blob/%23155/src/libplctag/TagOfT.cs#L27

If you get a chance, could you take a look and check that this is roughly what you were thinking last year?

There's definitely room for sharing code with `TagOfMT` (I recall you pointing out that this is what it should have been called originally), and also I would like to refactor these classes to allow for DependencyInjection / Mocking so that they can be unit tested.

jkoplo, 2021-03-11:
I took a quick spin through it, but your other comment jogged my memory.

If we only specify the CLR type, how do we know what the PLC type is to create the tag? (Since they aren't 1-to-1). Or can we actually get by without knowing that?

Other than that, in terms of implementation I was thinking of using a dictionary or even a factory-pattern to get the mapper. That has the benefit of getting the case structure out of the class and allowing for runtime changes (adding custom mappers) with the trade-off of increased complexity. If we didn't want to DI the lookup, then we'd have to make it a static. That being said, defining it in code is more or less a static anyway.

The other route would be to go toward an actual factory-pattern for tags. I end up implementing that in my local code anyway. Create a Tag factory and set PLC path/params/etc and then just use it to instantiate tags. Each factory instance could hold an instance of the lookup that it uses to DI. At that point, we could just *pass* the proper mapper to the `Tag` and take that concern away from the `Tag` class. Thoughts?

timyhac, 2021-03-11:
I didn't think of being able to change it at runtime, that is much better! That makes it less of a problem that the mappings aren't 1-1, we can just take a guess and if its wrong the user can override.

jkoplo, 2021-03-11:
Want me to take a stab at it? It's mostly in my head already...

jkoplo, 2021-03-11:
Also, how do you feel about a tag factory? Is that going too far beyond the "thin wrapper" ethos?

timyhac, 2021-03-11:
Go for it!

My view would be that a Tag factory it is too far beyond the thin ethos. There is overlap there with the TagGroup / AttributeGroup idea that started out of [this post by Kyle](https://github.com/libplctag/libplctag.NET/issues/66#issuecomment-664769252).

TheFern2, 2021-03-12:
> The other route would be to go toward an actual factory-pattern for tags. I end up implementing that in my local code anyway. Create a TagFactory and set PLC path/params/etc and then just use it to instantiate tags with a Type and Name. Each factory instance could hold an instance of the lookup that it uses to DI. At that point, we could just pass the proper mapper to the Tag and take that concern away from the Tag class. Thoughts?]

This is exactly what I was going to try implement in my code, but I think is better to be done in this library. I called mine TagBuilder, but I do believe abstracting this layer from the end user will make tag creation 1000x easier.

TheFern2, 2021-03-12:
@jkoplo How do you normally tackle the tag factory in your code. Say I have 5 tags different types. Is it possible to build a list of different types with generics?

jkoplo, 2021-03-12:
@TheFern2  - there's some good-natured back and forth on whether the library acknowledges the existence of a controller or not. @kyle-github has done a lot of work to keep tags separate from any idea of a 'controller' abstraction. In my use-case, I'm generally building HMIs that are 1-to-1 with a controller, so having a controller object is nice and gets used for things other than creating tags.

Here's a really simple implementation that makes the controller into a sort of tag-factory. Really it's just letting me create tags in other parts of my code without constantly specifying all parameters. You'll see I also have a watchdog property in there that I use in other parts of my code.

```C#
    public class PLCController
    {
        public PLCController(string iPAddress, int watchdogPollRate)
        {
            if (string.IsNullOrEmpty(iPAddress))
            {
                throw new ArgumentException("Must specify controller IP", nameof(iPAddress));
            }

            if (watchdogPollRate < 100)
            {
                throw new ArgumentException("Poll rate must be > 100 msec", nameof(watchdogPollRate));
            }

            this.IPAddress = iPAddress;
            this.WatchdogPollRate = watchdogPollRate;
        }

        public string IPAddress { get; }

        public int WatchdogPollRate { get; }

        public Tag<M,T> CreateTag<M,T>(string tagName) where M:IPlcMapper<T>, new()
        {
            var tag = new Tag<M, T>()
            {
                Name = tagName,
                Gateway = IPAddress,
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(3)
            };
            return tag;
        }
    }
```
You can't create a `List` of different `Tag<M,T>` if M and T aren't identical. You can create a 'List<ITag>' of different types and then (for example) call `Read` on all of them. But you can't access the values they read without casting them back to their actual types.

I've considered implementing some sort of `StringValue` on *all* tags. It would probably just be the JSON representation of that tag. This would make things a lot easier if you're final target is JSON anyway (you're hosting a WebAPI endpoint or want to save a bunch of Tag Values to a file for logging or configuration purposes).

We've gone back and forth about how low level the library should be. The goal of libplctag is to be PLC-agnostic and we've tried to stick to that. I have considered a `libplctag.Rockwell` library that depends on this one that would make things dirt simple (and include a controller abstraction). Maybe that's a good path forward at some point.

TheFern2, 2021-03-15:
I do think the perhaps a PLC object might be better in a rockwell lib, now that you explained the lib methodology. I'll keep working on my lib, and see where it goes, mine will only be for CLX PLCs though, have no interest in supporting DF1 and old protocols. If my lib goes well I'll reach out to you guys for a good name, prob libplctag.NET.CLX. Thanks again for the details.

jkoplo, 2021-03-19:
@timyhac and @TheFern2 take a look at my latest commit to branch `#155`, specifically here: [MapperLookup.cs](https://github.com/libplctag/libplctag.NET/blob/%23155/src/libplctag/DataTypes/MapperLookup.cs)

I basically wrote a thin wrapper around a simple IoC container (thanks Microsoft!). I think it accomplishes the goal but there are a couple things that come to mind:

1. Statics - using a static feels all kinds of bad and like it's going to cause future maintenance issues.
2. The files need better naming and I don't like that I dropped them into the Datatypes folder
3. Does this really make sense??

What I mean by number 3 is that I feel like this is a little weird... If I'm a PLC programmer dabbling in C# I'm going to be thinking of my types by the PLC names - that makes instantiating tags hard. I need to know that my **INT** should be instantiated as `Tag<short>`. If I'm a C# dev dabbling in PLCs, I need to know that I can't just instantiate all non-floating point numbers as `Tag<int>`.

I'm just wondering if this only makes things simpler for people that already know exactly how the two type systems match up...

Thoughts?

timyhac, 2021-03-20:
With respect to point 3, if the programmer gets it wrong and for example uses int instead of short, does that throw an error?
I don't have access to a CPU to test this at the moment.

I think it is fair that we require them to use the appropriate datatype, otherwise we would need to do something like infer the PLC type (using @tag listing or [similar](https://github.com/libplctag/libplctag/issues/252)) and then have a matrix of mappers to go between their chosen CLR type and the PLC type.

kyle-github, 2021-03-20:
Is there any way to make type equivalences or type synonyms?  I.e. in C and C++ you can do:

```c
typedef int16_t INT;
typedef int32_t DINT;
```

Then you can use `DINT` and the C/C++ code will understand.   I am not sure that there is a way to do it with Java.

timyhac, 2021-03-20:
There is the [using alias directive](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/keywords/using-directive) - how are you imagining this would be used?

kyle-github, 2021-03-20:
In C/C++, you have header files and the typedefs go there, so anyone using that header would be able to use the type `DINT` instead of `int`.  It is a type alias, but public to anyone using the header.

From the link you posted, it looks like a using alias is within a single file?   I just do not know enough about C# to really understand yet.

Using a typedef, in C++ you could do something like this:

```cpp
DINT sum = 0;
Tag<DINT, "protocol=ab-eip&....name=MyTag"> myTag;
...
sum += myTag;
...
```

Then people can use the "native" PLC types directly and in the generics (like `Tag` above).

kyle-github, 2021-03-21:
Hmm, now that I think about it, in C++ you could also probably create singular `Tag` types as above and array types.  Like this:

```cpp
DINT sum = 0;
ArrayTag<DINT, 400, "protocol=....name=MyArrayTag"> myArrayTag;
...
for(int i=0; i < myArrayTag.size(); i++) {
    sum += myArrayTag[i];
}
```

I am intentionally not showing a C++ `foreach` equivalent here, but that would probably work too.   Of course, C++ lets you override everything.

timyhac, 2021-03-21:
I think thats similar to the class that exposes both the local type and the PLC type (by way of declaring which mapper to use).

```csharp
// Instantiate the tag with the appropriate mapper and datatype
// The DintPlcMapper declaration tells our wrapper to use the Signed 32-bit integer accessors
var myTag = new Tag<DintPlcMapper, int>()
{
    Name = "PROGRAM:SomeProgram.SomeDINT",
    Gateway = "10.10.10.10",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};

myTag.Value = 20;
```

In order to get PLC data into a local variable, at some point we need to know which of the data accessor methods to use.
There are three ways of doing that that I can see:
* The programmer tells us (either directly using the data accessor methods, or using a `Mapper` class).
* The PLC tells us (by using tag listing or [similar](https://github.com/libplctag/libplctag/issues/252)).
* We make a guess based on the type of the local variable.

We've got the first option [already](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/TagOfT.cs).
The third option is what we've been looking at so far in this ticket.

We don't really have anything for the second option, but is possible.

jkoplo, 2021-03-21:
Okay here's an idea.

Maybe we're overthinking this completely and wrapped up in our C# cleverness with generics...

What if we just created `TagDint` (and ALL the others) and inherited it from the proper type and mapper? That makes things really simple for anyone using the library, definitely simpler than even instantiating with a single generic.

The `TagOfT` would still be available for more advanced programming like UDT, etc.

It would be a bunch of classes and boilerplate, but we're already basically doing this with defining the .net to plc types.

TheFern2, 2021-03-24:
> Okay here's an idea.
>
> Maybe we're overthinking this completely and wrapped up in our C# cleverness with generics...
>
> What if we just created `TagDint` (and ALL the others) and inherited it from the [...]

I don't know about this one, I actually like the generic way of creating tags, because I can easily create on read and one write generic function, instead of N amount of read/write functions per type, but if I can still create tags with generics with TagOfT, then it might sound like a good idea.

I guess it doesn't really matter how much code duplication we have if we make a nice API for the user, but in the end it might be more maintenance.

TheFern2, 2021-03-24:
> @timyhac and @TheFern2 take a look at my latest commit to branch `#155`, specifically here: [...]

I definitely agree. As a c# programmer I should get an compiler error if my tag can't be an int, I should get a compiler error not a runtime error. Perhaps, having both mapper, and type wasn't such of bad idea like the current implementation.

closed 2021-12-03 by timyhac

turbozob, 2022-12-10:
@jkoplo how do you call function to create tag, read write value from another class?

TheFern2, 2022-12-11:
> @jkoplo how do you call function to create tag, read write value from another class?

Please don't [necro](https://en.wiktionary.org/wiki/thread_necromancy) threads. This question has nothing to do with this discussion.

Post questions to libplctag forum or open another issue with your question.
https://github.com/libplctag/libplctag#libplctag-forum

PR #215 (2021-12-03): Implementing really simple Tag classes
Closes #155 

I felt a little dirty writing this code, but the intended end user is for devs that are new-er to C#. Generics can be confusing and even more so when a class is defined with two generics. This is really just some helper syntax to keep things simpler. I put these definitions in a single file in their own `libplctag.Datatypes.Simple` - I figured this should help keep them out of the way of more experienced developers.

I'm open to using different naming conventions for the namespace and for the classes - these were just the first that came to  [... cut]