libplctag/libplctag.NET #157: Projects with "#" in the name causes extraction of the library to the incorrect folder

timyhac, 2021-03-12:
My project has "#" in the name and that's where uri escapes and directs extraction of the library to the parent folder of the project instead of the Debug folder where the application is.

I used the libraries from the latest libplctag release and all these files in your project amount to 2.2 MB.
Some people might opt to go with a zip file instead of clone and that's when they will realize that it doesn't work properly.

_Originally posted by @GitHubDragonFly in https://github.com/libplctag/libplctag.NET/issues/137#issuecomment-797308965_

timyhac, 2021-03-13:
> Have you tried using this: var extractDirectory = Directory.GetCurrentDirectory();

_Originally posted by @GitHubDragonFly in [#137](https://github.com/libplctag/libplctag.NET/issues/137#issuecomment-797325772)_

timyhac, 2021-03-13:
Hey there @githubdragonfly - I've broken out this problem into its own ticket.

It has been a while since I implemented this, but I recall that there being an issue with using Current Working Directory is that it isn't thread-safe, so some other part of the application might change it just prior to extraction, leading it to be in the wrong place.

I also vaguely recollect there being something with .NET Core when packaging as a single file, the executing location of the assembly can be different to the current working directory.

----

Doing a bit of research into this problem it looks like there is an alternative to[ `Assembly.CodeBase`](https://docs.microsoft.com/en-us/dotnet/api/system.reflection.assembly.codebase?view=netstandard-2.0) called [`Assembly.Location`](https://docs.microsoft.com/en-us/dotnet/api/system.reflection.assembly.location?view=netstandard-2.0) which returns the file path in a different format and negatives the need to use [`UriBuilder`](https://docs.microsoft.com/en-us/dotnet/api/system.uribuilder?view=netstandard-2.0).
[#158 contains the change](https://github.com/libplctag/libplctag.NET/blob/%23157/src/libplctag.NativeImport/LibraryExtractor.cs#L31).

This seems like it would solve the immediate problem of projects with # in the path.

GitHubDragonFly, 2021-03-13:
That fix appears to be working fine, at least in Windows.

You should probably close this issue then (people can reopen it if they need to).

timyhac, 2021-03-13:
Thanks @GitHubDragonFly for checking, I'll make a few more minor modifications and release the new nuget package.

closed 2021-03-13 by timyhac

timyhac, 2021-03-13:
Done in v1.0.12