libplctag/libplctag.NET #162: Expose remaining attributes on `Tag` or `Tag<M,T>`

timyhac, 2021-03-13:
There are many more attributes that could be configured but are currently not exposed on `Tag` and `Tag<M,T>`.
https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes

ShadowMarker789, 2021-06-09:
I would also like this functionality exposed to C#. 

I've had great success with LibPlcTag.NET so far, this would be a very welcome addition.

timyhac, 2021-06-09:
Which ones in particular do you need?

ShadowMarker789, 2021-06-10:
Ah, apparently I've misunderstood. 

I somehow got the impression that 'attributes' was talking about data present in the CIP Service Success messages being returned that wasn't exposed to the library, or tag extended properties. 

For instance, Allen-Bradley document 1756-pm020 shows that the PLC-side data-type is specified in the frame along with the data. 

E.g.: [ snippet from Page 18 ] 
[image]

But as far as I'm aware there's no methods for obtaining that on the C# side. Don't know if that's exposed on the C-side either.

timyhac, 2021-06-10:
Yes, that is different to the intent here and I think is exactly what is discussed on https://github.com/libplctag/libplctag/issues/252, and is not yet possible in the core library.

timyhac, 2021-11-03:
Hi https://github.com/rastreus - just wondering if you have had a chance to do any testing on this?

rastreus, 2021-11-03:
No. I have not yet, but I am still working on a project that will use libplctag.NET and this functionality. I hope to get this tested with a real PLC soon.

timyhac, 2022-03-05:
Hi @rastreus - have you had a chance to test your project as yet?

closed 2022-03-05 by timyhac