libplctag/libplctag.NET #163: Question: Array offset, and ranges

TheFern2, 2021-03-15:
I am in a bit of a pickle when it comes to array ranges. I have managed to clean most of my code. The way my API works is as follows:

Two functions which can read any tag or array (1d, 2d, 3d), although I mostly care for 1d:

These return a Response object which has tagname, value as string or string[], and status.
Read and Write.
```csharp
await myPLC.Read("BaseBOOL", TagType.Bool);
await myPLC.Read("BaseBOOLArray", TagType.Bool, 128);
await myPLC.Write("BaseDINT", TagType.Dint, 545437493);
```

ReadTag and WriteTag also return a Response object, but value is an actual type:
```csharp
await myPLC.ReadTag<DintPlcMapper, int>("BaseDINT");
await myPLC.WriteTag<DintPlcMapper, int>("BaseDINT", 545437492);

[Response: BaseDINT -545437484 Success]
```
So far so good, Read and Write for single tags, I have cleaned up and have almost no code duplication. For arrays is another story. I have went about how to solve this for a few days but I can't seem to be able to get it right.

For Array Read Write ranges, for example I only want to read/write from 0 to 10:
```csharp
await myPLC.Read("BaseSTRINGArray", TagType.String, 128, 0, 10);
await myPLC.Write("BaseSTRINGArray", TagType.String, boolList.ToArray(), 128, 0, 10);
```

I wanted to have one function that can read an array that is the easy part, however I want to get a range, I can do this easily when I am explictily casting Array.ConvertAll but there is too much code duplication my example below I have for each data type. So I am wondering if there is a better way.

```csharp
private async Task<Response<string[]>> _ReadArray<M, T>(string tagName, int arrayLength, int startIndex = 0, int count = 0) where M : IPlcMapper<T>, new()
{
    var results = await ReadTag<M, T>(tagName, new int[] { arrayLength });
    if (results.Status == "Success")
    {
        string[] arrString = Array.ConvertAll<T, string>(results.Value, Convert.ToString);
        // Sanity check to make sure we don't try to access items indices bigger than array
        if (startIndex + count > arrayLength)
        {
            return new Response<string[]>(tagName, "Failure, Out of bounds");
        }

        if (count > 0)
        {
            List<T> tagValueList = new List<T>(arrString);
            return new Response<string[]>(tagName, tagValueList.GetRange(startIndex, count).ToArray(), "Success");
        }

        return new Response<string[]>(tagName, arrString, "Success");
    }
    else
    {
        return new Response<string[]>(tagName, "Failure");
    }
}
```

My question to you guys, how do you deal with ranges, is there an easier way, or do you think my solution could work somehow?

timyhac, 2021-03-15:
When the TagOfT is initialized, they [initialize](https://github.com/libplctag/libplctag.NET/blob/8ad4d0809e978d7f02018b5d10cc4ba67c6b4f8b/src/libplctag/TagOfT.cs#L135) the `Value` on the tag.
So rather than the tag value containing a subset of the full array, it always reads/writes the full array. Rather, the application reads writes to the subset.

Say you want to access MyTag[4-7]

```csharp
var ValuesToWrite = new []{12,23,34};
Array.Copy(ValuesToWrite, 0, MyTag.Value, 4, 3);
```

I think to write to a range in the PLC, the tag name would need to include the first element you want to write to ( and the ElementCount is the length of array.
I don't know how this works for 2d, or 3d arrays.

İf you're looking to support just CompactLogix you could look at building the tag name from the base tag name and adding the first index that you're accessing.

TheFern2, 2021-03-15:
That's a good idea for the base tag. In my code I read a full array, then either return or write only the range I want. The rest get repacked the same. So for the user it looks like they're working with ranges. 

Just wanted to abstract that portion to a function, to avoid duplicated code per type, but is difficult to cast a T value to a List. The other part is that I didn't want to write to values that don't need to be written in case another part of the PLC program write to those indices in the next plc scan.

I'll play around with this idea to see what I can come up with. Thanks for your thoughts.

timyhac, 2021-03-16:
Cool, I'm sure you'll come up with something great.

The conclusion I've come to in my projects is that tags should either read **_or_** write the PLC - not both.

closed 2021-03-16 by timyhac

kyle-github, 2021-03-16:
I seriously considered making tags be only read or only write.   I too found that it was really rare that I needed both.   The only major exception was when I needed to RMW a bit within a tag.   Now that I have bit support, even that goes away.

TheFern2, 2021-03-16:
> That's a good idea for the base tag. In my code I read a full array, then either return or write only the range I want. The rest get repacked the same. So for the user it looks like they're working with ranges.

Ah I see this statement is a bit of a mess. Above statement are two functions, not one.

For the Write function where I do Write array range, internally I read the whole array first just so I can have the original array in memory before writing to it again. So I just swap the range the user wants with the new values, copy whatever wasn't change to a new buffer. Then I write the entire new array. But my goal is probably to see if there is a better way with the Native API to only write what I need not the whole array.

timyhac, 2021-03-16:
@TheFern2 - where you are heading with this looks like there will be a bit of friction with making use of the `TagOfT` class - you might want instead to use [`Tag`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/Tag.cs) and do all of the mapping yourself.

TheFern2, 2021-03-16:
@timyhac Thanks, looks like that might be the best route.