libplctag/libplctag.NET #169: access omron NX1P series PLC

moonps1, 2021-05-09:
HI.
It's a very nice and powerful lib for EIP.But i always need to commnunicate with  omron PLC in project.
And i really want to know  that does this dll support connecting NX1P Plc?

Best Regards.

jkoplo, 2021-05-09:
So the libplctag.net library has mostly been focused on connecting to Rockwell PLCs since that's what most of us work with in our professional lives. But with that said, the nativeimport library is just a C# wrapper around the base libplctag C library.

@kyle-github have you had any experience with Omron PLCs or their protocols? Are they supported in the base library?

kyle-github, 2021-05-10:
The Omron code is a contribution from a user of the library and supports NX/NP models.   I do not really know more than that!  Those models support a version of the Rockwell protocol.  I say "a version" because it has some slight differences:

- You must provide the IP address of the Omron in the path as if you were using bridged path mode.
- You get the entire tag even if you ask for one element of an array.
- It does not appear that you can have multiple requests in one packet.
- I really do not know what happens if the tag does not fit into a single packet.   I have no hardware on which to test this.

I hope that helps.

jfontsaballs, 2021-05-14:
I'm also trying to use your library to connect to the Omron NX1P2 PLC (which is one of the NX family of PLCs). I used this code:

```csharp
static void Main()
{
	var int1 = new Tag<IntPlcMapper, short>
	{
		Name = "ProvaInt1",
		Gateway = "172.26.5.11",
		Path = "1,0",
		PlcType = PlcType.ControlLogix,
		Protocol = Protocol.ab_eip,
		Timeout = TimeSpan.FromMilliseconds(500),
		UseConnectedMessaging = true
	};

	int1.Read();

	Console.WriteLine($"Value: {int1.Value}");
}
```

I get this exception upon execution of `Read()` method:

```
libplctag.LibPlcTagException
  HResult=0x80131500
  Message=ErrorUnsupported
  Source=libplctag
  StackTrace:
   at libplctag.NativeTagWrapper.Initialize() in C:\Users\jfont\Desktop\libplctag.NET-master\src\libplctag\NativeTagWrapper.cs:line 184
   at libplctag.NativeTagWrapper.InitializeIfRequired() in C:\Users\jfont\Desktop\libplctag.NET-master\src\libplctag\NativeTagWrapper.cs:line 413
   at libplctag.NativeTagWrapper.Read() in C:\Users\jfont\Desktop\libplctag.NET-master\src\libplctag\NativeTagWrapper.cs:line 229
   at libplctag.Tag.Read() in C:\Users\jfont\Desktop\libplctag.NET-master\src\libplctag\Tag.cs:line 170
   at libplctag.Tag`2.Read() in C:\Users\jfont\Desktop\libplctag.NET-master\src\libplctag\TagOfT.cs:line 123
   at ProvaCommsOmron.Program.Main() in C:\Users\jfont\Desktop\libplctag.NET-master\src\ProvaCommsOmron\Program.cs:line 22
```

Checked what is going on with WireShark and seems that `libplctag` is using service `0x52` to request the tag and the PLC answers `Service not supported`.

After that I checked the code for the original C library and I saw that there is an option to enable communication with Omron PLC which makes it use service `0x4c` instead of `0x52` (see file src/protocols/ab/defs.h, line 82 and src/protocols/ab/eip_cip.c, line 427 to 432). 

I confirmed that this service code should work by inspecting the output of Cx-Compolet with WireShark (the official Omron Ethernet/IP CIP driver), which also uses service `0x4c`. I don't want to use the official driver as it is closed source and requires additional software to be installed in the computer, which is only available for Windows.

However, I haven't been able to convince libplctag.NET to use service code `0x4c`. I'm a .NET programmer and I know little C. I would be very grateful if someone could show some C# code that makes communication using service code `0x4c`.

lcoeli, 2021-05-14:
PLC type should be Micro800 right?

jfontsaballs, 2021-05-17:
Tried:

```csharp
static void Main()
{
	var bool1 = new Tag<IntPlcMapper, short>
	{
		Name = "ProvaInt1",
		Gateway = "172.26.5.11",
		Path = "1,0",
		PlcType = PlcType.Micro800,
		Protocol = Protocol.ab_eip,
		Timeout = TimeSpan.FromMilliseconds(500),
		UseConnectedMessaging = true
	};

	bool1.Initialize(); // This line throws LibPlcTagException
	Console.WriteLine("Initialized");

	bool1.Read();
	Console.WriteLine($"Value: {bool1.Value}");
}
```

Got this exception on `bool1.Initialize()`:

```
libplctag.LibPlcTagException
  HResult=0x80131500
  Message=ErrorUnsupported
  Source=libplctag
  StackTrace:
   at libplctag.NativeTagWrapper.Initialize()
   at libplctag.Tag.Initialize()
   at libplctag.Tag`2.Initialize()
   at TestComms.Program.Simple() in C:\Users\jfont\Desktop\TestComms\TestComms\Program.cs:line 53
   at TestComms.Program.Main() in C:\Users\jfont\Desktop\TestComms\TestComms\Program.cs:line 12
```

Wireshark analysis confirms service `0x52` is still being used, which is not supported by Omron PLC.

```
Source                Destination           Protocol Length Info
172.26.254.60         172.26.5.11           CIP      120    'ProvaInt1' - Service (0x52)
172.26.5.11           172.26.254.60         CIP      104    Service not supported: 'ProvaInt1' - Service (0x52)
```

jfontsaballs, 2021-05-17:
Finally got it working:

```csharp
static void Main()
{
	var tagHandle = plctag.plc_tag_create("protocol=ab_eip&gateway=172.26.5.11&path=1,0&plc=omron-njnx&name=ProvaInt1", 1000);
	
	var statusAfterCreate = (STATUS_CODES)plctag.plc_tag_status(tagHandle);
	if (statusAfterCreate != STATUS_CODES.PLCTAG_STATUS_OK)
	{
		throw new Exception($"Something went wrong: {statusAfterCreate}");
	}

	var statusAfterRead = (STATUS_CODES)plctag.plc_tag_read(tagHandle, 1000);
	if (statusAfterRead != STATUS_CODES.PLCTAG_STATUS_OK)
	{
		throw new Exception($"Something went wrong: {statusAfterRead}");
	}

	var theValue = plctag.plc_tag_get_uint16(tagHandle, 0);

	Console.WriteLine(theValue);

	plctag.plc_tag_destroy(tagHandle);
}
```

The .NET library's enum libplctag.PlcType does not have an option for Omron PLCs. I believe that the solution will be as simple as adding it (and handling it in the appropriate places).

kyle-github, 2021-05-19:
Thanks for pointing out that this is not documented.  I will fix that this week!

jfontsaballs, 2021-05-19:
I have added a [pull request](https://github.com/libplctag/libplctag.NET/pull/170) that allows access to the Omron PLC option using Tag<,> object.

moonps1, 2021-05-20:
@jfontsaballs @kyle-github @jkoplo Thanks a lot.

closed 2021-08-19 by timyhac