libplctag/libplctag.NET #171: Hi Error Connect with Micro800 Time Out Error

nainesh72, 2021-05-18:
Hi
I not able to Read Tage from Micro800, 
var myTag = new Tag<IntPlcMapper, short>()
            {
                Name = "C1_Priority",
                Gateway = "190.168.1.21",
                Path = "1,0",
                PlcType = PlcType.Micro800,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5),
                DebugLevel = DebugLevel.Detail,
            };
            
I send detail also
2021-05-18 09:54:29.855 thread(1) tag(0) INFO plc_tag_create:578 Starting
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:177 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO lib_init:130 Setting up global library data.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO lib_init:132 Creating tag hashtable.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO hashtable_create:75 Starting
2021-05-18 09:54:29.855 thread(1) tag(0) INFO hashtable_create:99 Done
2021-05-18 09:54:29.855 thread(1) tag(0) INFO lib_init:138 Creating tag hashtable mutex.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO lib_init:144 Creating tag tickler thread.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL thread_create:791 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL thread_create:825 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO lib_init:150 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2021-05-18 09:54:29.855 thread(2) tag(0) INFO tag_tickler_func:197 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO ab_init:121 Initializing AB protocol library.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO ab_init:130 Finished initializing AB protocol library.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO mb_init:1813 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mb_init:1815 Setting up mutex.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:633 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL mutex_create:660 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO mb_init:1824 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO initialize_modules:231 Done.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2021-05-18 09:54:29.855 thread(1) tag(0) INFO ab_tag_create:171 Starting.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:178
2021-05-18 09:54:29.855 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0000028063043190
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL ab_tag_create:184 tag=0000028063043190
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL get_plc_type:908 Found Micro8xx PLC.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL ab_tag_create:236 Micro800 needs connected messaging.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL get_plc_type:908 Found Micro8xx PLC.
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2021-05-18 09:54:29.855 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2021-05-18 09:54:29.855 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL session_create_unsafe:449 Session should use connected messaging.
2021-05-18 09:54:29.870 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2021-05-18 09:54:29.870 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00000280630EBB80
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 1,0.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 1.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 2 in string 1,0.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 0.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2021-05-18 09:54:29.870 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router object.
[... 122 lines cut ...]
2021-05-18 09:54:34.948 thread(1) tag(0) DETAIL remove_session:379 Done.
2021-05-18 09:54:34.948 thread(1) tag(0) INFO session_destroy:784 Session sent 2 packets.
2021-05-18 09:54:34.948 thread(1) tag(0) DETAIL session_destroy:790 Destroying session thread.
2021-05-18 09:54:34.963 thread(3) tag(0) DETAIL purge_aborted_requests_unsafe:1250 Session thread releasing aborted request 00000280630D0780.
2021-05-18 09:54:34.963 thread(3) tag(0) DETAIL rc_dec_impl:242 Calling cleanup functions due to call at purge_aborted_requests_unsafe:1257 for 00000280630D0780.
2021-05-18 09:54:34.963 thread(3) tag(0) INFO refcount_cleanup:256 Starting
2021-05-18 09:54:34.963 thread(3) tag(0) DETAIL request_destroy:2376 Starting.
2021-05-18 09:54:34.963 thread(3) tag(0) DETAIL request_destroy:2385 Done.
2021-05-18 09:54:34.963 thread(3) tag(0) INFO refcount_cleanup:268 Done.
2021-05-18 09:54:34.963 thread(3) tag(0) DETAIL purge_aborted_requests_unsafe:1265 Removed 1 aborted requests.
2021-05-18 09:54:34.963 thread(1) tag(0) INFO session_unregister:741 Starting.
2021-05-18 09:54:34.963 thread(1) tag(0) INFO session_unregister:745 Done.
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL session_destroy:838 Destroying session mutex.
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL mutex_destroy:748 destroying mutex 00000280630ECBE8
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL mutex_destroy:761 Done.
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL session_destroy:844 Cleaning up allocated memory for paths and host name.
2021-05-18 09:54:34.963 thread(1) tag(0) INFO session_destroy:860 Done.
2021-05-18 09:54:34.963 thread(1) tag(0) INFO refcount_cleanup:268 Done.
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL mutex_destroy:748 destroying mutex 00000280630431B8
2021-05-18 09:54:34.963 thread(1) tag(0) DETAIL mutex_destroy:761 Done.
2021-05-18 09:54:34.979 thread(1) tag(0) DETAIL mutex_destroy:748 destroying mutex 00000280630431C0
2021-05-18 09:54:34.979 thread(1) tag(0) DETAIL mutex_destroy:761 Done.
2021-05-18 09:54:34.979 thread(1) tag(0) INFO ab_tag_destroy:849 Finished releasing all tag resources.
2021-05-18 09:54:34.979 thread(1) tag(0) INFO ab_tag_destroy:851 done
2021-05-18 09:54:34.979 thread(1) tag(0) INFO refcount_cleanup:268 Done.

nainesh72, 2021-05-18:
Hi
I get Solution for micro800 i remove  path from tag reading follow

 var myTag = new Tag<IntPlcMapper, short>()
            {
                Name = "C1_Priority",
                Gateway = "190.168.1.21",
               // Path = "1,0",
                PlcType = PlcType.Micro800,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5),
                DebugLevel = DebugLevel.Detail,
            };

closed 2021-05-18 by nainesh72

kyle-github, 2021-05-18:
Yep, it is in the errors:

```text
2021-05-18 09:54:29.980 thread(3) tag(0) WARN receive_forward_open_response:2164 Forward Open command failed, response code: Port Not Available (1)
```
There should probably be a better way for me to return the underlying error.   Earlier, it is noted in the log output that a path is not supported.  Perhaps I should make this a hard error.  I need to think about this.