libplctag/libplctag.NET #172: How to reading multiple tags

TrongViet94, 2021-05-19:
Dear @timyhac ,
I ReadAsync for 1000 tag REAL finished about 19 seconds. 
Is there a way to make it update faster?
This my code:
        public static async Task Run()
        {
            List<Tag<RealPlcMapper, float>> myTags = new List<Tag<RealPlcMapper, float>>();
            for (int i = 0; i < ListTagTest.List1.Count(); i++)
            {
                var myTag1 = new Tag<RealPlcMapper, float>()
                {
                    Name = ListTagTest.List1[i],
                    Gateway = "150.168.168.230",
                    Path = "1,0",
                    PlcType = PlcType.ControlLogix,
                    Protocol = Protocol.ab_eip,
                    Timeout = TimeSpan.FromSeconds(1000)
                };
                myTags.Add(myTag1);
                await myTag1.InitializeAsync();
            }
            ///
            Console.WriteLine("begin");
            var asyncStopWatch = Stopwatch.StartNew();
            foreach (var item in myTags)
            {
                await item.ReadAsync();
                Console.WriteLine(item.Value);
            }
            asyncStopWatch.Stop();
            Console.WriteLine($"\ttook{(float)asyncStopWatch.ElapsedMilliseconds}ms on average");

        }

timyhac, 2021-05-19:
You are awaiting for each read to finish before starting the next one, wait for them concurrently - maybe use Task.WaitAll or similar

TrongViet94, 2021-05-20:
Oh thanks @timyhac .
it improves speed quite a lot (1000tag => 0.8 s).
But it's not as good as I wanted.
16000tag =>13s.

jkoplo, 2021-05-20:
@kyle-github has a great thread on optimizing read performance here: https://groups.google.com/g/libplctag/c/iDS9Aeh00ug/m/jW_WnK4CCwAJ

TL; DR: - there are a lot of factors

TrongViet94, 2021-05-26:
I have a problem using  Task.WaitAll  and ReadAsync();
When the PLC connection is lost, the CPU(core I7)  increases to 90% power.
Thanks @jkoplo @timyhac !

kyle-github, 2021-05-26:
Hi @TrongViet94,

Can you please capture debugging output when this CPU load happens?  There is enough evidence for me to believe that there may be a load problem in the core C library under Windows.   A Core i7 should have more than enough "horsepower" to run against many PLCs at the same time without going over 1 or 2% load!

jkoplo, 2021-05-26:
Our async implementation is... bad. We basically just stubbed it in because we couldn't sort out all of the C to C# marshalling of callbacks to async stuff. I think we just create a new thread for each call and don't use @kyle-github nice async methods.

timyhac, 2021-05-29:
One way to test this is to just run 16,000 concurrent calls to Task.Delay(60000)  (instead of tag.ReadAsync()) and see what happens to CPU usage.'

This is turning into a discussion - we've answered the original question right? @jkoplo - we could have another crack at the async implementation, we still have that early effort here: https://github.com/libplctag/libplctag.NET/blob/async/src/libplctag/Tag.cs. I'll create a ticket.

closed 2021-07-21 by timyhac