libplctag/libplctag.NET #173: How to read/write strings

tneon1, 2021-05-24:
I am connecting to a Contrologix process and can read and write all data types except the string data type. the string tag is in the processors main tag list and I am using the example code from the libplctag.net. I have my code is a separate function that is called to read or write the specific data type needed. Not sure what I am missing or if this is correct forum for this question.
Here is the code to try and write the string tag tot he PLC tag. The error is that the stringtag.value is not created.
 
                    var stringTag = new Tag<StringPlcMapper, string[]>()
                    {
                        Name = "AlitempString",
                        Gateway ="192.168.2.3", //SetupCfg.IpAddr,
                        Path = "1,0",
                        PlcType = PlcType.ControlLogix, //(PlcType)SetupCfg.SystemType,
                        Protocol = Protocol.ab_eip,
                        Timeout = TimeSpan.FromSeconds(5000),
                        ArrayDimensions = new int[] { 100 },
                    };

                    //stringTag.Initialize();
                    var r = new Random((int)DateTime.Now.ToBinary());
                    for (int ii = 0; ii < 100; ii++)
                    {
                        stringTag.Value[ii] = r.Next().ToString();
                    }

                    stringTag.Write();
                    //Result = stringTag.Value;
                    //stringTag.Dispose();
                    break;

jkoplo, 2021-05-25:
Hmm, I took a quick look at your code and my first thought is that this _may_ be a bug with initializing the actual `Value` array within the tag object. Can you try doing a `stringTag.Read()` before your loop that writes the values? See if it gets the current values back from the PLC.

tneon1, 2021-05-25:
Thank You for the reply,
I added the stringtag.initilization() back in and get the following error
[image]

Here is the output.

'WindowsFormsApp7.exe' (CLR v4.0.30319: WindowsFormsApp7.exe): Loaded 'C:\Users\tfrink\source\repos\WindowsFormsApp7\bin\Debug\libplctag.NativeImport.dll'. Symbol loading disabled by Include/Exclude setting.
'WindowsFormsApp7.exe' (CLR v4.0.30319: WindowsFormsApp7.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Runtime.InteropServices.RuntimeInformation\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.InteropServices.RuntimeInformation.dll'. Symbol loading disabled by Include/Exclude setting.
Exception thrown: 'libplctag.LibPlcTagException' in libplctag.dll
An unhandled exception of type 'libplctag.LibPlcTagException' occurred in libplctag.dll
ErrorOutOfBounds

Unhandled Exception: libplctag.LibPlcTagException: ErrorOutOfBounds
   at libplctag.NativeTagWrapper.Initialize()
   at libplctag.Tag`2.Initialize()
   at WindowsFormsApp7.GBL.GetTag(enType DataType, String TaskName, String TagName, SetupItems SetupCfg, Boolean write, Boolean Val) in C:\Users\tfrink\source\repos\WindowsFormsApp7\GBL.cs:line 310
   at WindowsFormsApp7.Form1.Form1_Load(Object sender, EventArgs e) in C:\Users\tfrink\source\repos\WindowsFormsApp7\Form1.cs:line 84
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
The program '[1860] WindowsFormsApp7.exe' has exited with code 0 (0x0).

jkoplo, 2021-05-26:
I have access to a CLogix PLC again, so I'll try giving this a shot. Are you sure you're actual tag is a string array with 100 elements?

tneon1, 2021-05-26:
My string is a single string that is 25 to 35  characters. I did try 1 and 0 for the array size.ThanksTomSent from my T-Mobile 5G Device

tneon1, 2021-05-27:
Here is the actual string tag and entry.
[image]

Thank You
Tom

jkoplo, 2021-05-27:
Ah, you're C# code is trying to read a tag that is an **array** of 100 strings, not a single string. In C#, strings are not treated as an array of characters, but as an individual value.

You'll want to do something like this, where the Tag is instantiated as `string` instead of `string[]`. Also there's no need to loop.
```C#
//Instantiate the tag with the proper mapper and datatype
var myTag = new Tag<StringPlcMapper, string>()
{
    Name = "PROGRAM:SomeProgram.SomeSTRING",
    Gateway = "10.10.10.10",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};

//Initialize the tag to set up structures and prepare for read/write
//This is optional as an optimization before using the tag
//If omitted, the tag will initialize on the first Read() or Write()
myTag.Initialize();

//The value is held locally and only synchronized on Read() or Write()
myTag.Value = "Test12345";

//Transfer Value to PLC
myTag.Write();

//Transfer from PLC to Value
myTag.Read();

//Write to console
string myString = myTag.Value;
Console.WriteLine(myString);

```

timyhac, 2021-05-29:
As @jkoplo mentioned this issue was caused by incorrect type arguments.

closed 2021-05-29 by timyhac

tneon1, 2021-05-29:
Thank you for the information, it worked great.
Thanks again

kaptindan, 2023-02-28:
I tried this and it does not work for me. 

//Instantiate the tag with the proper mapper and datatype
var myTag = new Tag<StringPlcMapper, string>()
{
    Name = "PROGRAM:SomeProgram.SomeSTRING",
    Gateway = "10.10.10.10",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};

//Initialize the tag to set up structures and prepare for read/write
//This is optional as an optimization before using the tag
//If omitted, the tag will initialize on the first Read() or Write()
myTag.Initialize();

//The value is held locally and only synchronized on Read() or Write()
myTag.Value = "Test12345";

//Transfer Value to PLC
myTag.Write();

//Transfer from PLC to Value
myTag.Read();

//Write to console
string myString = myTag.Value;
Console.WriteLine(myString);