libplctag/libplctag.NET #174: Intermittent Timeout Exception when calling Tag.Initialize()

WetTheSystem, 2021-05-27:
Reposting here from the libplctag github. Developer suggested to post it here as it is likely a unique issue for libplctag.net

libtagplc v1.0.3

libplctag.LibPlcTagException
HResult=0x80131500
Message=ErrorTimeout
Source=libplctag
StackTrace:
at libplctag.Tag.Initialize()
at PLCTagInterfaceHandler.Service1.CheckUnackedAlarms(Object source, ElapsedEventArgs e) in C:\Users\tjohnson8\source\repos\PLCTagInterfaceHandler\Service1.cs:line 284
at System.Timers.Timer.MyTimerCallback(Object state)

Some context: This code is installed on 2 servers and running concurrently. I occasionally get a timeout that throws an unhandled exception, but only ever on one server.

Here's the snippet that creates the exception:

var unackedArray = TagGenerator.TagGenArray("AlarmCheck", 44, "bool", ###.###.###.###");
unackedArray.Initialize();

Here's a code snippet from class TagGenerator that creates this bool array:

var boolTag = new Tag<BoolPlcMapper, bool[]>()
{
Name = tag_name,
Gateway = gateway,
Path = path,
PlcType = PlcType.ControlLogix,
Protocol = Protocol.ab_eip,
ArrayDimensions = new int[] { array_length, array_length },
Timeout = TimeSpan.FromMilliseconds(1000),
};
return boolTag;

Can you give some suggestions on where this timeout originates from so I can troubleshoot it, or if you have seen it in the past provide some insight? Is it because I'm grabbing such a large array of bools (a length value of 44 results in a bool array of size 1952 which may be some issue itself. I derived this value by just checking my result to get an array size as close as possible to 2000)

jkoplo, 2021-05-27:
A timeout of 1000 msec seems pretty low. PLC network comms are pretty lousy and industrial networks are often noisy. Maybe try increasing that value, at least as a debug test?

I'm not sure why a Bool array of 44 elements would be 1952 bytes. Where are you getting that size from?

timyhac, 2021-05-29:
I can't see this being an issue with libplctag.NET - I guess its not impossible that it is somehow to do with libplctag core, but more likely to do with other factors.

P.S. you have a square 2d array dimensions, but the PlcMapper type argument is 1d array. 44*44 = 1936. The next multiple of 32 after that is 1952 which I guess is where 1952 comes from. Something is wrong here. 

Having 2 computers simultaneously request this much data from the PLC could starve it when you're only waiting 1sec, they are competing for network capacity, which could explain why its an intermittent problem.

Try upping the timeout as Jody mentioned.

I'm quite sure this is not a libplctag.NET issue.

closed 2021-05-29 by timyhac