libplctag/libplctag.NET #175: Event-based async implementation

timyhac, 2021-05-29:
The current async implementation is a stub, in that it uses a polling approach to check whether the Read/Write action is completed.

It is likely that it would be a more efficient use of system resources to use the callback/event system available in libplctag core.

This ticket captures to implement an event-based async implementation and determine whether it is more efficient and should be the default implementation for libplctag.NET.

timyhac, 2021-07-21:
Hi @jkoplo - I've released an alpha version of the implementation as 1.0.7-alpha.2.

I've been testing it with my Human-machine-interface app and is working OK, including throwing errors. Is this something you'd be interested in testing before releasing as a patch version?

It is a bit hard to tell but the CPU performance is better I think (we no longer need to do all of those Task.Delay()s). The network performance is about the same.

Tim

jkoplo, 2021-07-21:
What kinda errors? I skimmed your commits today, but didn't actually dig in enough to see what's going on.

timyhac, 2021-07-21:
@jkoplo - like TImeouts and the other normal failures.

e.g. if you run ReadAsync() concurrently on the same tag, one of them works and the other one throws an ErrorBusy, which is the same as what you get if you do the same with the native API.

jkoplo, 2021-07-21:
Oh, "expected" errors! 😆

I'll try to play with it soon. Releasing as an alpha probably makes sense as long as you've done some testing on it.

timyhac, 2021-08-16:
Hi @jkoplo - this has been released as v1.0.7

closed 2021-08-16 by timyhac

jkoplo, 2021-08-16:
Very cool. I'm hoping work settles down and I can dive into this again.