libplctag/libplctag.NET #177: IOException: Sharing violation on path "..\plctag.dll"

the-pied-shadow, 2021-06-02:
Hello. I'm attempting to run code very similar to your example.cs to read and write to a boolean tag. However, when it is run I get this error: "IOException: Sharing violation on path C:\Users\<user>\Desktop\<project>\Assets\Plugins\plctag.dll"

I've been unable to understand the issue anymore through research. 

Here is the callback:

```
IOException: Sharing violation on path C:\Users\bhilsdon\Desktop\PLC Networking\Assets\Plugins\plctag.dll
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <9577ac7a62ef43179789031239ba8798>:0)
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize) (at <9577ac7a62ef43179789031239ba8798>:0)
(wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int)
System.IO.File.Create (System.String path, System.Int32 bufferSize) (at <9577ac7a62ef43179789031239ba8798>:0)
System.IO.File.Create (System.String path) (at <9577ac7a62ef43179789031239ba8798>:0)
System.IO.File.WriteAllBytes (System.String path, System.Byte[] bytes) (at <9577ac7a62ef43179789031239ba8798>:0)
libplctag.NativeImport.LibraryExtractor.ExtractAppropriateLibraryToDirectory (System.String outputDirectory) (at <50abcd0f95e747348b5f7c669256d26d>:0)
libplctag.NativeImport.LibraryExtractor.Init (System.Boolean forceExtract) (at <50abcd0f95e747348b5f7c669256d26d>:0)
libplctag.NativeImport.plctag.ExtractLibraryIfRequired () (at <50abcd0f95e747348b5f7c669256d26d>:0)
libplctag.NativeImport.plctag.plc_tag_create (System.String lpString, System.Int32 timeout) (at <50abcd0f95e747348b5f7c669256d26d>:0)
libplctag.NativeTag.plc_tag_create (System.String lpString, System.Int32 timeout) (at <18439e085ec54e2891bd246d29cef40d>:0)
libplctag.NativeTagWrapper.Initialize () (at <18439e085ec54e2891bd246d29cef40d>:0)
libplctag.Tag.Initialize () (at <18439e085ec54e2891bd246d29cef40d>:0)
libplctag.Tag`2[M,T].Initialize () (at <18439e085ec54e2891bd246d29cef40d>:0)
```

As mentioned above my code is very similar to example.cs. The variables used below are set elsewhere. 

```
        Tag<BoolPlcMapper, bool> mytag = new Tag<BoolPlcMapper, bool>()
        {
            Name = TagName,
            Gateway = GatewayIPAddress,
            Path = path,    
            PlcType = plc,
            Protocol = protocol,
            Timeout = TimeSpan.FromSeconds(5)
        };

        mytag.Initialize();
        mytag.Value = true;
        mytag.Write();
        mytag.Read();
        Debug.Log(mytag.Value);
```

The notable outlying factor here is I'm attempting to use in Unity Engine. 

Thank you for this tool and your help.

the-pied-shadow, 2021-06-02:
Upon some further investigation I've realized that what first happens is I get an "EntryPointNotFound" exception. After this exception is thrown and the program is run again, I get the sharing violation instead. This process can be reproduced when I delete the plctag.dll. So the original issue may just be that I have the wrong information for the tag.

jkoplo, 2021-06-02:
You may realize this (or have figured it out) but in order to support multiple platforms, we extract the proper base libplctag DLL to a directory before importing it and wrapping those calls in our C# library. There's more information about this in main Readme.

What it looks like is that you might have a Windows permission issue based on where it's extracting and how your app is running. That might be caused by your initial error or not...

the-pied-shadow, 2021-06-07:
I appreciate the reply. I'm afraid I'm not well experienced with PLCs or Ethernet/IP communication and I understand if you don't have time to explain things that aren't related to your package. 

When I get the error:
`EntryPointNotFoundException: plc_tag_create`

Am I correct in interpreting this as it failing to connect to the PLC or could this be that it did connect and didn't find the tag I specified?

timyhac, 2021-06-07:
No, this issue is all related to the inability of the NativeImport to extract the appropriate libplctag core library properly.

That error is saying that PInvoke can't find the C function in the native dll.

To get around this issue, get a copy of the appropriate plctag.dll and configure ForceExtract to false.
Reading this will help:
https://github.com/libplctag/libplctag.NET#libplctagnativeimport

the-pied-shadow, 2021-06-07:
Okay thank you that does bring some clarity for me. Could you point me in the right direction for getting a copy of the appropriate plctag.dll?

timyhac, 2021-06-07:
You can download them from the libplctag releases page
https://github.com/libplctag/libplctag/releases

timyhac, 2021-06-07:
In terms of what this project could do to resolve this issue, I believe that #137 is key

closed 2021-06-07 by timyhac

lakerty, 2022-03-02:
Hi just a quick question has anybody done a visual studio xamarin forms  android  phone project for CompactLogix plc

jkoplo, 2022-03-02:
I don't believe so, but it should be possible (maybe). Can you open a new issue and we can discuss in there?

timyhac, 2022-03-03:
Yep it worked last time I checked as long as you follow the [README](https://github.com/libplctag/libplctag.NET#developing-for-systems-with-immutable-application-directories).

lakerty, 2022-03-03:
Hi I have read this file a few times my knowledge of c# and xamarin forms are limited as I am just starting to learn these.As I use mainly vb.net and my main strengths are plc programs I am getting lost on readme file. Many thanks