libplctag/libplctag.NET #179: Possible bug for individual bool array index

TheFern2, 2021-06-15:
I think this is a possible bug, I can read one index of any type array except for boolean array.
I am on the latest release v1.0.5, Contrologix PLC

With this code I can read any individual tag, and also one individual index from an array, but not from boolean arrays.

```csharp
public async Task<Response<string>> Read(string tagName, TagType tagType)
        {
            switch (tagType)
            {
                case TagType.Bool:
                    return await _ReadTag<BoolPlcMapper, bool>(tagName);
                case TagType.Bit:
                    return await _ReadTag<BoolPlcMapper, bool>(tagName);
                case TagType.Dint:
                    return await _ReadTag<DintPlcMapper, int>(tagName);
                case TagType.Int:
                    return await _ReadTag<IntPlcMapper, short>(tagName);
                case TagType.Sint:
                    return await _ReadTag<SintPlcMapper, sbyte>(tagName);
                case TagType.Lint:
                    return await _ReadTag<LintPlcMapper, long>(tagName);
                case TagType.Real:
                    return await _ReadTag<RealPlcMapper, float>(tagName);
                case TagType.String:
                    return await _ReadTag<StringPlcMapper, string>(tagName);
                default:
                    return new Response<string>(tagName, "Wrong Type");
            }
        }

private async Task<Response<string>> _ReadTag<M, T>(string tagName) where M : IPlcMapper<T>, new()
        {
            var results = await ReadTag<M, T>(tagName).ConfigureAwait(false);
            if (results.Status == "Success")
            {
                return new Response<string>(tagName, results.Value.ToString(), "Success");
            }
            else
            {
                return new Response<string>(tagName, "Failure");
            }
        }

```

```csharp
public async Task<Response<T>> ReadTag<M, T>(string tagName) where M : IPlcMapper<T>, new()
        {

            var tag = new Tag<M, T>()
            {
                Name = tagName,
                Gateway = _ipAddress,
                Path = _path,
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(Timeout)
            };

            try
            {
                await tag.InitializeAsync().ConfigureAwait(false);
                await tag.ReadAsync().ConfigureAwait(false);
                var tagValue = tag.Value;
                tag.Dispose();

                return new Response<T>(tagName, tagValue, "Success");
            }
            catch (Exception)
            {
                return new Response<T>(tagName, "Failure");
            }
        }
```

```
var result2 = await myPLC.Read("BaseBOOLArray[4]", TagType.Bool);

"C:\Program Files\JetBrains\JetBrains Rider 2021.1.2\plugins\dpa\DotFiles\JetBrains.DPA.Runner.exe" --handle=4108 --backend-pid=26984 --detach-event-name=dpa.detach.4108 D:/git/clx.libplctag.NET/clx.libplctag.NET.TestProgram/bin/Debug/netcoreapp3.1/clx.libplctag.NET.TestProgram.exe
libplctag.LibPlcTagException: ErrorBadParam
   at libplctag.NativeTagWrapper.ThrowIfStatusNotOk(Nullable`1 status)
   at libplctag.NativeTagWrapper.InitializeAsync(CancellationToken token)
   at libplctag.Tag`2.InitializeAsync(CancellationToken token)
   at clx.libplctag.NET.PLC.ReadTag[M,T](String tagName) in D:\git\clx.libplctag.NET\clx.libplctag.NET\PLC.cs:line 756
[Response: BaseBOOLArray[4]  Failure]

```

Line 756 is the ReadTag function which works fine for other types:
```
var result2 = await myPLC.Read("BaseDINTArray[4]", TagType.Dint);
[Response: BaseDINTArray[4] 999585050 Success]

```

TheFern2, 2021-06-16:
Btw latest bool mapper works great for single bool tags, and boolean arrays. The issue only occurs for single index on a boolean array.

TheFern2, 2021-06-18:
I was able to work some magic by mapping tag manually:

Input would be something like BaseBOOLArray[6] then I would extract index and provide to below function.

```csharp
public async Task<Response<string>> ReadBoolArraySingle(string tagName, int index)
{
    var tag = new Tag();
    tag.Name = tagName;
    tag.Gateway = _ipAddress;
    tag.Path = _path;
    tag.PlcType = PlcType.ControlLogix;
    tag.Protocol = Protocol.ab_eip;
    tag.Timeout = TimeSpan.FromSeconds(Timeout);
    tag.ElementCount = (index / 32) + 1;

    try
    {
        await tag.InitializeAsync().ConfigureAwait(false);
        var tagValue = tag.GetBit(index);
        tag.Dispose();

        return new Response<string>(tagName, tagValue.ToString(),"Success");
    }
    catch (Exception e)
    {
        return new Response<string>(tagName, e.Message);
    }
}
```

timyhac, 2021-06-19:
Thanks for checking this @TheFern2. İ have a feeling this is due to some underlying limitations with the protocol. Are you able to check by using the command line tools provided by libplctag?

TheFern2, 2021-06-19:
Sure thing, I can check Monday should I just check with the tag_rw binary and put the brackets inside the tag name string?

timyhac, 2021-06-21:
Yes that's right, i think the type needs to be "bit".

TheFern2, 2021-06-28:
Closing this, I am unable to verify if core libplctag can read one bool array bit, it doesn't like brackets. Either way I got around this, so is a non issue for me.

closed 2021-06-28 by TheFern2