libplctag/libplctag.NET #180: Pulling whole tag list

th3dj, 2021-06-24:
I am having issues with trying to pull the whole tag list. I looked at the C# versions and I think I supposedly have it right.

        Me.tags = New Tag(Of TagInfoPlcMapper, TagInfo)() With {
            .Name = "@tag",
            .Protocol = 0,
            .Gateway = ip,
            .PlcType = PlcType.ControlLogix,
            .Path = path
        }
The error I'm coming back with is:
Type argument 'TagInfoPlcMapper' does not inherit from or implement the constraint type 'IPlcMapper(Of TagInfo)'.

I am kinda new to using this library so i9 apologize if this is a stupid question

timyhac, 2021-06-25:
İt needs to be a Tagİnfo array, not a single Tagİnfo.

timyhac, 2021-06-26:
Also, `@tags` not `@tag` - please see the example code here: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleListTags.cs

closed 2021-06-26 by timyhac

th3dj, 2021-06-28:
Thank you, I appreciate the help in the matter!