libplctag/libplctag.NET #181: Error with SetInt8() and SetUInt8()

joao147, 2021-07-08:
I'm integrating Omron-nj-nx series on a SCADA and using Tag(it isn't Tag<M, T>) to read/write all Get/Set all types are working perfectly, expect set to Sint(byte).

timyhac, 2021-07-09:
Hi @joao147 thanks for the feedback.

Are you able to confirm this by using the native API? Something like the Run() method in this example: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20NativeImport/NativeImportExample.cs#L9

joao147, 2021-07-12:
Hello @timyhac thank you to answer really quickly,

I tried as you said, but continue return error 29. The thing that I think really strange is GetInt8 and GetUInt8 are work correct.
[image]

timyhac, 2021-07-14:
Thanks for confirming. This means that it's an issue with the core project, or i guess an issue with your device. Are you able file a bug on the core project?

joao147, 2021-07-14:
Ok, I tried to use a old wrapper that we have here from core project and receive the same error, just telling you, I going to open a issue on core project, maybe they know if an error with my device or core project. 
Thank for listening.

closed 2021-07-14 by joao147

timyhac, 2021-07-15:
Thanks @joao147